arendst/Tasmota #2809: mqtt broker on restart power off device

rt400, 2018-05-23:
hi ,
i have HA on raspberry with mqtt broker ( i install it no the emmeded)
when i do restart to mosquitto service all the sonoff devices got power off...
sometime its power on some device..
its make me crazy...
neeed help..plz..tnks

Jason2866, 2018-05-23:
I think your broker has stored retained messages and sends these again and again...
Delete the retained messages, and it will work.

rt400, 2018-05-23:
how do i delete those messages and how i keep that this wont happened again ?

Jason2866, 2018-05-23:
See #2140

ascillato, 2018-06-04:
Besides the mosquitto config, please be aware of having different mqtt topic for each sonoff and different client id

ascillato, 2018-06-04:
So, for allowing us to check that. Please use the provided template for troubleshoting. Share your mosquitto config (without your  password obviously) and the status 0 of a couple of your devices also.

GasTurbineMan, 2018-06-04:
In my setup, I am using Homeremote with my MQTT Broker running on Raspberry Pi 3.  In Homeremote, I send on  and off commands via topic cmnd/sonoffxx/POWER and payload either on or off.  I use topic stat/sonoffxx/POWER for status feedback.

I can send open and close commands via Homeremote and Sonoff works as expected.  If the Sonoff loses communication with the Broker and has to re-establish link - (or if I manually restart via web interface), it will start back up in the same state as what it was in before.  But, if the state of the relay is changed from the web interface or if you have a timer set up to switch the relay and then it loses comm with the Broker, it initially starts back up in the correct state but then switches to the incorrect state.

Example 1>  (how we want the Sonoff to work)
a.  Homeremote sends Power ON command.
b. Relay is now ON.
c.  Sonoff loses comms with Broker (or I manually restart via web interface)
d. Sonoff powers back up in the ON state and remains in the ON state.

Example 2>  (not desired)
a.  Homeremote sends Power ON command.
b.  Relay is not ON.
c.  Relay is toggled OFF via web interface (or a timer could have issued OFF command)
d.  Relay is now OFF.
e.  Sonoff loses comms with Broker and has to re-establish its link to Broker.
f.  Sonoff initially powers back up in the OFF state, but then switches to the ON state.
(see attached text from console, lines 9,10 and 11)
[MQT Issue 6-3-2018.txt](https://github.com/arendst/Sonoff-Tasmota/files/2068264/MQT.Issue.6-3-2018.txt)

I think I read where the Sonoff does not publish topics.  Is this correct?  So in example 2 above, we toggled the relay to the OFF position and sonoff console shows stat/sonoffxx/POWER = OFF but this message is not going to the Broker?  Is this the root of my issue?

rt400, 2018-06-04:
`# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example

pid_file /var/run/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log

include_dir /etc/mosquitto/conf.d
`
this is my conf of mosquitto server and i dont have password

GasTurbineMan, 2018-06-04:
I am just a beginner with MQTT, but I do have a user name and password set up for my broker.  My current user config file only has 2 lines in it:
       password_file /etc/mosquitto/passwd
       allow_anonymous false
 When I added the lines from your config above, and stopped and restarted my broker, none of the clients can access the broker, so I reverted back to my original conf file.

Back to the issue I attempted to describe and how it is affecting me in real life, I am using a Sonoff with timers set up to control outside lights on my house.  I have them set up to come on 2 hours before sunrise, go off 30 minutes after sunrise, then come back on 20 minutes before sunset and go off at 11:30pm.

I look at my Homeremote app on my tablet and I see the lights are ON and it is 10:30 am (well after sunrise) so I turn them OFF using the Homeremote app.  Lights will stay OFF, even if losing connection with broker from time to time, because the last command was OFF from the Homeremote app.

Then at 20 minutes before sunset, the Sonoff Timer turns the lights ON as desired.  Normally, these would be on until 11:30 PM but say around 9pm, the Sonoff loses connection with the Broker, has to re-establish connection and instead of remaining ON, they turn OFF - I assume because the last command from the Homeremote app was OFF at 10:30 am earlier that day.  Why doesn't the ON command issued by the Timer replace the earlier OFF command retained by the Broker?

GasTurbineMan, 2018-06-04:
Doing further testing, with my Homeremote app set up as a client and subscribing to the following 2 topics:
cmnd/sonoffxx/POWER
stat/sonoffxx/POWER
And publishing cmnd/sonoffxx/POWER with either on or off as payload, from the Homeremote app, I see where both of my subscribed topics update to ON and OFF as I toggle the switch via Homeremote app.
But, while monitoring topic status in Homeremote app (client) and toggling the relay via web interface, I see where topic stat/sonoffxx/POWER updates with each toggle from the web interface, but topic cmnd/sonoffxx/POWER never changes.
So is our problem that the Sonoff does not actually "publish" the state of cmnd/sonoffxx/POWER ?

I tried to set something up in Node-Red on my RPi3 that would monitor stat/sonoffxx/POWER and publish cmnd/sonoffxx/POWER accordingly, but ended up with a loop hammering the relay with on-off-on-off commands.  Had to break the connection in Node-Red to get out of the loop.

Jason2866, 2018-06-04:
@GasTurbineMan 
Every teleperiod time you get the status -> Power
You can do a status request with sending Power via mqtt with empty Payload. You will get immediatly the status.
`20:19:09 CMD: Power
20:19:09 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
20:19:09 MQT: stat/sonoff-065CA6/POWER = ON
20:19:16 CMD: Power off
20:19:16 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
20:19:16 MQT: stat/sonoff-065CA6/POWER = OFF
20:19:20 CMD: Power
20:19:20 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
20:19:20 MQT: stat/sonoff-065CA6/POWER = OFF`

Jason2866, 2018-06-04:
For clearance:
cmnd  is used for feedback for sended commands and will only appear one time as answer for the sended command
stat is used for (what it name suggests) for state and can be requested as often you want

GasTurbineMan, 2018-06-05:
So why cannot the Sonoff and the MQTT Broker get in synch with the current state of the relay when the relay state is changed either from a Web Interface "toggle" command or a "timer" instruction to change states.  The problem appears to be that neither of these updates the topic cmnd/sonoff/POWER.  Then when the relay goes offline and has to reestablish itself, the relay starts back up in the correct state, but then reaches out to the broker and broker changes it based upon feedback from cmnd/sonoff/POWER.

No doubt I need to change something on my side to deal with this issue, but right now, just not sure what.  Most likely I need to put Node-Red in between my Homeremote App and the Broker.

GasTurbineMan, 2018-06-05:
I don't really have a solution to this issue since I cannot control how Tasmota reacts to MQTT broker during restart.  I did some testing with topic cmnd/sonoffxx/POWER with various combination of retain flag set with my Homeremote app and Sonoff device.  Then simulated the various power on/off commands coming from either the HR app via Broker or the Tasmota Web Interface.   The PowerOnState of my Sonoff has always been set to 3 and in all cases, the Sonoff initially powers back up in the correct state - but then appears to get over-ruled by a command apparently from MQTT state.

Attached is a summary of my testing .  In no instance did the device pass all 4 testing scenarios.
[Tasmota MQTT PowerOn State testing.xlsx](https://github.com/arendst/Sonoff-Tasmota/files/2072374/Tasmota.MQTT.PowerOn.State.testing.xlsx)

Jason2866, 2018-06-05:
Why do you not query the state? If you unsure about the state.
There is always the possibility that the devices has a different state if something was "down" 
If the device changes its state during broker restart the problem is caused from the broker because
he sends something to the device (retained messages).
In my setup no device changes its state when wifi is lost or broker goes down and came back to working.
I use nowhere retain states (it is simply garbage) . After restart i just query the devices states to get the correct  view in the Grafic Panel.
May you think about your setup if it ever can handle your wishes or you need a other solution

GasTurbineMan, 2018-06-06:
Jason, thanks for the help.  You have been most helpful on another issue that I got resolved.

 

Here is the deal.  I am leaving it up to the Sonoff Timer in this case to turn the lights on and off.  It turns the lights ON at say 1 hour before sunrise and I expect them to stay on until the timer turns them off (say 30 minutes after sunrise).  The problem is that for unknown reasons, the Sonoff loses connection and has to restart.  It initially starts up with lights ON but then it turns them OFF.  The Broker is not telling the lights when to come on or off, right now it is the timer.  And, if the timer worked correctly with Sonoff Tasmota being able to accommodate communication disturbances, the Homeremote app would not have to inject any manual ON / OFF commands.

 

Using Node-Red, I have tried to set something up to monitor the current state based upon topic stat/sonoffxx/POWER and feed that back to cmnd/sonoffxx/POWER  in the event they get out of synch, but every time I do, I end up with a LOOP that sends the Sonoff relay into a frenzy.  To me, when the Sonoff Timer tells the relay to turn ON, it should update the topic cmnd/sonoffxx/POWER to show ON.  As it stands, when the timer tells the relay to go ON, stat/sonoffxx/POWER shows to be ON but cmnd/sonoffxx/POWER still shows OFF.  When the Sonoff loses comms with the broker, for whatever reason being a Sonoff issue or a network issue – it seems the Sonoff is getting its powerup state from the last cmnd/sonoffxx/POWER state which is incorrect in this scenario.

 

Other folks are reporting similar issues, so I don’t think this issue is isolated to me, but I will gladly accept any recommendations how to work around this problem.

 

Thank You!

rt400, 2018-06-06:
@Jason2866 can you tell me how to it ?

Jason2866, 2018-06-06:
@rt400 i dont understand your question. Be a bit more precise. 
I cant help you with hassio! OpenHab user....

GasTurbineMan, 2018-06-14:
@Jason2866, just wanted to follow up with my current status.  I disabled MQTT in my Sonoff's and the timers worked as expected for several days. So this confirmed I must still have retained messages.  Going back to your original recommendation regarding "retained messages",  using Node-Red as my client, I was able to finally get rid of all retained messages.  I had to leave the "retain" box checked when I sent my initial empty payload to my topic so that it would actually replace the retained command.

Regarding Tasmota and MQTT, the documentation is a bit confusing to me regarding what Tasmota does and does not publish.  Tasmota is "publishing" the topic "stat/sonoffxx/POWER" - correct?

And regarding "PowerRetain" as being ON or OFF, I assume this is the "retain" flag for the published topic stat/sonoffxx/POWER.  In my case, I choose to set PowerRetain ON.  This way, Tasmota publishes the current status for POWER and tells Broker to retain the message.  Then when new clients come on line and subscribe to topic stat/sonoffxx/POWER, they get the current status for the device and do not have to wait for the next state change of the relay when Tasmota publishes the new state.

While is took me a while, does my understanding seem correct?

Thank you for your help.

Jason2866, 2018-06-14:
To be honest, i dont know what is going on when using PowerRetain On.
I never used it, and i dont miss anything.
After restarting the HA System (OpenHab), i query the devices to get correct state.
If a device fails (for what reason ever) after teleperiod time i get the status update...

ascillato2, 2018-08-18:
Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-18 by ascillato2