arendst/Tasmota #2813: QUESTION: ! implementing IR control for an HVAC - with data 112 bits long

gnkarn, 2018-05-23:
i have completed the first draft to add a new AA i will manage using mqtt from Home assistant  , the code is [here](https://github.com/gnkarn/Multi-IrControl/tree/multi_bluesky), but I'm unsure on how to modify the sonoff.ino callback function in order to be able to parse a data payload that **is 14 bytes long( ir code is 112 bits) .**

something like : 
cmnd/Multi-IR/irsend {"Protocol": "BLUESKY","Bits":120,"data": "[0xC4, 0xD3, 0x64, 0x80, 0x00, 0x24, 0x80, 0xA0, 0x14, 0x00, 0x00, 0x00, 0x00, 0xE6]" }

**i appreciate if you can guide me to where in the code should I  focus , in order to study and understand the mqtt subscribe and parsing .**

arendst, 2018-05-24:
Just find the IRSend command code handler. No need to change anything in sonoff.ino

It's in the xdrv_02_irremote.ino file so you should be able to add your BlueSky to this file and process your data using ArduinoJson.

I suggest you start making changes to the latest development branch (currently 5.14.0b) to not deviate too much from the future.

gnkarn, 2018-05-24:
thank you , i understand will go in that direction, sending my own one byte commands and  decode into the bluesky IR send function .

gnkarn, 2018-07-08:
i have completed the implementation, and it works the same as the toshiba and mitsubishi hvac . would like too know how to proceed in case this can be added on  your main code .

ascillato2, 2018-07-10:
> i have completed the implementation, and it works the same as the toshiba and mitsubishi hvac . would like too know how to proceed in case this can be added on your main code .

Please, do a Pull Request. See github help for that. Thanks.

closed 2018-07-10 by ascillato2

6s1gm4, 2018-10-04:
instead of 

> something like :
> cmnd/Multi-IR/irsend { . . . }

  try with **CAPITAL LETTERS** at the last topic like this:

  cmnd/Multi-IR/**IRSEND** { . . . }