arendst/Tasmota #2814: Sonoff Basic - Tasmota 5.14a - Alexa

GasTurbineMan, 2018-05-24:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : 
- [x] Provide the output of command ``status 0`` :  
07:11:34 CMD: status 0
07:11:34 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Living Room Ceiling Fans"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
07:11:34 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T23:48:55","StartupUTC":"2018-05-23T12:22:39","Sleep":0,"BootCount":17,"SaveCount":183,"SaveAddress":"F5000"}}
07:11:34 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0a","BuildDateTime":"2018-05-21T04:38:05","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:11:34 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wind",""],"TelePeriod":300,"SetOption":["000080E9","55A18000"]}}
07:11:34 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":504,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F402794","14000000","01021790","00000000"]}}
07:11:34 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4930","IPAddress":"xxx.xxx.xxx.xxx","Gateway":"xxx.xxx.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx.xxx.xxx.xxx","Mac":"DC:4F:22:2B:B5:92","Webserver":2,"WifiConfig":3}}
07:11:34 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx.xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2BB342","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:11:34 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu May 24 12:11:34 2018","Local":"Thu May 24 07:11:34 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:15","Sunset":"20:30"}}
07:11:34 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-05-24T07:11:34","Switch1":"OFF"}}
07:11:34 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-05-24T07:11:34","Uptime":"0T23:48:55","Vcc":3.145,"POWER":"ON","Wifi":{"AP":1,"SSId":"Wind","RSSI":100,"APMac":"1C:B7:2D:C8:31:C0"}}}
07:12:25 MQT: tele/sonoff/STATE = {"Time":"2018-05-24T07:12:25","Uptime":"0T23:49:46","Vcc":3.143,"POWER":"ON","Wifi":{"AP":1,"SSId":"Wind","RSSI":100,"APMac":"1C:B7:2D:C8:31:C0"}}

Have posted this issue on a couple of other supports sites (one listed in above checklist) but not getting much help.  I am trying to get Alexa to recognize my Tasmota 5.14a flashed Sonoff Basics.  I have some emulating WeMo and some emulating Hue just to see if it will pick up either of them.  I have Smart Home skill enabled in Alexa and it picked up some stock Smart Life devices - they work as expected.  But will not pick up any of my emulated devices.
I tried enabling the WeMo skill in Alexa, but since I do not have any stock WeMo devices in my network, nor do I have a WeMo account to link to, I could not complete adding the skill.

Any instructions would be appreciated.  Thanks.

Also, in my Status 0 print out above, I edited some information for privacy.  Each of my Sonoff devices have a unique topic.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-05-24:
Your flash file size is for version 5.14.0a with sunrise/sunset function enabled really small.
You must have disabled other functions. Maybe here is a side effect.
In general 5.14.0a works with ALEXA. I use this version on my test sonoff basic since a few days
without a problem at all...
Could you post your userconfig.h?

GasTurbineMan, 2018-05-24:
As you would guess, I am a newbie using Tasmota and may have started trying to reduce the size of my firmware load before I really understood what I am doing.  At the current time, my Sonoff device collection consists of 1 Dual R2 and 10 Sonoff Basics.  They are currently running reliably via MQTT.  I am running my MQTT Broker on my Raspberry PI.

I am using PlatformIO to build and upload my firmwares.  Had to slow down my serial speed to 115200 in order to get reliable flashes.  Not having real good luck with OTA upgrades.  Doing the "Be Minimal" thing - but still not real good luck via OTA.   After some OTA upgrades, web interface does not load fully, some pages fail to load and overall response is very slow.  Had to reflash via serial and change CFG_HOLDER back and forth.

Attached is my userconfig.h and as you will see, I did disable several of the sensors which I did not believe I needed for the Sonoff Basic.

Hopefully you can help me get this Alexa thing to work.  Thank you.

[user_config.zip](https://github.com/arendst/Sonoff-Tasmota/files/2035869/user_config.zip)

GasTurbineMan, 2018-05-24:
One other thing, for Alexa to work with 5.14.0a, please confirm which Esp8266 core for Arduino version I need to be using.  Currently, I have gone back to 2.3.0

Jason2866, 2018-05-24:
Using 2.3.0 is the best choice. Only with this version everything in Tasmota works reliable.
@arendst is gone back to this version with his prebuild images after many user issues here with newer versions...
I recommend you to try the prebuild image. Your settings will NOT be lost if you update.
Just download the minimal image and upload it over the web interface.
After that load the final firmware. Try if Alexa now discovers the the device.
If a device is already discovered delete it before! I had problems after firmware upgrade
that this formerly working devices didnt respond. After delete and new discovery the are working reliable again.
P.s. i will take a look in your config.h

Jason2866, 2018-05-24:
`/*
  user_config.h - user specific configuration for Sonoff-Tasmota

  Copyright (C) 2018  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_

/*********************************************************************************************\
 * This file consists of TWO sections.
 *
 * SECTION 1:
 * The first section contains PARAMETERS overriding flash settings if define CFG_HOLDER is CHANGED.
 * All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
 *   Corresponding MQTT/Serial/Console commands are shown in [brackets]
 *
 * SECTION 2:
 * The second section contains Enabled and Disabled FEATURES allowing different program size.
 *   Changes in this section do NOT need a change of define CFG_HOLDER.
 *
 * ATTENTION:
 * Users are advised to use the user_config_override.h file for most changes.
\*********************************************************************************************/

//#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md

/*********************************************************************************************\
 * SECTION 1
 * - After initial load any change here only take effect if CFG_HOLDER is changed too
\*********************************************************************************************/

// -- Master parameter control --------------------
#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load SECTION1 configuration parameters to flash  default ...209

// -- Project -------------------------------------
#define PROJECT                "sonoff"          // PROJECT is used as the default topic delimiter
#define MODULE                 SONOFF_BASIC      // [Module] Select default model from sonoff_template.h (Should not be changed)

#define SAVE_DATA              1                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             1                 // [SetOption0] Save changed power state to Flash (0 = disable, 1 = enable)

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "xxx.xxx.xxx.xxx"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "xxx.xxx.xxx.xxx"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "xxx.xxx.xxx.xxx"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "SSID"                // [Ssid1] Wifi SSID
#define STA_PASS1              "wifi password"                // [Password1] Wifi password
#define STA_SSID2              ""                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)

// -- Syslog --------------------------------------
#define SYS_LOG_HOST           ""                // [LogHost] (Linux) syslog host
[... 252 lines cut ...]

/*********************************************************************************************\
 * Select KNX without Emulation to save space
\*********************************************************************************************/

//#define USE_KNX_NO_EMULATION                     // Create sonoff-knx with KNX but without Emulation (See sonoff_post.h)

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
\*********************************************************************************************/

//#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _USER_CONFIG_H_
`

Jason2866, 2018-05-24:
Just a few hints:
Use DHCP it is more elegant and handling is easier
Disable mDNS. It makes somtimes troubles 
If not using Home Assistant disable it
Do you have enabled Hue emulation in web interface. In config it is enabled but not activated
Give every device a unique friendly name1 (menu configure other) This is the name you call Alexa tu turn on  off

GasTurbineMan, 2018-05-24:
Following you on most everything except where you say (regarding Hue emulation) "In config it is enabled but not activated".  I will start reading, but how do I activate it?

Thanks for your time.

GasTurbineMan, 2018-05-24:
Also, in addition to how to "activate" Hue emulation, I do give each of my Sonoff devices a unique Friendly Name during my testing and configuration of my flashed devices.  Each device has its own name.

GasTurbineMan, 2018-05-25:
I appreciate you advice on user_config.h settings.  Now more questions specific to getting Alexa to work:

 

1.	Which emulation do you recommend:  Wemo or Hue?
2.	You commented that I have Hue emulation enabled, but not “activated”.  How to I activate Hue emulation in Tasmota?
3.	On Alexa, what “skill” needs to be added?  I would assume either Hue or Wemo, but I cannot add neither because I don’t have any of their devices which prevents me from creating an account to link the skill to.

 

Maybe if I can get an answer to question 2 regarding “how to activate” Hue emulation in Tasmota, my Alexa might see my device.

 

Thanks

GasTurbineMan, 2018-05-25:
Reading a bit more on the net, everything I find says that I do not have to do anything special on Alexa with regards to having to add a new skill.  (ie Hue or Wemo skill).  Says the emulation takes care of that need.

So I guess I just need to know what you mean when you say I need to activate the emulation on my sonoff.

Jason2866, 2018-05-25:
In your userconfig.h you have not selected an Emulation. This is my wording for activated.
#define EMULATION EMUL_NONE // [Emulation] Select Belkin WeMo (single relay/light) or Hue Bridge emulation (multi relay/light) (EMUL_NONE, EMUL_WEMO or EMUL_HUE)
This can be done later in the Web Interface. But it has to be done, with your userconfig.h
[image]
Hue Emulation is the better choice...

GasTurbineMan, 2018-05-30:
I just wanted to follow up and let you know everything is working now with Alexa and Tasmota.  Initially, our “Alexa” device was just our FireHD tablets, and while they were able to detect my “Smart Life” devices, they would not detect the Sonoffs.  Today I installed an Echo Dot and it detected all of the Tasmota Sonoffs with Hue enabled.

 

Thanks for your guidance on helping me get my user_config.h file set up correctly.

I will mark this topic as closed.

 

 <https://user-images.githubusercontent.com/24528715/40559159-19352122-6056-11e8-8411-9fb738041a4d.png>

GasTurbineMan, 2018-05-30:
This is a great support group.

closed 2018-05-30 by GasTurbineMan

fragolinux, 2018-11-11:
hi
which is the correct command to turn on or off "hue emulated" devices? If i say "alexa accendi alza" or "alexa accendi abbassa" (italian for: alexa, turn on ALZA, or alexa, turn on ABBASSA, which are the names of the 2 relays in a sonoff dual r2) she says these commands are not available in the device...