arendst/Tasmota #2815: d1 mini +1 relay + 3 reed switch

lscosta90br, 2018-05-24:
Dear Ones,

There are only 1 and 3 reed switches.

Here's my setup:

It's possible ?

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

lscosta90br, 2018-05-24:
I was able to get to this configuration:

[image]

Questions:
All reed switch are without magnet

[image]

- toggle 1 => relay ok!
- toggle 2 => reed switch: status ON - OK
- toggle 2 => reed switch: status OFF - How do I switch to ON with magnet removed?
- toggle 3 => reed switch: status OFF - How do I switch to ON with magnet removed?

Frogmore42, 2018-05-24:
You didn't really state why you need 3 reed switches, so it is hard to say how you can achieve your goal.  There are some cases where it might be possible to do this without an external system.  But, I would use some sort of home automation or NodeRED to handle the control of the relay with the switches as input.

closed 2018-07-10 by ascillato2