arendst/Tasmota #2816: Password clear text

think-free, 2018-05-24:
I've found a security issue with the wifi password.

When accessing the web page http://x.x.x.x/w0 looking for the html code you can see the wifi password

If you have a power outage, when the module restart and can't connect to the wifi (router take too long to start, ...) the module start in ap mode and is accessible by everybody at proximity. So anybody is able to see your wifi password.

I know that you can put an admin password but it's not required ... (and I don't think that's the solution)

MatzeJoerling, 2018-05-24:
https is not possible ... try WifiConfig 4 or WifiConfig 5 to prevent open tasmota hotspots ...

ascillato, 2018-05-24:
Hi,

I tried what you say but I could not find the password.

Which tasmota version are you using? Because that issue was solved at #1727

think-free, 2018-05-24:
I Will try the last version tomorrow
Saw It on the chrome dev tools on the HTML code

curzon01, 2018-05-25:
- display password in clear text on http://x.x.x.x/w0 w0 was solved since release 5.12.0
that's the reason why people are asked about `Status 0` info when opening an issue!

- module restart in AP mode can be prevent setting "Default Wifi Tool" to  `WIFI_RETRY` or `WIFI_WAIT` using _user_config.h_ define `WIFI_CONFIG_TOOL` or [command](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi)
 `WifiConfig 4` or `WifiConfig 5`

- prevent resetting the device to default (which might be enable the point above) using the button for 40 sec can be prevent by `SetOption13 ON`

ascillato, 2018-06-02:
Hi @think-free 

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-08 by ascillato2