arendst/Tasmota #2817: Wemos D1 mini With Tasmota

Ronniec10, 2018-05-24:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

Ronniec10, 2018-05-24:
Hello ,

I flashato a Wemos d1 mini with the software tasmota for sonoff (and 'right?). I managed to flash it correctly and see it properly connected to my network.
Now I want to control a relay (powered at 5V) and to do this I set a relay1 on pin D1. If I press the on / off from the tasmota interface the relay does not switch. Can you help me understand why? thank you

RaymondMouthaan, 2018-05-24:
Could be handy to provide some more info ...

MatzeJoerling, 2018-05-24:
hardly believe that wemos d1 mino can power a relay. it outputs 3,3v with 12mA max . do you have a shield? 

why dont you even provide any of the requested information?
 do you thing we are some sort of clairvoyant ...?

Ronniec10, 2018-05-24:
I have 5v on my d1.

Jason2866, 2018-05-24:
Do you have a driver transistor or ic for the relay. I have 5 volt do say nothing.
Describe your setup as exactly as you can. Only with informations someone can help

Ronniec10, 2018-05-25:
ok.
i have flashato tasmota sonoff (latest version) on wemos d1 mini using the Arduino ide:
sheet: Wemos D1 mini lite
flash: 1m no spiffs
ipwvariant: 1.4 prebuilt
cpu frequency: 80 Mhz
upload spreed: 115200
I connected my relay shield between vcc and Gnd 5v of the D1 and I connected the pin in to d5 on which I have programmed from the tasmota relay1 interface. When I press on | off it does not switch the relay.
I tried to manually close the pin on the Gnd and rele 'switch
What should I say in more?

Ronniec10, 2018-05-25:
ok
I measured between d1 (I have configured this pin) and gnd, When I press on | Off I get: on: 3.2 and off 0
If I connect my relay it does not switch.

RaymondMouthaan, 2018-05-25:
As @MatzeJoerling said, the Wemos doesn't supply enough current to power your relay, you need a driver between it like @Jason2866 said...

MYLE-01, 2018-05-25:
Had the same problem with my wemos then I put the relay on the 3v side and everything work

Ronniec10, 2018-05-25:
Can you explain better? I have already tried to power my 3 v relay but
it does not work the same

MYLE-01, 2018-05-25:
D6 relay1 D1 switch1

Ronniec10, 2018-05-25:
???

reloxx13, 2018-05-26:
yeah, what about the button? you need to setup it too on the configuration page

switch1 = relais1
switch2 = relais2 
and so on....

closed 2018-07-10 by ascillato2