arendst/Tasmota #2818: Sonoff Pow Jack Plug

jannnfe, 2018-05-24:
Hello,
I was wondering if it is possible to retrofit a Jack Plug with the Sonoff Pow, for example to connect a thermometer. Then I looked at both devices (Sonoff Pow and Sonoff TH10 / 16) a little closer and found that this should actually be possible. This is also possible with the Sonoff 4CH, as you can already read in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-4CH).

You just have to buy a Jack Plug and connect it to GPIO3 Serial In. How exactly you have to connect this I wanted to ask from you here. Can someone please tell me exactly what I need and how I have to wire the whole thing for it to work?

Some Pictures:
[image]
[image]
[image]
[image]

Thanks! :)

gtortone, 2018-05-24:
Hi,
I did it on Sonoff-4CH:
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-4CH

jannnfe, 2018-05-24:
Hey @gtortone,
Yes, I have already said in my post, but the whole thing on the Sonoff Pow looks a little different. There is no R35 pad and how do I have to wire the entire Sonoff Pow. I'm really bad in something and I hope someone can say exactly what I have to do.
Thanks :)

MatzeJoerling, 2018-05-24:
take rx/tx and use them for i2c or connect a sensor! dissable SERIAL logging in tasmota, done.
Connector J3! 
https://www.itead.cc/wiki/images/5/52/Sonoff_POW_Schematic.pdf

jannnfe, 2018-05-24:
Hello @MatzeJoerling,
So I do not need a 4.7Kohm pullup resistor between DATA pin and 3.3V as it @gtortone did? I simply use either RX or TX for data transfer right?
[image]

The jack plug is not required but is the more elegant solution. That's why I wanted to ask what exactly on which "ring" is the jack connection?
[image]
Thanks! :)

MatzeJoerling, 2018-05-24:
Hi, pullup would be nice, more reliable. with only one ds18b20 it will work without, i belive.
according to https://www.itead.cc/wiki/images/3/39/Sonoff_TH10A%2816A%29_schmatic.pdf J2 i would suggest that 
4: GND / 2: VCC and 3/1 Data for 2 inputs?
[image]

try it.

Jason2866, 2018-05-24:
The resistor is neccersary, without it will not work!

jannnfe, 2018-05-24:
Okay so i put one 4.7Kohm pullup resistor between VCC and RX right? I'm not using TX.
[This resistor](https://www.ebay.de/itm/200x-SMD-0805-Widerstand-4-7k-1-0-125W-150V-ROYAL-OHM-0805S8F4701T5E/181750949484?hash=item2a5133666c:g:2bEAAOSwpDdVXFDg) or [this resistor](https://www.ebay.de/itm/10x-Widerstand-4-7KOhm-DHT-DS18B20-Metallfilm-Resistor-4-7K-Arduino-Raspberry/252712608097?hash=item3ad6d86161:g:vCQAAOSwVtZaD3J-) or it doesn't matter?
[image]

I connect VCC, GND and RX to jack plug.
According to [this file](https://www.itead.cc/wiki/images/4/40/DS18B20.pdf) about DS18B20 Sensor, one of the four Pins of the Jack is not connected, so we have only 3 Pins. Wiring the jack plug i have to try i think :D
But [this 4 pin Jack socket](https://www.aliexpress.com/item/10pcs-2-5mm-Phone-Jack-4-contacts-4-conductors-TRRS-Audio-Stereo-Socket-Through-Hole-Right/32788968290.html?spm=a2g0s.8937460.0.0.4ae52e0euU8v08) should work right?

Thanks for all your help! I do not want to do anything wrong and do not order anything wrong, because things from China are on the way for a long time! :)

arendst, 2018-05-24:
If you continue on this path I would like to send you my best wishes in your afterlife...

As stated on the pow wiki page never connect a sensor to a pow. Your choosen sensor is my favorite as the casing is connected to Gnd/AC power on Pow.

May the force be with you (and wear a rubber underpants).

Frogmore42, 2018-05-24:
Please re-read the wiki page on the Pow and then reread it again.

Connecting sensors to the pow is at best dangerous and possibly deadly.  While it is possible to do it, Tasmota (by design) does not support it and it will allow to configure the sensor.

More than one person has destroyed a pow by accident trying to do this.

It is a best practice to separate your low voltage sensors from you high voltage things.  The pow makes it impossible to do this.  The TH10/16 have some separation and are reasonably okay, but I am no longer adding sensors to them. I now use wemos D1 Mini devices for all my sensors.  This is much safer but a little more expensive.

jannnfe, 2018-05-24:
Okay thanks for the feedback! That would have become something. This would have done the subject then. Thank you anyway for the lively help until then.

Jason2866, 2018-05-24:
Thx @arendst for your comment. I didnt realized that the device is a pow!
I use all my sensors on Wemos minis too so i do not think about ac power.
In future i will double check before writing maybe dangerous....

jannnfe, 2018-05-24:
@arendst @Frogmore42 One last question. Whats about the Sonoff Dual? Can I add a Sensor as described above on this Sonoff?

ascillato, 2018-05-25:
Please,

This is a very **_DANGEROUS_** situation. **_DO NOT TRY THIS_**.

## Please, change the title of the issue to NOT DO THIS.

# **_You WILL BE ELECTROSHOCKED!!!!_**

Only use sensors on a WEMOS, NODEMCU, SONOFF SV (Safe Voltage) and on any Sonoff **_prepared_** for a sensor.

AVOID PUT PEOPLE LIFES IN DANGER !!!!!

AlexTransit, 2018-05-25:
https://github.com/arendst/Sonoff-Tasmota/issues/1941#issuecomment-366917098

ascillato, 2018-05-25:
The devices that uses Mains AC (220v or 110v) and also uses sensors or parts that a human could touch, those devices use a optocoupler that separate the main AC (220v or 110v) from the 5v or 3.3v. This optocoupler link is a link by light. Is a device with a LED emiter and a LED Receiver. No direct connection. So the low voltage section is safe and disconnected from the mains.

Sonoff POW do not use an optocoupler. It uses a direct link between the power measurement chip (that is connected to main ac) and the ESP8266 chip.

That's why, your sensor WILL BE on mains AC and you WILL BE ELECTROCUTED.

ascillato, 2018-05-25:
@jannnfe 

If you need the temperature value, use a sonoff TH.

If you do not want to buy that, use a cheap nodemcu with the temp sensor and took the decisions on that device and send the command to your sonoff dual. you can use MQTT, KNX, etc. There are plenty of safe options.

closed 2018-05-28 by arendst

zauberer69, 2018-06-18:
[image]

Solution:
4 -> VCC
3 -> GND
1 -> Data

ascillato, 2018-06-18:
@zauberer69

That is not a solution. That is a call for death. Please DO NOT CONNECT SENSORS TO A SONOFF POW. 

VERY DANGEROUS

read all the thread before posting.

AlexTransit, 2018-06-18:
3 -> GND
GND internal connected to ACC line.
https://github.com/arendst/Sonoff-Tasmota/issues/1941

DIYglenn, 2020-09-15:
Hmm, I just did the 4ch mod, is it just as dangerous there?