arendst/Tasmota #2819: RfReceive not working correctly on Sonoff RF Bridge

Flori1989, 2018-05-24:
On my Sonoff RF Bridge I was able to learn some buttons from my RF remote. The learned data looks as following:
`21:15:14 CMD: RfKey3 5`
`21:15:14 MQT: stat/sonoff-bridge/RESULT = {"RfKey3":{"Sync":9690,"Low":320,"High":930,"Data":"401151"}}`
`21:15:22 CMD: RfKey4 5`
`21:15:22 MQT: stat/sonoff-bridge/RESULT = {"RfKey4":{"Sync":9670,"Low":320,"High":920,"Data":"401154"}}`

So far everything is working just fine. But when pressing the same two buttons on my remote I get this:
`21:15:36 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9730,"Low":310,"High":940,"Data":"40155F","RfKey":"None"}}`
`21:15:38 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9670,"Low":320,"High":930,"Data":"401154","RfKey":4}}`
`21:15:38 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:15:41 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":330,"High":920,"Data":"40155F","RfKey":"None"}}`
`21:15:45 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9710,"Low":310,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:15:47 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9690,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:15:50 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:15:53 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9680,"Low":330,"High":920,"Data":"401154","RfKey":4}}`
`21:15:56 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:16:17 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:16:20 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9690,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:16:24 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:16:26 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"401151","RfKey":3}}`
`21:16:26 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9740,"Low":310,"High":940,"Data":"40155F","RfKey":"None"}}`
`21:16:34 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`
`21:16:41 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":9700,"Low":320,"High":930,"Data":"40155F","RfKey":"None"}}`

For some reason only in some few cases the correct data is shown.

Jason2866, 2018-05-24:
Let me guess, poweroutlets.
The sonoff bridge has problems to decode this code (know as Intertechno) always correct.
No Tasmota issue because the decoding is done on a independent ic not the esp.

Flori1989, 2018-05-25:
Thank you for the quick answer @Jason2866 

I was able to solve the problem by flashing the EFM8BB1 with this firmware:
https://github.com/Portisch/RF-Bridge-EFM8BB1

closed 2018-05-25 by Flori1989

Jason2866, 2018-05-25:
I know this firmware. I just had not the time to flash it. Is it really working more reliable?
I ask because portisch tested a long time different timing setups with mixed results...

Flori1989, 2018-05-25:
It's not perfect, but at least the RfKey is now detected correctly. There are still some problems with the values in the Data field.