arendst/Tasmota #2820: Sonoff Basic pinheader holes shut

KaleunBronco, 2018-05-24:
Not sure if this is the right forum for this. It's also my first post here, so be patient.

I just received another batch of 4 Sonoff Basic directly shipped from China. To my surprise I noticed that all hole for pinheader had been soldered shut. I successfully soldered pinheaders on before and flashed a good dozen of Basics but this was new. I tried to heat up the solder in the holes and sucked out the solder but this is a right mess.

Has anyone come across this issue?

jernauH, 2018-05-24:
Same here with the two I recently bought just flow some fresh solder and use a solder sucker

cHunter789, 2018-05-25:
If you don't have solder sucker try with desoldering wire and toothpick.

BTW: it's stupid move Sonoff! People will stop buying your stuff...

curzon01, 2018-05-25:
Hint:
Before I'm opending and soldering a new Sonoff I always try to update my Sonoffs using [SonOTA](https://github.com/mirko/SonOTA). It's not working on all Sonoffs/Firmware but a trial is to be worth.

@KaleunBronco - this is an [issue list](https://en.wikipedia.org/wiki/Issue_tracking_system), not a [forum](https://groups.google.com/forum/#!forum/sonoffusers) :sunglasses:

KaleunBronco, 2018-05-26:
Thanks for the feedback guys. Never tried the OTA method. Got a solder sucker and will try to use it. Stupid move by Sonoff to do that.

reloxx13, 2018-05-26:
maybe some of the devices fail for the Quality controll and get flashed and tested manual and this are the leftovers.

put on more tin and than use the solder sucker, should be no problem.

ascillato2, 2018-06-27:
Hi,

If your issue is not solved, please reopen it. Thanks

closed 2018-06-27 by ascillato2

davidamit87, 2018-09-17:
My device was destroyed by an attempt to open the holes