arendst/Tasmota #2824: Sonoff US Model of 3-button T1

cirolaferrara, 2018-05-25:
Flashed without problem with mirko/SonOTA five Sonoff US Model of 3-button T1 but after firmware upgrade, i bricked one item.
Can you help me to repair it? No known method to put it into flash mode? Thanks

curzon01, 2018-05-25:
[Sonoff-T1 Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU) does not help?

cirolaferrara, 2018-05-25:
It says:

```
US Model of 3-button T1, received direct from Sonoff China 2018-05-12
Front photo (Google photos) Rear photo (Google photos) (Sorry about the tape smudges).

The header labels on J3 are accurate (apparently some boards have TX and RX reversed; this doesn't seem to suffer that).

Flashed with SONOTA.exe. Above method of getting into flash mode doesn't seem to work.
```

curzon01, 2018-05-25:
I read it too

cirolaferrara, 2018-05-25:
ehmm how?

curzon01, 2018-05-26:
Does your board looks same like on the photos on [US Model of 3-button T1, received direct from Sonoff China 2018-05-12](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU#us-model-of-3-button-t1-received-direct-from-sonoff-china-2018-05-12). 

Unfortunately a) I don't have a T1 US and b) the photos are to bad to find something.

Do it yourself: Search for the ESP chip and then search for the chip layout (use Google) and find GPIO 0.Thas all what you need.

If you don't find all of it, do you have detailed photos from the front and rear of your device?

cirolaferrara, 2018-05-26:
These are photos:
https://pasteboard.co/HmVXARu.jpg
https://pasteboard.co/HmVXUVx.jpg
https://pasteboard.co/HmVYD5C.jpg

curzon01, 2018-05-26:
Have a look on U4 on your board - that's the ESP.

And the ESP Pinout:
[image]

So using the dot * on upper left corner (same for your U4 chip) the GPIO0 should be here:

[image]

cirolaferrara, 2018-05-26:
Doh. I need a very good solder. Mine is very bad to do this work

curzon01, 2018-05-26:
Yeah, that's my origin question - it does not work with an wood painting solder :grinning:
May be you have an ohmmeter, follow the circuit path and find a better junction...

cirolaferrara, 2018-06-06:
Done! (I hope so). And now?

curzon01, 2018-06-07:
Connect pin GPIO 0 to GND during (only necessary during power up, not steady) power up to switch device into flash mode, then flash it