arendst/Tasmota #2825: Sonoff basic as a numb switch to respond on switch only

c3reeman, 2018-05-25:
Hi,

First of all loving the tasmota and thanks to those involved, I baught some cheap Sonoff basics as a trial, and I love them so much with Tasmota I’ve just orderd 40 more and a few Pow R2’s and finally some Sonoff temp sensors, I’m building a junction box full Of tasmota’s as a lighting wiring hub for a barn conversion I’m currently on with.  

Anyways my question I’m linking a Sonoff basic into some existing motion sensors The motion sensors are driving a 230v (uk) relay that are then linked into the Sonoff basic for the 5v switching and it’s working much the same as a basic light switch so running on gpio14 09 switch 1, and it’s working great motion sensed turning Sonoff on after physical motion sensor timer runs 

and using 
Sorry for what might be an obvious question

c3reeman, 2018-05-25:
One solution I did think of was to use the pulse timer function and set it at about 10 minutes ( the outside motion sensors are set at 8 minutes ) this way if it did some how get inverted it would always reset it’s self after 10 minutes ? Appreciate it if anyone has a better simpler option

Dickesplumpel, 2018-05-26:
I am not really shure what you want to know but it looks like your switch is set to TOGGEL (default) that sometimes can flip around so that on and off is changed, set it in the console to FOLLOW (switchmode1 1) or FOLLOW inverse (switchmode1 2) - example is for switch1. See tasmota commands for more details: 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

c3reeman, 2018-05-28:
Ah that’s the one, thankyou ... I had a play with switch modes before posting and didn’t think it worked as I needed but after you suggested I’ve just had another play and noted It actually works perfectly

closed 2018-05-28 by c3reeman