arendst/Tasmota #2826: Flash pin locations of sonoff touch t1 R2

roccotocco, 2018-05-25:
I bought some sonoff touch t1 R2 2 gang, and im ready to upload the firmware.
But i couldn't find a picture of the RX, TX GND pins of the touch t1 R2 version, i only found the method for uploading for the old version of the sonoff touch.
Does anyone have any links or pictures, so i can connect it to the right pins and upload the firmware?
Attached 2 pictures of my sonoff touch t1 R2 2 gang

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : 
- [x] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**
[image]
[image]

timosntr, 2018-05-25:
Maybe I'm misunderstanding your question, but it is unclear to you to what pins you should connect to?
Have you tried connecting to the labeled pins? As I can see from the board it won't matter which GND and VCC you use.

dcazarez05, 2018-05-26:
I have the same problem,  maybe somebody can clarify this to us.

crockernr, 2018-05-27:
i too have this versi 3 channel)

wongnam, 2018-05-28:
Hi
Look at the info from this link, special at the bottom of the page i found people have flashed The T1 R2 US TOUCH BOARD V1.0 by enters flash mode by touching pad P3 while pressing switch S1. 
https://github.com/arendst/Sonoff-Tasmota/issues/1424
Below is screen shot FYR

[image]

crockernr, 2018-06-04:
I too cannot get this to flash.
Does anyone have any step by step info on getting
this to work?
Some users have stated they have got it flashed with sonota.exe but I cannot get it to display an SSID to connect to.

crockernr, 2018-06-06:
FYI - I have now managed to get this to flash with SONOTA.exe
To do this I pressed and held Button 1 (Top Switch) until this beeped for a second time (there will be a pause inbetween) Once this is done the wifi light should start flashing on and off.
Once this was done I repeated the above steps but this time holding Button 3 (Bottom Switch). Once this had completed it created its own SSID and was able to flash ! now have MQTT working ! Happy days

HellfireZA, 2018-06-07:
@dcazarez05 how did you get the network to show up the to flash with SONOTA.exe?

dcazarez05, 2018-06-08:
@HellfireZA You need to put on flash mode twice, after push and hold for 5 seconds and "beep" take off and push again for 5 seconds, after the second "beep" you will see the network.

dcazarez05, 2018-06-15:
@HellfireZA PASS is 12345678

HellfireZA, 2018-06-16:
@dcazarez05 Thank you so much, the problem I was actually having was that the ITEAD network would only appear for 5 seconds or so. So I followed the method of holding down B1 and then B3, and it worked. Got it flashed :)

luketoh, 2018-06-22:
Hi All!  I need help!
I bought the T1 US R2 with single switch.  I can get it to broadcast its SSID.  But when I run Sonota.exe it will go to stage 2 but then cause the T1 to lose connection and it never connects to the "server" which is my PC.  I am on 1.8 firmware for the T1.  Can someone do a video or give me steps to flash this?

HellfireZA, 2018-06-22:
@luketoh my one also did that at first. To get it to show permanently you need to figure out the button combo on your unit.

luketoh, 2018-06-22:
@HellfireZA What I did was press and hold the light switch for 7 secs..release, and press and hold for another 7 secs.  Then the light will blink continuously...which is when it broadcasts the ITEAD SSID.  Then I would load Sonota and enter my PC's IP, router's SSID and password.  Then connect to the ITEAD SSID...and then after this stage it will ask me to connect back to my own SSID. But the T1 device will go into 2 blink mode and not connect to the "server".  

It seems to be due to this issue https://github.com/mirko/SonOTA/issues/58 maybe.

May I ask what version of the firmware was on your T1 US R2 device at the time you flashed it with Sonota?

HellfireZA, 2018-06-22:
@luketoh You're doing it right.

So you choose your PCs Device IP which is most likely (0), you put in the wifi name + password.
Then it's saying please connect to the ITEAD SSID, you do so. It'll do it's thing very briefly and disconnect from the ITEAD SSID, then it'll come up with the message about "Wait for the FinalStage..." network. At that point you connect to the Wifi network you put the details in at the beginning. Now it actually starts to communicate with the T1 and soon you'll see FinalStage, give it up to 2 minutes to be sure. Let me know.

luketoh, 2018-06-23:
@HellfireZA There's a problem...

Everything is working up to the part where my PC disconnects from ITEAD and back to my own router.  Thereafter the T1 device will not connect to my network and my PC is left waiting for it to connect to the "server".   The T1 will flash twice and pause, flash twice and pause (no longer the continuous flashing).

luketoh, 2018-06-25:
I didn't manage to flash using Sonota.  But I did flash it with the FTDI method after several tries.  The hard part was knowing when the device was in flash mode.  I still don't know how I did it...but there were no blinking leds or anything and the device was able to be flashed.

zanattabruno, 2018-06-30:
I did flash it with the FTDI method. To put then in flash mode press 2 times quickly the internal switch S1

rport, 2018-07-01:
Here is a photo of the T1 R2 US Touch Board V1.0 (Front and Back) - I've also published it via Google Drive if you want to zoom in or comment! http://bit.ly/2yY2wBw
[image]

ascillato2, 2018-07-21:
Can you add this information to the wiki?

Thanks.

closed 2018-07-21 by ascillato2

macgeeknz, 2018-12-14:
FWIW, I couldn't get my 3-gang US touch into programming mode no matter how many times I tried the various button-pressing combinations. Nor could I get it to accept OTA updates. So I ended up figuring out where to access GPIO0 and put the unit into flash mode that way instead.

I used a 1.5K resistor tacked on to the left side of R20. This pad is connected to GPIO0 on the ESP8255 via a 400 ohm resistor (R20). It is also connected to U1, hence why I used a resistor instead of a wire link (a wire link would probably work OK but I wanted to play it safe). I bridged held the other side of the resistor onto a nearby ground pad while powering on and it then flashed fine.

I've attached a photo showing where to make the connection.

[image]

wongnam, 2018-12-15:
**Yes, used jumping-wire method will work for my Touch T1-US 3Gang, it will go to flash mode 100% success.**

FYR,
1- jump the wire as attached picture
2- still hold the wire and plug the FTDI USB to the computer
3- wait for 1-2 seconds then release the jumping-wire
4- ready process for flashing my Sonoff Touch T1-US 3C

[image]

sundeepgoel72, 2020-03-20:
What is the U2 touch controller used - need to troubleshoot a module

pablacho, 2020-04-03:
> What is the U2 touch controller used - need to troubleshoot a module

try this page https://github.com/arendst/Tasmota/issues/5449

I am trying to find out what is U1, did you find anything on that?