arendst/Tasmota #2827: Saving Timer settings

Lawrencezarb, 2018-05-25:
Can someone confirm if the settings setup on the Timers are actually saved with the backup configuration.

I had to restore a configuration to my Sonoff 4CH, but the timer setting did not restore.

ascillato, 2018-05-25:
Yes, they are saved.

Lawrencezarb, 2018-05-25:
Any idea why the restored file would not restore them?

ascillato, 2018-05-25:
Hi,

I just backup the configuration of a sonoff
Then I erase all flash
Then upload the last tasmota firmware
Then I restore last config

And everything was there including my timers.

So, I can not reproduce your issue. Sorry.

Please Try Again

reloxx13, 2018-05-26:
savedata was off and it did not get saved?

arendst, 2018-05-27:
What is the name of the file you used to restore your 4CH?

Lawrencezarb, 2018-05-27:
Config_Main____LED____Co2___Pump_5.14.0a.dmp

arendst, 2018-05-27:
What is the size of the file?

Lawrencezarb, 2018-05-27:
2560 (3k)

Lawrencezarb, 2018-05-27:
I also have changed the number of MAX_TIMERS to 32. Not sure if that is relevent.

arendst, 2018-05-27:
There you have it. This upsets the Settings area with the result you experience.

Lawrencezarb, 2018-05-27:
Ok, so then to clarify. I change the MAX_TIMERS to 32. Upload the BIN file. Set the timings on the wegui, then perform a backup. Then the backup will not save the timer settings. 

Is my understanding correct?

If this is the case what can I do to backup the timer settings.?

arendst, 2018-05-27:
It should save the timer settings as expected. The problem is your current file-length; to me it seems it does not have the extra timers in it and therefore the restore failed.
With your extra timers the settings area should be larger than 2560 and so should your hbackup file.

In short.

-Recompile with your extra timers.
-Upload and then make a backup and verify the file lenght to be more than 2560
-Restore the new back up file.

Lawrencezarb, 2018-05-27:
I have done as instructed and my backup is now 2624

ascillato, 2018-06-02:
Hi @Lawrencezarb 

If your issue was solved, please close it. Thanks :+1:

Lawrencezarb, 2018-06-02:
No, its still the same!

ascillato2, 2018-06-27:
Hi

I really don't know what is going on with your modified Tasmota. I can not reproduce your issue with the original firmware.

Just to know, Have you managed to solve your issue?

I'm closing this issue due it has no activity. Thanks.

closed 2018-06-27 by ascillato2