arendst/Tasmota #2828: Time zone table

pmknowles, 2018-05-25:
The TimeSTD and TimeDST has been driving me mad. I live in the UK and expect to be in TimeZone 0. However, TimeZone 0 seems to default to GMT/UTC+60 and GMT/UTC+120 for STD and DST respectively. A table showing what result each TimeZone will give will save many people from having to set TimeDST/TimeSTD and TimeZone to 99. It might then be possible to set the TimeZone via the webserver instead of via console.

arendst, 2018-05-25:
See wiki commands

In short, timezone -13 to 13 set a fixed timezone without daylightsaving. Timezone 99 uses the offsets entered with TimeDst and TimeStd at the configered time periods.

ascillato, 2018-05-25:
Hi,

I tested `timezone 0` and works as expected. It gives UTC +0 h

If you are in the UK at this moment you are in UTC +1 h

See extracted from https://24timezones.com/world_directory/time_in_london.php

[image]

So, for UK, your configuration, to be typed on the console, should be:

```
TimeZone 99
TimeDST 0,0,3,1,1,+1
TimeSTD 0,0,10,1,2,0
```

So, if your country has DST, you will have to use timezone 99 and config the STD and DST times.

There are some countries that have the same Timezone for all the year, so, for those, you use just the timezone command alone.

All the information regarding TimeZone, DST and STD is on the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management

> It might then be possible to set the TimeZone via the webserver instead of via console.

Yes, would be nicer, of course, but an extra webpage requires more FLASH and RAM memories that are very restricted resources in the ESP8266. So, as the Sonoff-Tasmota is an alternative Firmware for Sonoff Devices to be controlled by MQTT, a console for the one-time configuration should be enough. Remember that the original sonoff firmware does not have a webpage. All the configuration is being made by an app on your phone.

At this moment, to configure Tasmota, there are some external softwares as:

* TasmoADMIN (https://github.com/reloxx13/tasmoAdmin)
* TasmoADMIN Add-on for HomeAssistant (https://github.com/hassio-addons/addon-sonweb)
* Yeti Phone App (https://github.com/arendst/Sonoff-Tasmota/wiki/Yeti)
* ESP8266 smart config android app. (https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch)

Those allows you to configure some Tasmota features using an UI (User Interface). I believe that more options will be added to those softwares, making the Tasmota Webpage used only for advance configurations/debug/etc.

Hope this helps.

If this solves your issue, please close it. Thanks.

ascillato, 2018-05-25:
Corrected commands. See updated comment above.

pmknowles, 2018-05-25:
Thank you it explains why the time wasn’t correct. It’s not clear that the timezones other than 99 don’t adjust for DST.
Decided to back up a ‘standard’ configuration and upload it each time I get a new device.

closed 2018-05-28 by arendst

ripvega, 2019-03-05:
Thank you for taking the time to do this. It really helps and I appreciate it.

mrdvt92, 2020-03-21:
I think this is EST5EDT

TimeZone 99
TimeDST 0,2,3,1,2,-240
TimeSTD 0,1,11,1,2,-300

from Wikipedia 

TZ='EST5EDT,M3.2.0/02:00,M11.1.0/02:00'

nikolicjakov, 2020-08-23:
Hi everyone,

i not quite sure what command should i execute to set my timezone to Europe/Berlin. Could anyone share TimeDST and TimeSTD command to set correct time offset for Berlin Europe?

ascillato, 2020-08-23:
@nikolicjakov Jakov,

If you follow the procedure on [this previous comment of this issue](https://github.com/arendst/Tasmota/issues/2828#issuecomment-392147269), you can look for DST and STD at https://24timezones.com/Berlin/time. From there, you can take the needed data. Then from the [Tasmota Docs](https://tasmota.github.io/docs/Commands/#management), you can read how to input that data into `timezone` commands.

* Commands to be input in the Tasmota's console for Berlin DST and STD:

```lua
TimeZone 99
TimeDST 0,0,3,1,2,120
TimeSTD 0,0,10,1,3,60
```

ascillato, 2022-04-15:
> A table showing what result each TimeZone will give will save many people from having to set TimeDST/TimeSTD and TimeZone to 99.

From PR to Docs: https://github.com/tasmota/docs/pull/941

Table of Timezones w/STD and DST: https://gist.github.com/cpfair/253d4fb5bab9b6d58e0a204a78003118

pmknowles, 2022-04-15:
Adrian, are the Belfast and Dublin ones correct? As far as I know the times are the same in NI and Eire. In fact, the London one looks wrong as we 'spring forward' but it could be that the 'base' NTP server is set to 'European time'
People have been struggling with the time settings for years. Perhaps it's time for a different approach?
The NTP server should default to UTC then, perhaps, it's worth considering setting a 'base' timezone  - UK would be 0 and a DST string 1 (to enable),DST month,DST week, DST day, DST hour, STD month, STD week, STD day, STD hour, DST offset so it would be one command
For the UK
Timezone 0,1,3,0,1,1,10,0,1,2,60
I'm not sure if hemisphere would be needed but it depends on how the system would cope with the DST month being higher than the STD month. You also might not need to have the DST 'flag' as the second parameter and just go straight to DST month.

What do you think?

Regards

Phil K

From PR to Docs: tasmota/docs#941<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Ftasmota%2Fdocs%2Fpull%2F941&data=04%7C01%7C%7C7050ade87649472d5d0c08da1eaf2aaf%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637856034929399764%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=qAgPfPXLs5Gj%2FkLMk3Tsv8R%2FWGGHB6WPuJHgwiVMWPw%3D&reserved=0>

ascillato, 2022-04-15:
> are the Belfast and Dublin ones correct?

I have reported to the author of the automatic table that some of the cities are not correct. Please check the following: https://github.com/tasmota/docs/pull/968#issuecomment-1099927345

Thanks.

cpfair, 2022-04-15:
I believe I have corrected the "spring backwards/fall forwards" issue in the linked timezone table.

mrdvt92, 2022-04-15:
> I believe I have corrected the "spring backward/fall forward" issue in the linked timezone table.

New York is now correct! Thanks

Also, I spot checked this entry.

```
Pacific/Kiritimati | This timezone has a UTC offset outside the range Tasmota supports.
```
The table states that it is not supported. But based on the information from http://www.timezoneconverter.com/cgi-bin/zoneinfo?tz=Pacific/Kiritimati and https://timezonedb.com/time-zones/Pacific/Kiritimati

It appears that they changed the clocks a few times in the past and are now fixed at UTC + 14:00.

Is there a way to catch that programmatically and return as "Timezone +14:00".

cpfair, 2022-04-15:
According to the documentation, the `Timezone` command only accepts offsets in the range [-13:00,13:00]. A quick glance at [the code](https://github.com/arendst/Tasmota/blob/b8cf5eee7cc012fdecd048f2c263b8eb9aae1d22/tasmota/support_command.ino#L2011) makes me think the supported range is actually wider, but I based the table on what was documented.