arendst/Tasmota #2830: Sonoff Touch Flash error:

shayshtern, 2018-05-25:
any way I get to this message:
warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
I connect it ok!
why?!

Frogmore42, 2018-05-26:
There are lots of other issues in the list with similar sounding problems.  There is some information in the wiki on how to flash.  There are you tube videos on it too.  Please review those first to see if they solve your problem.

There are many things that can cause the issue you are seeing.  None of them are actually Tasmota issues.  They are usually caused by user error or bad power or USB serial converters.

shayshtern, 2018-05-26:
Is it possible to flash it OTA? SonOTA Works with Sonoff Touch? Any experience?

reloxx13, 2018-05-26:
check sonota wiki for supported devices.

appde, 2018-06-20:
@shayshtern Reducing upload_speed in platformio.ini file fixes most of the issues:
```
upload_speed = 115200
;upload_speed = 512000
```

closed 2018-07-10 by ascillato2