arendst/Tasmota #2833: WPA2-PSK authentication bug

Phil795, 2018-05-26:
I flashed my S20 socket and have problems with mixed WPA/WPA2 PSK (TKIP, CCMP). TKIP or CCMP are set to automatic. Is it possible that ESP cannot negotiate this automatically?

If I now switch the encryption to WPA-PSK CCMP, the device connects and the connection works. Now I can switch the encryption back to mixed WPA/WPA2 PSK (TKIP, CCMP) and the device remains connected.
Only the first connection negotiation does not work with the mixed mode.

My hardware:
- Linksys WRT3200AC (OpenWRTT last stable build)
  Wifi 2.4GHz WPA/WPA2-PSK, auto (TKIP, CCMP or TKIP and CCMP)
- Sonoff S20 (ESP8266 Core v2.3.0)

Jason2866, 2018-05-26:
No problems here with a Lancom and a Bintec accesspoint with mixed or fixed mode setup.
I use core 2.3.0 too.

reloxx13, 2018-05-26:
check your wifi channel and try a change under channel 11.

also be sure you dont have 2,4GHz and 5GHz Wifi with the same name.

Jason2866, 2018-05-26:
@reloxx13 i have 2.4  and 5 Ghz with the same name. This is no problem at all...
Esp supports only 2.4!

ascillato, 2018-06-08:
Hi,

Have you solve your issue?

closed 2018-06-13 by ascillato2