arendst/Tasmota #2835: Issue flashing Sonoff T1 1CH and T1 2CH

tuchocaldera, 2018-05-26:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**
I have been flashing Sonoff T1 1CH and T1 2CH and have experienced the following: After successfully flashing Sonoff T1 1CH I proceeded to configure it as "T1" but I realized that physically the button didn't work (!?). I decided then to configure it as "Touch" and it worked normally there. Then I flashed the T1 2CH and configured it as "T1 2CH". Again I could verify that one of the buttons physically didn't work and I could not recover it. Through mit netværk and my phone the 2 channels work perfectly. I don't know what is happening. Is there anyone who has experienced something similar?. Is there a solution to the problem?
I am using firmware Sonoff-Tasmota Arduino IDE

Leseratte10, 2018-05-26:
Are you sure these are legitimate Sonoff devices and no cheap chinese knockoffs? Try setting the type to Generic and try all the GPIOs until you find the one that works.

tuchocaldera, 2018-05-26:
They are legitimate Sonoff, quite sure.

Frogmore42, 2018-05-26:
You didn't provide status 0, so no one knows how you really configured it.  You, most likely, either have a hardware issue or a configuration issue.

Did the hardware work before you flashed it with Tasmota?

tuchocaldera, 2018-05-26:
Yes, the hardware worked good. I have flashed many Sonoff: Basic, RF, Dual
and I have not had problem at all with my configuration.

Frogmore42, 2018-05-27:
Well, you are having a problem now.  It could be hardware, but you said it was not that.  It could be software, but no one else is having problems with it.  Or, it could be configuration, but you have said there is no problem with yours.  That makes it a mystery.  If you are satisfied with that answer, you need do nothing more.  If you want more help, you will probably need to provide all the things requested in the issue template as that is the only way anyone can help other than guessing.  In particular, you need to provide the log for the time showing the problem and also status 0.

wongnam, 2018-05-27:
@tuchocaldera Sound like you are talking about the new model Sonoff T1 US: 1-3 Gang US AU Standard WiFi RF Smart Wall Touch Light Switch. This is a new model that it is not T1 UK nor is it Touch US.

tuchocaldera, 2018-05-27:
I see that you are somewhat annoyed by my answer to your questions. I want
you to understand that my intention is to know if other people have had the
problem that I have. Apparently you already know it when you write: "no one
else is having problems with it". Well, what I've done so far is follow
what others have done and the information I've found in Arendst /
Sonoff-Tasmota / Wiki. I'm going to review everything again.
Thanks for your comment

Frogmore42, 2018-05-27:
I am not really annoyed.  But, this is not supposed to be a forum.  This is supposed to be a place for reporting issues with Tasmota.  The set of questions that are in the template are there to make sure people do at least a little homework and see if they can solve the issue themselves.

The wiki has this:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU

Which has a link to this:
https://github.com/arendst/Sonoff-Tasmota/issues/1424

That issue has some stuff in it that is probably relevant.

I believe your issue is a duplicate of 1424.

I don't have a Touch or T1 devices, but have seen (and made) many of the mistakes that people usually make when starting out.  Since it was frustrating for me when I started out, but I now have a better understanding, I spend time helping other people.  The reason I asked about status 0 is that shows exactly what is configured, and is sometimes helpful.  I am sorry if it came across as angry.  But, look at 1424 I think it will help.

tuchocaldera, 2018-05-27:
Thank you very much for your words. I think you are right about #1424 and
I'm going to try there.

closed 2018-06-08 by tuchocaldera