arendst/Tasmota #2837: Crash right after TLS fingerprint verification

younesmaia, 2018-05-26:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Sonoff Basic
- [X] Provide the output of command ``status 0`` :

00:00:11 RSL: 58264D/status/STATUS = {"Status":{"Module":1,"FriendlyName":["xxxxxx"],"Topic":"58264D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":3,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:11 RSL: 58264D/status/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"xxxxs","OtaUrl":"http://10.10.10.200/firmware.bin","RestartReason":"Exception","Uptime":"0T00:00:10","StartupUTC":"","Sleep":0,"BootCount":47,"SaveCount":50,"SaveAddress":"FA000"}}
00:00:11 RSL: 58264D/status/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-26T13:18:13","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:11 RSL: 58264D/status/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ABnet",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:00:11 RSL: 58264D/status/STATUS4 = {"StatusMEM":{"ProgramSize":447,"Free":556,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:00:11 RSL: 58264D/status/STATUS5 = {"StatusNET":{"Hostname":"58264D-1613","IPAddress":"192.168.15.101","Gateway":"192.168.15.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.15.1","Mac":"5C:CF:7F:58:26:4D","Webserver":2,"WifiConfig":1}}
00:00:11 RSL: 58264D/status/STATUS6 = {"StatusMQT":{"MqttHost":"mytlsmqtthost.local","MqttPort":8883,"MqttClientMask":"prjk-%06X","MqttClient":"prjk-58264D","MqttUser":"device001","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:11 RSL: 58264D/status/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:11 1970","Local":"Thu Jan 01 00:00:11 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99}}
13:37:02 RSL: 58264D/status/STATUS10 = {"StatusSNS":{"Time":"2018-05-26T13:37:02"}}
13:37:02 RSL: 58264D/status/STATUS11 = {"StatusSTS":{"Time":"2018-05-26T13:37:02","Uptime":"0T00:00:07","Vcc":3.085,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ABnet","RSSI":100,"APMac":"00:24:36:AC:00:F3"}}}
**(Please, remember to close the issue when the problem has been addressed)**

PROBLEM:
The device crashes as soon as TLS fingerprint verification 1 finishes.
I've already disabled most of the sensors, web server, etc.
I'm using a Mosquitto MQTT server with self-signed TLS cert (2048 key size, SHA256). 
Other devices (not Sonoffs) connect to this server normally.

Below what is shown by the serial terminal right before the crash. 
If I try to change fingerprint 1 (mqttfingerprint command), Tasmota crashes as well.

00:00:05 WIF: Connected
00:00:07 MQT: Connecting
00:00:07 MQT: Verifying TLS fingerprint
00:00:07 MQT: TLS figerprint 1 verified

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont
sp: 3fff32b0 end: 3fff3bb0 offset: 01a0

>>>stack>>>
3fff3450:  3fffafc4 3fffaf24 3fffaf6c 402546f3
3fff3460:  3fffafc4 3fffaf24 3fffaf24 402555de
3fff3470:  00000002 00000001 3fffaf24 40254758
3fff3480:  00000100 00000100 3fffb48c 3fffaf94
3fff3490:  3fffb2a8 00000100 3fffaf24 4025667c
3fff34a0:  3fffb48c 3fff46fc 00000003 4025a32c
3fff34b0:  3fff3884 00000011 00000000 00000000
3fff34c0:  00000100 3fff926b 3fff3884 4025a82a
3fff34d0:  3fff46fc 3fffb48c 3fff3884 4025a753
3fff34e0:  3fffb27c 00000011 3fff3884 4025a554
3fff34f0:  00000020 3fff469c 3fff926b 00000000
3fff3500:  00000004 3fffb27c 3fff926b 40253528
3fff3510:  000002c0 00000000 8ce5c831 00bb89e3
3fff3520:  4e81f63b afc61d02 9279b19f 677283b0
3fff3530:  4fedbda0 febcc41d 69aee4b9 1d0ff523
3fff3540:  c563dfc1 309a26a9 3aa39ff5 79b890cc
3fff3550:  a8be10ba bfd799c3 20f75ba6 639034db
3fff3560:  00010302 00008001 00000000 00000000
3fff3570:  00000000 68150000 389c27f8 b3eae0e4
3fff3580:  4cf3a406 765a8c74 4ab09347 582d40bd
3fff3590:  e4ec8c72 dac40c04 61da9cce 6dfc4363
3fff35a0:  7d646ec8 b5e57968 6ba57ece 6e3b2d32
3fff35b0:  07a417cb 78ab0ca7 f71c6404 5fd07071
3fff35c0:  dd2d4919 5117935b 3dceaded 8beadd08
3fff35d0:  f4c744e1 76726573 66207265 73696e69
3fff35e0:  5b646568 4c707c4f ffaa7800 a55cc575
3fff35f0:  1baa0250 81ed4629 d3d66757 6dfb4976
3fff3600:  3defda5f bb5896b7 eb062b7d 912b5b2a
3fff3610:  2fc58310 7730d2f0 59b7a05b b01ca925
3fff3620:  9227dfbf 80000000 00000000 00000000
3fff3630:  00000000 00000000 00000000 00000000
3fff3640:  00000000 00000000 00000000 000002a0
3fff3650:  ece44696 8e58f7ab 9720fa89 b9f1df67
3fff3660:  38fe5028 2e41f512 73e3becf 71fca510
3fff3670:  854b6709 fb7692c9 cdb8bf33 797570dc
3fff3680:  871185c3 c7f29103 cce5110b f16babb0
3fff3690:  72fa942a 0075561c 261b7c45 387f9d07
3fff36a0:  98f27a71 7ca97a89 0464ab94 c627cc31
3fff36b0:  8d406d32 0c243f3b 552b1509 177023d4
3fff36c0:  be45a305 3e36aa66 5a161ac3 12f92531
3fff36d0:  cf2e5e77 dc71ddc6 f2766c19 3911345b
3fff36e0:  00000008 00000000 00000020 40100836
3fff36f0:  3ffeb498 401004e1 00000002 cbc864d9
3fff3700:  00000005 00000008 40100396 00000002
3fff3710:  00000002 40100197 00000002 3ffef5e0
3fff3720:  00000001 00000000 00000020 40100836
3fff3730:  40103b00 3ffeec28 3ffef5e0 00000001
3fff3740:  00000001 40102d0f 3ffef658 00000008
3fff3750:  40103172 3ffeec28 3fff51b4 00000001
3fff3760:  4021f746 3ffeec28 3fff51b4 3ffeec28
3fff3770:  40102c56 00000037 7fffffff 00000002
3fff3780:  00000137 40102a85 00040000 3fff5097
3fff3790:  40101bb7 00080000 3fff04e0 40100836
3fff37a0:  00000000 00000000 00002200 4000050c
3fff37b0:  3fffc278 401018b4 3fffc200 00000022
3fff37c0:  3fff37d0 3ffeef20 9fb76b55 43b94ae9
3fff37d0:  40255d26 00000030 0000000b ffffffff
3fff37e0:  1780503a 6160cd10 b5d7b797 ebd523cb
3fff37f0:  6bc73aa6 00000000 00000000 00000000
3fff3800:  00000000 00000000 00000000 00000000
3fff3810:  00000000 00000000 00000000 00000000
3fff3820:  00000000 00000000 00000000 00000000
3fff3830:  00000000 00000000 00000000 51580000
3fff3840:  31c8e58c e389bb00 3bf6814e 021dc6af
3fff3850:  9fb17992 b0837267 a0bded4f 1dc4bcfe
3fff3860:  28db5c4a bdc902bd 40f4c9b7 9033a0cf
3fff3870:  0dd6b74e 6bd37b69 71e8bb82 acec41fe
3fff3880:  00000433 000002cd 0000031f c34f46f6
3fff3890:  3fff46d4 3fffae64 55c800ca 0e993342
3fff38a0:  584d6ce4 5367264a d7ed19f4 c1c5e368
3fff38b0:  839d64c2 0183ebe8 63ff5a71 3fff8e99
3fff38c0:  3fff46d4 00000433 000003d2 40251bd6
3fff38d0:  000007fe 3fff77ac 00000805 00820011
3fff38e0:  7a2a6322 d8a5d360 78b2fd70 127081c8
3fff38f0:  8034729d c0caf6ea 01c10e41 00000805
3fff3900:  00000805 3fff8e99 3fff77ac 40252e0d
3fff3910:  3f18bb7f 9682f13f 712bee64 66183400
3fff3920:  39ea7b52 fce984f3 c816d7f4 1fd8200c
3fff3930:  5fbb7eca 3fff4aa4 3fff4a64 00000780
3fff3940:  00000805 3fff9680 0000001d 3fff4aa4
3fff3950:  3fff77ac 3fff8e99 00000805 00000805
3fff3960:  00000805 3fff8e99 3fff77ac 40252884
3fff3970:  00000000 3fff8e99 3fff77ac 40252c2c
3fff3980:  00000014 3fff21e9 00000015 402074fc
3fff3990:  3fff39e0 0000000b 00000010 00000000
3fff39a0:  3fff459c 3fff21e9 3fff77ac 01000000
3fff39b0:  3fff39e0 3fff1f28 3fff77ac 402529dc
3fff39c0:  40201edc 20393320 3fff2b90 00001387
3fff39d0:  0000212a 3fff1f28 3fff876c 40202599
3fff39e0:  000022b3 3fff8784 3fff1f28 4020194e
3fff39f0:  84e94334 20413320 36203136 3fff3a90
3fff3a00:  000022b3 3fff1f28 3fff21e9 402028e5
3fff3a10:  3ffe9a28 84e94334 3ffe9a28 84e94334
3fff3a20:  00000001 3fff148c 3fff108c 40203d69
3fff3a30:  000003e8 3fff2200 00000000 4020421d
3fff3a40:  3fff222d 3fff224e 00000000 3fff1e02
3fff3a50:  00000001 00770bb4 23002200 0000000c
3fff3a60:  3fff5100 3fff3af4 3fff108c 3fff222d
3fff3a70:  3fff224e 3fff2200 3fff108c 40217beb
3fff3a80:  00000001 3fff1864 00000013 ffffffff
3fff3a90:  726f6f64 676e616d 38352f72 44343632
3fff3aa0:  6c65742f 74656d65 4c2f7972 ff005457
3fff3ab0:  ffffffff 3fffc6fc 00000001 00000000
3fff3ac0:  00000000 3fffdad0 3fff2b7c 00000030
3fff3ad0:  4022507a 3fff2a10 3fff14da 4010731a
3fff3ae0:  402013f3 3fff3b10 00000001 40216aec
3fff3af0:  650fa8c0 40213a00 00000000 402181d8
3fff3b00:  402181c0 00000000 00000000 00000000
3fff3b10:  40225ed3 3fff1f98 3fff1f49 0000006e
3fff3b20:  3ffe84d8 3fff1f98 3fff1f49 40217c34
3fff3b30:  3fff1f64 3fff1f98 3fff1f49 40218154
3fff3b40:  3fff1de1 3fff1f78 00000000 3fff2b7c
3fff3b50:  00000001 0000001c 00000000 4020b634
3fff3b60:  3fffdad0 3fff2b7c 40206038 402160a4
3fff3b70:  00000000 3fff1f0c 3fff2b75 3fff2b7c
3fff3b80:  3fffdad0 00000000 3fff2b75 402181a0
3fff3b90:  3fffdad0 00000000 3fff2b75 40206084
3fff3ba0:  feefeffe feefeffe 3fff2b90 402077a0
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ascillato, 2018-05-27:
Hi,

`Exception (29)` sometimes is due to insufficient memory.

Try to disable more sensors and features and recompile

younesmaia, 2018-05-28:
@ascillato 
My build is already very lean. Almost everything (that takes RAM) has already been disabled.
I've got no sensors (they're all undefined at "user_config_override). 
Obviously, the web server has been disabled in order for TLS to be activated. 
Only timers and NTP are enabled.

arendst, 2018-05-28:
See #2845 regarding TLS status and future expectations...

closed 2018-05-28 by arendst

majherek, 2018-05-28:
@younesmaia,

I disable most of sensors, use esplib 2.4.1 with LWiP v2 Lower Memory and have no error on 5.14.0b.

fr34kyn01535, 2018-08-04:
Using the Sonoff S20, LWiP Lower Memory and everything beside TLS disabled i still get the memory crash shown above. If it can't run with TLS the whole project is pretty much useless to those devices.

`#undef USE_DISCOVERY                          // Enable mDNS for the following services (+8k code, +0.3k mem)
#undef USE_WEBSERVER                          // Enable web server and wifi manager (+66k code, +8k mem)
#undef USE_TIMERS                             // Add support for up to 16 timers (+2k2 code)
#undef USE_TIMERS_WEB                         // Add timer webpage support (+4k5 code)
#undef USE_SUNRISE                            // Add support for Sunrise and sunset tools (+16k)
#undef USE_RULES                              // Add support for rules (+4k4 code)
#undef USE_DOMOTICZ                           // Enable Domoticz (+6k code, +0.3k mem)
#undef USE_I2C  
#undef USE_MHZ19                                // Add support for MH-Z19 CO2 sensor (+2k code)
#undef USE_SENSEAIR                             // Add support for SenseAir K30, K70 and S8 CO2 sensor (+2k3 code)
#undef USE_PMS5003                              // Add support for PMS5003 and PMS7003 particle concentration sensor (+1k3 code)
#undef USE_NOVA_SDS                             // Add support for SDS011 and SDS021 particle concentration sensor (+0k7 code)
#undef USE_PZEM004T                             // Add support for PZEM004T Energy monitor (+2k code)
#undef USE_SERIAL_BRIDGE                        // Add support for software Serial Bridge (+0k8 code)
#undef USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+4k code, 0k3 mem, 48 iram)
#undef USE_IR_RECEIVE                         // Support for IR receiver (+5k5 code, 264 iram)
#undef USE_WS2812                               // WS2812 Led string using library NeoPixelBus (+5k code, +1k mem, 232 iram) - Disable by //
#undef USE_ARILUX_RF                            // Add support for Arilux RF remote controller (+0k8 code, 252 iram (non 2.3.0))
#undef USE_SR04                                 // Add support for HC-SR04 ultrasonic devices (+1k code)
#undef USE_HOME_ASSISTANT                       // Enable Home Assistant Discovery Support (+2k code)
`

theminor, 2018-08-26:
I have this same issue. Exception (29) right after Verify TLS fingerprint...  It used to work in older versions, but now I can't get it to work even if I use an older version of Tasmotta.

Has anyone gotten it to work?

majherek, 2018-08-28:
Hi,

in https://github.com/arendst/Sonoff-Tasmota/issues/2764#issue-324427694 there is my user_config_override.h. Try it...

theminor, 2018-08-28:
> Hi,
> 
> in #2764 (comment) there is my user_config_override.h. Try it...

No luck. I'm still getting Error (29)...

majherek, 2018-08-31:
Did you uncomment user_config_override.h in user_config.h?

theminor, 2018-08-31:
Yes, and my MQTT settings seem to be working fine - as with the OP, @younesmaia, it apparently connects and verifies the MQTT connection and fingerprint, but then crashes immediately.

majherek, 2018-09-04:
Hmm...

Do you use Arduine IDE? Which ESP8266 Board version do you use? I test it on 2.4.1 with IwIP Variant: "v2 Lower Memory"