arendst/Tasmota #2839: Support for Ifan02

sleezy100, 2018-05-26:
Hi, Just ordered 2 of these and was wondering if there will be any support for them in the near future ?

Sonoff iFan02: Convert Non-smart LED Ceiling Fan to WiFi Smart Ceiling Fan with Light
Model: IM170811003

Remote ON/OFF–Turn on/off the fan and light from anywhere
Control Fan Speed–Change the fan speed level: 1/2/3/Smart
Adjust the light intensity–Dimmable lights support: white/warm white/warm

Thanks

abzman, 2018-05-27:
I just ordered 9 on the same assumption.  There's probably just a pin to PWM for the LED, not sure about the speed.  And getting that remote to still work will be fun.

arendst, 2018-05-28:
Support might me added but do not expect it within 8 weeks.

MYLE-01, 2018-05-29:
I can wait I just ordered 1

Snipercaine, 2018-06-10:
Me too.  Got one on the way.

ageorgios, 2018-06-12:
Really great if support is added!

palitu, 2018-06-15:
Im going to assume it works exactly the same as everyother fan controller, it just uses a different comms protocol to communicate to it than the switch.

But these look out standing, and i will be upgrading as soon as we have some custom firmware for them!

palitu, 2018-06-15:
to be honest - i wont be waiting for firmware - i'll just use the controller for now

digiblur, 2018-06-15:
Anxious to see who gets one first and pops it open.

palitu, 2018-06-15:
we have a couple en route...

palitu, 2018-06-15:
I just love that it is a pop in replacement, and has WAF via the remote
control (i already have a remote on my fans) and will integrate into Home
Assistant, so i can turn on and off with motion sensors etc.

And in the mean time, i can just use it as is!

tyjtyj, 2018-06-15:
What the fun without tasmota.... Hehe

palitu, 2018-06-15:
oh no - i want Tasmota on it, but i can improve what i have in the mean time :)

PrathikGopal, 2018-06-18:
I can wait , thanks 👍

WayneManion, 2018-06-21:
I ordered a few of them, but the ebay vendor canceled my order. Now the item appears to have disappeared from ebay entirely?

arendst, 2018-06-23:
My ifan02 is on the way. When received all I need is a fan (LOL) and good weather...

nmatt25, 2018-06-25:
ordered 2 of them and are being delivered via DHL today ... will post pictures as soon as they get here .... feels like Christmas morning ... lol

nmatt25, 2018-06-25:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
now for the install

nmatt25, 2018-06-25:
[image]
[image]
[image]

nmatt25, 2018-06-25:
Good News:
OK ... Install went painless as expected. Just swapped out old harbor breeze wireless controller with new sonoff new one. Once powered up I easily paired controller with the eWeLink app on my phone and then proceeded to discover new devices via Alexa app and all went well and I can control everything via the remote, eWelink app, and Alexa via voice. 

Bad News:
When I proceeded to test all the functions I started with the easy part which was turning on and off the light which worked but only On or Off no "dimming" but maybe I am doing something wrong but will look into that further. Next went to test the fan .... Let me start with saying that yes I did make sure that I had the pull chain set in the position for the fan to be on HIGH/Full speed. When I turned on the fan via the remote by pressing 3 (high) the fan came on but did not seem like it was the full speed (doubled checked the chain setting) so I figured I would check the 2 (medium) button and the fan slowed but seemed like it was WAY slower then medium and then tried 1 (low) which resulted in the fan barely even spinning. I am not sure if this is something that can be corrected via firmware or not but it is very disappointing and think I am going to wait to install the second one until further testing.

Next Project: 
now to see about integration with Home Assistant!

Snipercaine, 2018-06-25:
Thanks Matt!

digiblur, 2018-06-25:
I don't recall the light being dimmable on their description. Curious to see someone put a volt/current meter measurements on it. I do see a few jumper pins.

nmatt25, 2018-06-25:
this is from there site but like I said maybe I'm missing something ... lol 

    Remote ON/OFF–Turn on/off the fan and light from anywhere
    Control Fan Speed–Change the fan speed level: 1/2/3/Smart
    Adjust the light intensity–Dimmable lights support: white/warm white/warm
    App Support –Free iOS and Android mobile App eWeLink
    Sync Status–Real-time device status provided to App
    Timing–Set scheduled/countdown to turn on/off at a specified time
    Share Control– Control your smart home together with your family
    Scene–Turn on/off a gang of devices with one tap
    Smart Scene–Triggered on/off the fan or light by environmental conditions from sensor
    Compatibility –Works perfectly with Amazon Alexa, Google Assistant, Google Nest

abzman, 2018-06-26:
I don't know how they're claiming that there's a bicolor, color-temp adjustable LED output on this thing, it's clearly just one relay for the output on the light.  I'll trace more of it later, interested in getting under the can and also sniffing the remote (probably 433mhz)

one of these even: https://www.alibaba.com/product-detail/Hot-selling-FANHAR-W11-1A2STE-5VDC_60553617588.html

palitu, 2018-06-26:
I would imagine that there is a triac chopping the wave to dim the light a bit.

Hopefully the fan integration actually works!

abzman, 2018-06-26:
[image]
[image]

Take a look at these pictures and correlate to the picture of the top showing 4 relays in a row, it's just the last relay that connects the light output to the live input.

abzman, 2018-06-27:
[image]
[image]
[image]
[image]

There is an arm just for the remote? god, why?

abzman, 2018-06-28:
I was wrong, it looks like the ARM controls the relays through the driver.  This may be harder than I thought...

arendst, 2018-06-29:
Received mine today and can verify that the light is simply controlled by 1 of the four relays. So it's on/off only as also stated on the ITEAD website. They do mention cold/warm control but that probably only works with a specific light connected to the relay.

The other three relays control the voltage to the fan; 1 connects power directly and would be fan speed 3, the other two connect power via a 2.5uF capacitor allowing to control the voltage send to the fan. I guess one of the two capacitor relays would make up fan speed 1, both capacitor relays on would be fan speed 2.

I expect control and sequencing is handled by the supporting microcontroller so all I need to do is finding out what GPIOs are used between the esp8285 and the microcontroller...

Stay tuned...

digiblur, 2018-06-29:
Nice work so far! Definitely going to pull the trigger on this.

My dream for this would be two buttons connected to GPIO pins. One would toggle the light. The other would toggle the fan through the 4 positions. A long press of the fan button would toggle the fan on/off using the last speed as the on state. Firmware doesn't have to do all that as some could be done via the home automation but the rules section might work out as well.

Snipercaine, 2018-06-29:
You all must live closer to China than I do... I'm still waiting...

sleezy100, 2018-06-29:
Mine finally arrived Wednesday, I'm in UK.     Would like to thank all you guys for spending the time to get these things working.

Cheers Guys

bluefoxlee, 2018-06-30:
For the light. I guess probably because most ceiling fan/light in China comse with another LED driver inside the light, which you can switch between warm/cold/both. (Since I just finish my installlation last week). 

I can take a photo of it, if anyone need it. But I guess it’s nothing to do with ifan02.

arendst, 2018-06-30:
Update.
- Flashed tasmota 6.0.0b by grounding gpio00 AND pressing the button at the same time.
- Configured as Sonoff 4CH Pro to see what happens regarding the four relays.
- Oh Joy. The remote still switches the relays as before toggling the 4 relays on the webpage when the different buttons are pressed.
- Need to figure out how to support this with a fan command...

arendst, 2018-06-30:
For now when configured as Sonoff 4CH you have control with Tasmota like the RF remote using the following commands either serial, web or mqtt:

- ``power1 2`` toggles the light
- ``backlog power2 1; power3 0; power4 1`` set fan to mode 3 (full speed)
- ``backlog power2 1; power3 1; power4 0`` set fan to mode 2 (mid speed)
- ``backlog power2 1; power3 0; power4 0`` set fan to mode 1 (low speed)
- ``backlog power2 0; power3 0; power4 0`` set fan to mode 0 (off)

arendst, 2018-07-01:
Release of Sonoff iFan02 in Tasmota.

[image]

closed 2018-07-01 by arendst

palitu, 2018-07-02:
Super Work!

Thanks Arenst!  I am al ittle bit wary of needing the connection to mains whilst flashing.  Unless im reading it wrong, we are going to need extremely detailed instructions for this one!

digiblur, 2018-07-02:
DrZzs.. Paging DrZzs!  I am sure he's already thinking of a video to show on this one. 

I only recently ordered mine so it would take me quite a while to get a flashing video out and it would be nowhere in comparison to his video anyways.

lorenzop, 2018-07-02:
If possible, I would use a bench supply to power it in the instructions

Snipercaine, 2018-07-02:
Great work Theo (and crew)!
Mine should be here soon.  Video not long after.  
Am I wrong assuming that we can flash with only the USB/Serial power without connecting mains?  As long as we don't activate the relays?

arendst, 2018-07-02:
Even with all relays off the USB/Serial converter still is not able to power the device for flashing. A bench supply as opted above should work.

I just my rubber underpants and managed to flash it while connected to AC. Tricky, so I would not advertise it  in the video.

digiblur, 2018-07-02:
It needs 1.21 gigawatts! Great Scott.

Makes sense with the extra hardware drawing too much current for the little USB flasher to power.

palitu, 2018-07-02:
jiga watts...

digiblur, 2018-07-03:
Depends on your way of spelling but it is pronounced like a J

palitu, 2018-07-03:
HAHAH! What have i started -
https://en.wikipedia.org/wiki/Giga-#Pronunciation

turns out the Yanks started saying it with a soft G in the 60s and 80s.
Potentially should be a hard G due to it being Germanic roots.

Either way - in text land, don't really care about the pronunciation do we!

thebradleysanders, 2018-07-03:
Are the functions of the remote retained in Tasmota? Thanks for any responses.

arendst, 2018-07-03:
Yes, remote works as expected.

MimbaMonkeyHouse, 2018-07-04:
Thanks @arendst. Ordere 6 units. You should really get a salary from Sonoff. Thanks for Tasmota!

arendst, 2018-07-04:
For those eager to see a schematics of the iFan02 see the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-iFan02

Courtesy of iTead.

digiblur, 2018-07-04:
That's awesome you got the schematic! Exactly what I wanted. Saves me from ohming stuff out when I get mine to see what is what when adding pull up resistors and such.

finity69x2, 2018-07-04:
Dumb questions probably but I'm not a prior Tasmota user....

I see the remote still works - great!

I assume the device is also controllable via a MQTT message? And that the status is sent via an MQTT message also?

So if I turn on/off the light or change fan speeds I can sniff the MQTT messages to get the actual status? And send an MQTT message to change the status?

digiblur, 2018-07-04:
Take a look at the console and see what messages are being sent. Should be pretty easy to figure the command messages of the console messages.

arendst, 2018-07-04:
@finity69x2 Yep.

finity69x2, 2018-07-04:
I haven't odered any yet.

I just wanted to make sure that I knew what to expect before I did order any.

I've been burned before on my literally year long quest to try to find a Home Assistant compatible Fan/light controller! Hunter fans are ridiculously difficult to properly integrate.

finity69x2, 2018-07-04:
Awesome!

I just dropped the hammer and ordered a few. 

Hopefully the wife won't drop the hammer on me! :)

jumblies, 2018-07-04:
Hey Theo.  Impeccable work.  

Got two yesterday and set about flashing.  I'm actually using ESPTool just because I'm familiar with it.  I made a few bonehead mistakes that might help others.  I started working with the J2 header and I thought that I was getting garbage on the serial, but not really.  Once I did the correct wiring to the J3 header, comms were up!

My uploads were failing at about 30%.  I cut the baudrate and this was my command
```
esptool.py -b 74880 --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
```

Still had trouble so I hooked the VCC back up in addition to mains power (rubber undies :) ) and that solved it completely.  Configured as 4 Ch and just like you said, worked like a charm with the remote and through the tasmota web interface.  

Once I get a binary or compile my own, I'll convert it over to your actual ifan2 module configuration.  

I really appreciate your efforts.  You have created an operating system that runs on a light switch.  Amazing.

palitu, 2018-07-05:
I can see from the schematic that there is a zero line detector, but i cannot see where it would be used.  I would have imagined that it is used for a triac for light dimming, but there doesn't seem to be anything obvious.

any ideas?

jumblies, 2018-07-05:
Okay,  this is a EE question, not a tasmota question at all.

Hooked it up upstream of the fan wiring (going into the fan where the pull chain resides)
- speed 1, it is way too slow, estimate 30 rpm (w/o ifan, prob double that)
- speed 2, passable as low
- speed 3, faster than normal; very windy

My understanding is that increasing capacitance in AC circuits increases the power factor and that explains the faster than normal, but I'm trying to wrap my head around how to fix the other two speeds.  I'm guessing this is meant to work on a single speed fan with no other capacitance than supplied by the iFan02.  

Anyone have any ideas?
I'd consider:
- Wiring tasmota fan directly to fan motor coil but this loses reverse functionality, I think. 
- bypassing the pull chain switch and associated fan housing capacitor 

BTW: It works fine as a fan component in homeassistant but speed selection is a little clunky on the UI.  Will work on that later.  

Shown below is the capacitor that is in the fan.  
[image]

mustangarcher, 2018-07-06:
Do you release the button AND disconnect GPIO00 after applying AC to enter it into programming mode?  Is there an led indication that it is in programming mode and ready to be flashed?

jumblies, 2018-07-06:
Having just done my second, yes.  I do this.  Short 0, and hold the button
down, pub in usb (or apply power to uart) then hit AC mains power.

jumblies, 2018-07-06:
Because I needed three arms and wasn't willing to wait for CRISPR to work,
used a foot switch.

[277 more replies]