arendst/Tasmota #2840: Sonoff User Password?

shayshtern, 2018-05-26:
I flashed sonoff touch with sonoff.bin - flash was OK .
I restarted the Sonoff  -  connect it after 4 press for 192.168.4.1 
the webpage open but asking for User Password...  what?
is there default?

Frogmore42, 2018-05-27:
From where did you get the bin file?  The default is a blank password.

shayshtern, 2018-05-27:
From the official site, here.

shayshtern, 2018-05-27:
SonOTA the software working. But I cant find the wifi "ITEAD0000XX"...
And I don't know why but there is user password:
https://cdn.pbrd.co/images/Hn6NMyj.png

seetendra, 2018-05-27:
Is the same happening on a Machine as well.

curzon01, 2018-05-27:
Try
User: admin
Password: <leave empty>

shayshtern, 2018-05-28:
As I said, I did install SonOTA - But can't find the Wi-Fi of the Sonoff "ITEAD000XXX...". 
Secondly with FTDI and Atom after 1,028 tries - failed every time. 
And I did success flash it with ESPEasy - but with user and password :( 
BTW - How can I get notifications from GitHub?

curzon01, 2018-05-28:
I realy don't understand the status - the picture above shows that you have installed Tasmota, next you say you'd installed ESPEasy. I suggested to try the login data above. Did you try that?

- The network "ITEAD000XXX..." is only available with origin firmware from Itea. If you already flashed another firmware (like Tasmota or ESPEasy) you can not have this Wifi network anymore.
- SonOTA only works if you have the original Itea Sonoff firmware on the device. SonOTA does not work if you already flashed another firmware (like Tasmota or ESPEasy).

shayshtern, 2018-05-28:
I did try Admin no password - didn't connect. 😕 
I have original Sonoff Touch that can connect to the original EWElink - power it - use the Ewelink and all works, it didn't ask me for find an Wi-Fi all automatically😟 dammm...

curzon01, 2018-05-28:
Ok, unfortunatelly I don't know the steps you did with SonOTA. The steps are
1. `cd SonOTA`
2. `python3 sonota.py --serving-host <your-pc-ip> --wifi-ssid <your-wifi-ssid> --wifi-password <your-wifi-pw>`
3. wait a moment and refresh your Wifi list on your PC, "ITEAD-XXXXX" should be visible.

If "ITEAD-XXXXX" is not visible, try the following:
- disable your PC Wifi and enable it again. Some PC do not display new Wifi immediately and will show the new Wifi only when you disable/enable it
- try step using the following additional parameters (in all comibnations) on `python3` line `--legacy` and `--slowstream` so you have 3 possible alternate command lines for step 3.
  3b) `python3 sonota.py --serving-host <your-pc-ip> --wifi-ssid <your-wifi-ssid> --wifi-password <your-wifi-pw> --slowstream`
  3c) `python3 sonota.py --serving-host <your-pc-ip> --wifi-ssid <your-wifi-ssid> --wifi-password <your-wifi-pw> --legacy`
  3d) `python3 sonota.py --serving-host <your-pc-ip> --wifi-ssid <your-wifi-ssid> --wifi-password <your-wifi-pw> --legacy --slowstream`

reloxx13, 2018-05-30:
this is NOT flashed with SonOTA.

SonOTA default firmware is tasmota 5.9.0 or older. i dont know exacty, but its an old firmware.

On the Screenshot it is 5.14.0 in the footer.

Source: https://github.com/mirko/SonOTA/tree/master/static (8 month old)

so he did flash some other firmware and he should had access to that device and maybe set the pw  himself and forgot or someone who has access to his wifi set it.

curzon01, 2018-05-31:
@reloxx13: I don't believe shayshtern has changed that but SonOTA uses the file `static/image_arduino.bin` to flash, which can replaced by any other image you want to have finally on your device - mine for example is always the current Tasmota I use, not 5.9.0

shayshtern, 2018-05-31:
SO, I flashed the sonoff again. same issue with the login page. 
I did flash another sonoff with the same way and the SAME BIN file, and it is works ok!!! no password in it

arendst, 2018-05-31:
@shayshtern your problem is within the settings area. When you reflash a firmware the settings do not change so you will still see the password request.

Solution:
1) log in via serial and execute command ``reset 1``. This will reset all settings and after reboot you can use the same commandline to set your required parameters using commands.
2) complete flash erase as documented in the wiki. Then reload firmware.

shayshtern, 2018-05-31:
I used original bin file from here in github. I didn't change any setting. so?

arendst, 2018-05-31:
Could be but.... The firmware looks for already present settings using a key at certain flash locations. If found it starts using that location as a valid settings area which sometimes results in your experience.

BTW. Why don't you reset the password with command ``webpassword 0`` if the password is the only problem you have with it...

shayshtern, 2018-05-31:
OK! I connect with console serial monitor from Atom, send 
> reset 1
now I'm able to search and connect to my network! 
CLOSE and SAVE for others,
thank you. appreciate your work!.

seetendra, 2018-05-31:
I have a Question.
For sonOTA the Touch shouls give a SSID with ITEDxxxxx .
But for some reason i am not able to see anything on the touch i got yesterday.

This is the process ism following 

1. Power on the device and then press and hold the first button (3gang) till 7 seconds the light started to blink but still no luck .
I dont see the SSID. 

Any suggestions except to try the Serial Port.

closed 2018-05-31 by arendst

unique904, 2019-06-15:
Try using 
admin
DVES_PASS

VivantSenior, 2019-07-19:
What are the restrictions of using password in Web Admin Panel? Special characters, length?