arendst/Tasmota #2842: Problems T1 to be recognized by Alexa V5.14

motorrad, 2018-05-27:
Problems T1 to be recognized by Alexa V5.14 German
5.13 German works without Problems

curzon01, 2018-05-27:
I don't think somebody can actual help to solve your issue if you skip complete all the infos requested by opening the issue. People can read a lot of things e.g. from `Status 0` info (self compiled? which library you are using? what's the config...)

motorrad, 2018-05-28:
Already compiled German bin

Jason2866, 2018-05-28:
What @curzon01 said. Without informations to your build, nobody can help...

Frogmore42, 2018-05-28:
I think he means he downloaded the pre-compiled binary.

In which case, you still need to provide the log and status 0 information.

I am not sure why some people are reluctant to provide that even after multiple requests.  It's like they really don't want their problem solved 😀.

motorrad, 2018-05-29:
Working with
Tasmota Version	5.13.1
Build-Datum & -Uhrzeit	2018.05.01 21:05:42
Core-/SDK-Version	2_4_0/2.1.0(deb1901)
Laufzeit	1T01:11:01
Anz. Flash Schreibzugriffe	657 at F8000
Anzahl Startvorgänge	71
Grund für Neustart	Software/System restart
Name [friendly name] 1	Badezimmerlicht
	 
AP1 SSID (RSSI)	XXXXXXX (88%)
Hostname	sonoff-7978
IP-Adresse	192.168.97.140
Gateway	192.168.97.50
Subnetzmaske	255.255.255.0
DNS-Server	192.168.97.50
MAC-Adresse	60:01:94:9A:FF:2A
	 
MQTT	deaktiviert
	 
Emulation	Belkin WeMo
mDNS-Ermittlung	aktiviert
mDNS-Bekanntmachung	Web-Server
	 
ESP Chip ID	10157866
Flash Chip ID	1327185
Realer Flash Speicher	1024kB
Ges. Flash Speicher	1024kB
Ben. Flash Speicher	530kB
Verf. Flash Speicher	472kB
Freier Arbeitsspeicher	19kB

Not working with
 v5.14.0  
 3ad1f43 
German.bin precompiled

Frogmore42, 2018-05-30:
Many people have had issues with core 2.4.x

Compile it with 2.3.0 and it will probably make it better.

curzon01, 2018-05-30:
see also #2414 2586, #2724

ascillato, 2018-05-30:
Please, also see https://github.com/arendst/Sonoff-Tasmota/issues/2814

closed 2018-06-08 by ascillato2