arendst/Tasmota #2843: Log Source IP address of State and Command requests

i3laze, 2018-05-27:
I've tried every log level (including 4), but there's no info about Source IP, which initiated the State or Power command, and used protocol (MQTT or HTTP-GET).

I'd like to distinguish between automation tasks (from HomeBridge IP), internal Timers and manual switches.

curzon01, 2018-05-27:
HTTP: True, you do not see the source of an HTTP request.

MQTT: Request are always coming from the MQTT-server where your Tasmota is connected to.
To distinguish sources you need to 'go' to your MQTT server logs, a MQTT client like Tasmota is the wrong place to search for such info.

i3laze, 2018-05-28:
I’m actually running with MQTT disabled, using “homebridge-sonoff-tasmota-http v1.2.0“ plugin and  wget scripts.

ascillato, 2018-05-30:
Hi @i3laze 

Now the SRC was added to Tasmota. I tested and works really good.

Please, also test it and if your issue was solved, please close it. Thanks :+1:

i3laze, 2018-05-31:
If only there were binaries released for 5.14.0b..
I’s better just believe you, or post a fragment of log (level 3) here to close the thread.

i3laze, 2018-05-31:
@ascillato , thanks! I was lucky with OTA this time (last time migrating 5.13 to 5.14 I had to TTL two units cause of #2602).

Works great! I still suggest moving it to weblog=2 (default). (or moving CFG writes to level 4).
```
11:01:12 SRC: WebGui from 10.0.0.100
11:02:06 SRC: WebCommand from 10.0.0.100
```

i3laze, 2018-05-31:
I also see there is an actual info duplicate on WebLog 3 (LOG_DEBUG):
`11:28:23 HTP: Command
11:28:23 SRC: WebCommand from 10.0.0.100`

curzon01, 2018-05-31:
Command via [Web](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#web).
Not the same as via GUI Console so you can also differ between sources coming from Web GUI Console or direct HTTP commands.

closed 2018-05-31 by i3laze