arendst/Tasmota #2845: LWiP 1.4 Sonoff Basic Status 0 cause crash

majherek, 2018-05-27:
Hi,

I found that my sonoffs basic crashed if I use STATUS 0.

```
00:00:00 CFG: Use defaults
00:00:00 Project sonoff11 Sonoff 11 (Topic sonoff11, Fallback sonoff11, GroupTopic sonoffs) Version 5.13.1-2_4_1
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff11...
00:00:04 WIF: Connected
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Verify TLS fingerprint...
00:00:06 MQT: Verified using Fingerprint1
00:00:06 MQT: Connected
00:00:06 MQT: tele/sonoff11/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff11/POWER = 
00:00:06 MQT: tele/sonoff11/INFO1 = {"Module":"Sonoff Basic","Version":"5.13.1","FallbackTopic":"sonoff11","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff11/INFO3 = {"RestartReason":"Power on"}
22:05:42 MQT: stat/sonoff11/RESULT = {"POWER":"OFF"}
22:05:42 MQT: stat/sonoff11/POWER = OFF
22:05:49 MQT: tele/sonoff11/STATE = {"Time":"2018-05-27T22:05:49","Uptime":"0T00:00:13","Vcc":3.358,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","RSSI":74,"APMac":"06:8D:DB:DB:97:47"}}
22:05:53 CMD: STATUS 0
22:05:53 MQT: stat/sonoff11/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff 11"],"Topic":"sonoff11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:05:53 MQT: stat/sonoff11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/sonoff11.ino.bin","RestartReason":"Power on","Uptime":"0T00:00:17","StartupUTC":"2018-05-27T21:05:36","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"3FA000"}}
22:05:53 MQT: stat/sonoff11/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-27T23:03:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:05:53 MQT: stat/sonoff11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix","MAJCOMNET HTC"],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
22:05:53 MQT: stat/sonoff11/STATUS4 = {"StatusMEM":{"ProgramSize":515,"Free":2556,"Heap":4,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
22:05:53 MQT: stat/sonoff11/STATUS5 = {"StatusNET":{"Hostname":"sonoff11","IPAddress":"192.168.20.11","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"5C:CF:7F:72:0A:1E","Webserver":2,"WifiConfig":5}}
22:05:53 MQT: stat/sonoff11/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"sonoff11","MqttClient":"sonoff11","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:05:54 MQT: stat/sonoff11/STATUS7 = {"StatusTIM":{"UTC":"Sun May 27 21:05:53 2018","Local":"Sun May 27 22:05:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"03:24","Sunset":"19:41"}}
22:05:54 MQT: stat/sonoff11/STATUS10 = {"StatusSNS":{"Time":"2018-05-27T22:05:54"}}
22:05:54 MQT: stat/sonoff11/STATUS11 = {"StatusSTS":{"Time":"2018-05-27T22:05:54","Uptime":"0T00:00:18","Vcc":3.357,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","RSSI":74,"APMac":"06:8D:DB:DB:97:47"}}}
22:06:12 MQT: Attempting connection...

Exception (29):
epc1=0x4022509c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff3d50 end: 3fff42f0 offset: 01a0

>>>stack>>>
3fff3ef0:  00000080 3fffbf94 3fffb544 40232867  
3fff3f00:  3fff87e4 3fffbf94 3fffb544 40233255  
3fff3f10:  3fff917c be793c72 d438b3cf 00000020  
3fff3f20:  3fffbf8c 3fffbf94 3fffbf74 00000020  
3fff3f30:  3fffbf8c 3fffbf94 00000010 4023354a  
3fff3f40:  3fff52dc 3fff9074 00000496 00000001  
3fff3f50:  00000001 3fffbf74 3fff688a 402322c4  
3fff3f60:  00000200 3fff902c 3fff6707 00000000  
3fff3f70:  00000200 3fff902c 3fff6707 40234649  
3fff3f80:  3fff3fb0 00000000 000001d0 00000030  
3fff3f90:  f7cc1954 3fff5f00 00000006 00000004  
3fff3fa0:  3fff6701 3fff62bc 3fff6701 4022f344  
3fff3fb0:  a3450303 4caa3164 59788c1f 685bc7e2  
3fff3fc0:  564e7f05 5bf7d7eb 05ad98c0 bea39c87  
3fff3fd0:  d175190e da000887 fde8d006 19543181  
3fff3fe0:  3fff6701 00000004 3fff62bc 4022f7d9  
3fff3ff0:  3fff52dc e158191f cda2843f 3fff609c  
3fff4000:  00000000 600011f0 3fff5efc 3fff609c  
3fff4010:  00000000 00000004 00000004 40218df8  
3fff4020:  0000002c 00000b6c 00000064 40220028  
3fff4030:  0000004a 00000000 0000002d 00000004  
3fff4040:  3fff6701 00000004 3fff62bc 00000004  
3fff4050:  3fff6701 00000004 3fff62bc 4022f14c  
3fff4060:  7d321a08 00000006 00000000 0000002e  
3fff4070:  00000063 3fff6701 3fff62bc 4022f528  
3fff4080:  00000017 3fff6734 3fff62bc 401004d8  
3fff4090:  3fff40e8 0000000e 00000010 3fff628c  
3fff40a0:  40218bac 3fff62bc 3fff40e0 01000000  
3fff40b0:  3fff40e8 0002270f 3fff62bc 4022f2b4  
3fff40c0:  40104ce4 3fff1dc1 3fff91a4 3fff628c  
[... 191 lines cut ...]
22:37:02 MQT: stat/sonoff11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/sonoff11.ino.bin","RestartReason":"Power on","Uptime":"0T00:00:15","StartupUTC":"2018-05-27T21:36:47","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"3FA000"}}
22:37:02 MQT: stat/sonoff11/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-27T23:35:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:37:02 MQT: stat/sonoff11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix","MAJCOMNET HTC"],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
22:37:02 MQT: stat/sonoff11/STATUS4 = {"StatusMEM":{"ProgramSize":516,"Free":2556,"Heap":6,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
22:37:02 MQT: stat/sonoff11/STATUS5 = {"StatusNET":{"Hostname":"sonoff11","IPAddress":"192.168.20.11","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"5C:CF:7F:72:0A:1E","Webserver":2,"WifiConfig":5}}
22:37:02 MQT: stat/sonoff11/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"sonoff11","MqttClient":"sonoff11","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:37:02 MQT: stat/sonoff11/STATUS7 = {"StatusTIM":{"UTC":"Sun May 27 21:37:02 2018","Local":"Sun May 27 22:37:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"03:24","Sunset":"19:41"}}
22:37:02 MQT: stat/sonoff11/STATUS10 = {"StatusSNS":{"Time":"2018-05-27T22:37:02"}}
22:37:02 MQT: stat/sonoff11/STATUS11 = {"StatusSTS":{"Time":"2018-05-27T22:37:02","Uptime":"0T00:00:15","Vcc":3.341,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","RSSI":76,"APMac":"06:8D:DB:DB:97:47"}}}
22:37:41 CMD: STATUS 0
22:37:41 MQT: stat/sonoff11/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff 11"],"Topic":"sonoff11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:37:41 MQT: stat/sonoff11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/sonoff11.ino.bin","RestartReason":"Power on","Uptime":"0T00:00:54","StartupUTC":"2018-05-27T21:36:47","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"3FA000"}}
22:37:41 MQT: stat/sonoff11/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-27T23:35:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:37:41 MQT: stat/sonoff11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix","MAJCOMNET HTC"],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
22:37:41 MQT: stat/sonoff11/STATUS4 = {"StatusMEM":{"ProgramSize":516,"Free":2556,"Heap":7,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
22:37:41 MQT: stat/sonoff11/STATUS5 = {"StatusNET":{"Hostname":"sonoff11","IPAddress":"192.168.20.11","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"5C:CF:7F:72:0A:1E","Webserver":2,"WifiConfig":5}}
22:37:42 MQT: stat/sonoff11/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"sonoff11","MqttClient":"sonoff11","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:37:42 MQT: stat/sonoff11/STATUS7 = {"StatusTIM":{"UTC":"Sun May 27 21:37:42 2018","Local":"Sun May 27 22:37:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"03:24","Sunset":"19:41"}}
22:37:42 MQT: stat/sonoff11/STATUS10 = {"StatusSNS":{"Time":"2018-05-27T22:37:42"}}
22:37:42 MQT: stat/sonoff11/STATUS11 = {"StatusSTS":{"Time":"2018-05-27T22:37:42","Uptime":"0T00:00:55","Vcc":3.340,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","RSSI":76,"APMac":"06:8D:DB:DB:97:47"}}}
```

Is this normal?

I use 4MB flash in my sonoff (I replace the original one).

ascillato, 2018-05-27:
compiled with Arduino IDE or Platform.io?

ascillato, 2018-05-27:
try the esplib 2.3.0 instead of 2.4.1 that you are using.

arendst, 2018-05-28:
It's caused by TLS. Lack of resources as of 5.14.0.

I guess it's finally time to say goodbye to TLS.

arendst, 2018-05-28:
TLS is very RAM hungry with your experience as result. More flash doesn't solve that.

I guess this is the end of TLS as new features use more RAM even if you disable all sensors as the settings area has permanently grown by rule save area to almost 4k eaten from RAM too.

majherek, 2018-05-28:
Hi @ascillato,
I compile it with Arduino IDE. I will try with esplib 2.3.0, but I think I already check it (I will check it for sure today). 

@arendst for me it doesn't matter -> I use LWiP v2 from long time and everything is working fine. 
Can you not to throw away TLS support? IMHO security is important too.

majherek, 2018-05-28:
@ascillato,

for me with esplib 2.3.0 is the same.

closed 2018-05-28 by majherek

younesmaia, 2018-05-31:
@arendst 
Is there any combination of features that allows for sufficient RAM for TLS to work?

majherek, 2018-06-04:
@younesmaia,

yes, I have working 5.14.0b with TLS.

In **user_config.h** uncomment:
```
#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md
```
and create **user_config_override.h** file like this (remember, to change NAMES, IP, etc.). 
```
/*
  user_config_override.h - user configuration overrides user_config.h for Sonoff-Tasmota

  Copyright (C) 2018  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// force the compiler to show a warning to confirm that this file is inlcuded
#warning **** user_config_override.h: Using Settings from this File ****

/*****************************************************************************************************\
 * USAGE:
 *   To modify the stock configuration without changing the user_config.h file:
 *   (1) copy this file to "user_config_override.h" (It will be ignored by Git)
 *   (2) define your own settings below
 *   (3) for platformio:
 *         define USE_CONFIG_OVERRIDE as a build flags.
 *         ie1 : export PLATFORMIO_BUILD_FLAGS='-DUSE_CONFIG_OVERRIDE'
 *         ie2 : enable in file platformio.ini "build_flags = -Wl,-Tesp8266.flash.1m0.ld -DUSE_CONFIG_OVERRIDE"
 *       for Arduino IDE:
 *         enable define USE_CONFIG_OVERRIDE in user_config.h
 ******************************************************************************************************
 * ATTENTION:
 *   - Changes to SECTION1 PARAMETER defines will only override flash settings if you change define CFG_HOLDER.
 *   - Expect compiler warnings when no ifdef/undef/endif sequence is used.
 *   - You still need to update user_config.h for major define USE_MQTT_TLS.
 *   - All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
\*****************************************************************************************************/

/*
Examples :

// -- Master parameter control --------------------
#undef  CFG_HOLDER
#define CFG_HOLDER        0x20161209             // [Reset 1] Change this value to load SECTION1 configuration parameters to flash

// -- Setup your own Wifi settings  ---------------
#undef  STA_SSID1
#define STA_SSID1         "YourSSID"             // [Ssid1] Wifi SSID

#undef  STA_PASS1
#define STA_PASS1         "YourWifiPassword"     // [Password1] Wifi password

// -- Setup your own MQTT settings  ---------------
#undef  MQTT_HOST
[... 365 lines cut ...]
#endif

// -- Low level interface devices -----------------
#ifdef USE_IR_REMOTE
#undef USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+4k code, 0k3 mem, 48 iram)
//#define USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+4k code, 0k3 mem, 48 iram)
#endif

#ifdef USE_WS2812
#undef USE_WS2812                               // WS2812 Led string using library NeoPixelBus (+5k code, +1k mem, 232 iram) - Disable by //
//#define USE_WS2812                               // WS2812 Led string using library NeoPixelBus (+5k code, +1k mem, 232 iram) - Disable by //
#endif

#ifdef USE_ARILUX_RF
#undef USE_ARILUX_RF                            // Add support for Arilux RF remote controller (+0k8 code, 252 iram (non 2.3.0))
//#define USE_ARILUX_RF                            // Add support for Arilux RF remote controller (+0k8 code, 252 iram (non 2.3.0))
#endif

#ifdef USE_SR04
#undef USE_SR04                            // Add support for Arilux RF remote controller (+0k8 code, 252 iram (non 2.3.0))
#define USE_SR04                                 // Add support for HC-SR04 ultrasonic devices (+1k code)
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_
```

Mebus, 2018-08-17:
Is it correct, that you disabled the web server to get MQTT to work with TLS?

Mebus

orcema, 2018-08-17:
Why don't we try to use dynamic memory allocation for the web server as well for the mqtt service.
Thus if the mqtt service in TLS mode encounter problems connecting to server it will disconnect, free memory and launch the the local web server to allow the user to check for problems. The user can then  proceed to updates and launch a new mqtt connection attempt. 
New mqtt connection attempt will stop the local web server in order to free memory and lanuch the mqtt service. This process can repeat until successful connection to mqtt server.
Would this be a potential solution to insufficient memory ?