arendst/Tasmota #2846: Esp8266 12e industrial and Tasmota

vontrapus, 2018-05-28:
Hi guys!

Ok, I have this problem, I flashed the esp8266 12e with Tasmota, everything went good but he stuck and dont connect to the network. I tried srduino Ide and nodemcu flasher. I changed the ssid and tryed without password on the network but still dont want to connect, the led on the esp is blinking and try to search the network without luck.
Can anybody help a litle. Sorry for my bed english.
Thanks

Frogmore42, 2018-05-28:
You need to include the serial log from power on.  It will indicate what is going on.
You also need to read about cfg_holder and all the commands that you can send over serial, or web or MQTT when you get it connected.

vontrapus, 2018-05-28:
Thank you wery mutch
I modified the cfg-holder with -1  and flashed two times and now everyting is working.
Thanks again :)

ascillato, 2018-05-28:
Great. Happy to know that it is working for you. Please, close this issue. Thanks!