arendst/Tasmota #2847: Configure 4CH Pro wifi from precompiled binaries

gogotop32, 2018-05-28:
I flashed the Sonoff 4CH Pro using the 5.14.0 precompiled binary with no errors.  But how do i connect wifi?

I looked at the [Button usage](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-Usage) page in the wiki, but it doesn't say anything for the 4CH Pro.

I run DD-WRT and they do not support Wifi Protected Setup (WPS).
I really don't want to have to compile anything to get this working.
Am I missing something about the wifi configuration for the 4CH Pro?
Does the precompiled software need such a complicated wifi setup?

Frogmore42, 2018-05-28:
If you don't want to compile you need to use the serial interface to send commands.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

There are commands for setting the SSID and password.

gheesung, 2018-05-28:
If you have a spare router or if dd-wrt can setup guest network, you can set the default ssid/password that is stored in the pre-compiled binary in the guest network. Then join your PC to the network and you should be able to see your Sonoff device and start reconfiguring it your home network.

curzon01, 2018-05-28:
Suggested solutions above works, I think the simplies way is also described on Button usage you already found, but do not use th WPS function:
- hit button 4 times, a new Wifi network will be visible
- after you have connected to this Wifi your PC will get an additinal IP from net 192.168.4.0/24
- call the web page http://192.168.4.1 and configure the final Wifi you want to connect to

gogotop32, 2018-05-28:
I went with Frogmore42's solution and now have it connected.  I originally tried connecting with Putty and that did not work, so I tried TeraTerm and was able to send it commands.
Thanks for all the information.

closed 2018-05-28 by gogotop32