arendst/Tasmota #2848: Sonoff doesnt work after update

MrJoki007, 2018-05-28:
Today i wanted to update my sonoff basic with the firmware 5.10.0 to the new 5.14.0
I downloaded the newest sonoff-DE.bin and wanted to upload it on the UI. I got an error and looked it up and some user said i should flash the sonoff-basic.bin and then reflash the normal one.
After uploading it, it said upload finished and that its restarting.
After that, the green led lights up, blinks about 30 times and you hear the relay clicking once. 
And that repeats every 30 secounds. 
In my router i can see that its connecting to wifi for 9 secounds and than disconnecting again.

ayancey, 2018-05-29:
same thing happening to me :(

andyfitter, 2018-05-29:
I’ve seen the same on some U.K. T1 units.

rmounce, 2018-05-30:
I am having a similar issue with 6x Sonoff Basics (from different batches) after OTA upgrading from 5.12/13 to 5.14. None of my other Tasmota devices (S20, S22, TH10, H801, Wemos D1) are affected. 2 of these 6 did eventually connect to the network, but after another reboot they are lost.

Good news is that fault goes away when I am testing at a different location! My working theory is that some combination of the below factors are at play here:
- Noisy 2.4GHz radio environment
- Cheap and nasty RF filtering (or lack thereof) on the Sonoff Basic
- Reduction of some re-try or timeout period between 5.13 and 5.14
- Possibly a neighbouring AP that is stuck in WPS config mode?

cHunter789, 2018-05-30:
For me sonoffs basic with tasmota  5.13 and later are unusable with precompiled/ota firmware. In 5.13 the problem was when trying connecting to mqtt server. Few days ago I have tried OTA to 5.14 and I stuck at minimal image. Even when i turned off a mgtt server it's not helped to get back a control over sonoff basic device.

Looks like no one testing the firmware with sonoff basic before release...

ascillato, 2018-05-30:
Hi,

I'm sorry to know that you are facing issues with last firmware. I have 12 sonoff devices at home working with last 5.14.0b without any problem.

Sometimes the issues comes from a problem in the configuration and/or Esp8266 SDK config in flash. It happened few times to me that upgrading by OTA from old versions of Tasmota, the OTA method failed and I have to erase all the flash and upload again but by cable. After that fresh start, it works really good.
**So I recommend to see troubleshooting on the wiki and/or flash by cable if OTA has failed.**

Each version is being tested before released, by a lot of people. In the development branch, bugs are fixed and some new features are added and tested. Then, after sometime, it is merged to the master branch and released as precompiled binaries.

So, Theo and a lot of people are spending their free time for free to contribute to this software and this github community. If you think, more testing is needed, **you are more than welcome** to test, find bugs and help other people in trobleshooting. Complaining is not a useful option. Helping is it.

reloxx13, 2018-05-30:
if not done already, try to let it full power loss for a few mins.

i had only 1 of 11 basics always restarting after 1-60mins, after i let it power loss for 10mins it got fixed.
i dont know why 😕 
i had this with 5.14.0a.
never had it before or after that, im always on the dev builds.

give it a try. i just released a fuse to get it power loss.

MrJoki007, 2018-05-31:
Plugged mine out for one day and now plugged it in again over the night. But still got the same problem

maciboy, 2018-06-01:
After my first Sonoff Basic died due to a failed OTA update some weeks ago, today my second (and last) one got the blinking issue after I’ve tried to update it to 5.14…
It is publishing a Wifi with its correct former host name, but I’m unable to connect :(
(I’ve no possibility to flash them by wire…)

Any tipps how I can fix it, since as I said I can see its wifi-network…?

ascillato, 2018-06-01:
Hi,

Sorry to know that you are unable to connect. Seems that your OTA has failed. In that case you need to flash by wire. There is an explanation on how to do that in the wiki.

maciboy, 2018-06-01:
Juhu, I’m back in the game (at least with the one that I bricked today^^)

What I did?
As you might know the Sonoff supports several actions when applying push-patterns on its button:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

I tried the ones that sound promising…
With the 7-short-presses option I had success :)
I was able to connect to its wifi, opened 192.168.4.1/up? and uploaded the sonoff.bin file manually

I really hope the same works for you!

stasrm, 2018-06-03:
I have 3 basic and 6 touch. I have successfully upgraded touch. But all basic were bricked after upgrade. I can't unbrick them even flashing through cable. I can see AP but i can't connect. I've flashed 5.13 and problem resolved.

Trago5, 2018-06-03:
@maciboy Thanks!

I had to use the short presses for both of my Sonoff Basics and upload sonoff.bin manually, too. Four presses worked for me.

However, after the upgrade most of my devices (2 Basics, 1 TH16) didn't work correctly with Home Assistant anymore. I could switch them on but couldn't switch them off. So I downgraded to 5.12 and now they work just fine again. Sonoff POW seems to be working with 5.14 without problems.

unscel, 2018-06-03:
I have all (8) my basic’s placed in wall switches. Each one of them have been soldered with resistors and capacitors to avoid interference for the switchmode1. Unfortunately all died (restart loop, blinking led) during ota uptate from 5.12 to 5.14. Deattaching, desoldering, erasing flash and reflashing thru wire, soldering and reattaching takes almost whole day but now all works fine at the end.

MrJoki007, 2018-06-04:
After a lot of time i got it in AP mode and connected to the wifi of the sonoff.
Then i changed the ip of my pc to 192.168.4.2 to connect to the sonoff webinterface.
It said sonoff basic please upgrade. So i uploaded the new sonoff.bin version 5.14.0.
But after the update, i cant find the wifi of the sonoff in ap-mode anymore.

MrJoki007, 2018-06-04:
Now i wanted to flash it manually but when i try to flash it i get this message:
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
Is it now completly bricked?

rmounce, 2018-06-07:
I have done some more testing with my Sonoff Basic R2 (v1.0 PCB) and found the following results, performing OTA upgrades back and forwards between versions.

v5.13.1 built myself with 1.5/2.3.0: Good
v5.13.1 binary release 1.6/2.4.0: Good
v5.14.0 binary release 1.5/2.3.0: Bad, cannot connect to WiFi when powered from AC
v5.14.0 built myself with 1.5/2.3.0: Bad, cannot connect to WiFi when powered from AC
v5.14.0 built myself with 1.6/2.4.0: Good
457f438bacc28554885b9363bd31ac4a83e82c54 / v5.14.0a built myself with 1.5/2.3.0: Good

So I can reliably reproduce this issue that is only affecting my Sonoff Basics with v5.14.0 from the master branch, built with Arduino core 2.3.0.

It is difficult to narrow this down further as all commits on the master branch are squashed together for releases, and the development branch doesn't seem to be affected at all (even at the commit that should be syncing it up with the 5.14.0 release).

I did, incidentally, bisect a different issue on the development branch. 0d5117cdca1b1c909d1250f55488e88695f033bd breaks WiFi entirely on my Sonoff Basic and necessitates a serial reflash.

ascillato2, 2018-07-21:
Please, see #3224

If your issue is not solved, please reopen it. Thanks

closed 2018-07-21 by ascillato2