arendst/Tasmota #2849: google cloud MQTT support

upendrawww9, 2018-05-28:
Do you support Google Cloud MQTT service.??
Now we have only cloudmqtt support.

Regards
Upendra

**Have you look for this feature in other issues and in the wiki?**
YES
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-05-28:
As it only supports TLS 1.2 tasmota does not support it. Not enough resources available on a tasmota running esp8266.

closed 2018-06-08 by ascillato2

mikizvi, 2018-12-08:
Hi,

I found a way to bridge MQTT's where your local home Mosquitto can talk TLS 1.2 to the remote MQTT server.  Details are in the Kapppelt gBridge doc, referenced from [here](https://github.com/arendst/Sonoff-Tasmota/issues/2146#issuecomment-445438821).

Michael.