arendst/Tasmota #2853: Sonoff Dual R2 restarts after simple rule execution

MykhailoKh, 2018-05-29:
Hardware:
Sonoff Dual R2. Result of Status 0 is in attachment.
Firmware: https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff.bin
I've checked the same behaviour with current development also.

Rule:
Rule1 on Event#fanspeed do Power2 1; endon

Result:
Reboot with a reason: Hardware Watchdog
I see that Power2 was triggered and reboot happens after rule completed.

Useful info:
[log.from.server.txt](https://github.com/arendst/Sonoff-Tasmota/files/2046828/log.from.server.txt)
[result.txt](https://github.com/arendst/Sonoff-Tasmota/files/2046829/result.txt)
[rule.1.txt](https://github.com/arendst/Sonoff-Tasmota/files/2046830/rule.1.txt)
[status.0.txt](https://github.com/arendst/Sonoff-Tasmota/files/2046831/status.0.txt)

arendst, 2018-05-29:
Confirmed. Investigating...

arendst, 2018-05-29:
Thnx for the useful info! Fixed in just released version.

MykhailoKh, 2018-05-29:
Thank you very much! It works as expected.