arendst/Tasmota #2854: Limitation of number of RF Remotes

iotomic, 2018-05-29:
Hi,

The standard RF Bridge is limited to 4 remotes , regardless of number of buttons configured. Does the same apply to Tasmota? 

Working on a concept where I need to maximise number of remotes for a single bridge / transceiver. RFXCOM is good but too expensive. Was hoping to make use of RF Bridge.

yoh-there, 2018-05-29:
The bridge will MQTT-publish the raw data so any "automation hub" (I use Node-RED) can pick that up and process it to whatever command you like. Decoding the raw data might not be what you want, but it does give you endless flexibility / button count.

A limitation I noticed in the bridge coprocessor firmware (has nothing to do with Tasmota) is that it is quite picky what remotes it will pick up. Then again, the price is a killer and it is a neat package. I run it with remotes it understands and Tasmota and I am extremely content with it.

iotomic, 2018-05-29:
I was more interested in how many Sonoff remotes it would work with. It mentions 4.

ascillato, 2018-05-29:
If you want to **send** by RF the learned codes, it stores 16 RF codes.
If you want to process the RF codes **received**, you can use rules (see wiki) and you have only the limit of 1.5Kbytes for rules (so, a lot of rf codes). Also you can resend them using MQTT to a home automation system to process them there.

Jason2866, 2018-05-31:
If you use a HA system you are not limited in any way. The code learning and receiving is done there.
Tasmota sends the received data via mqtt to the HA system. From the HA sytem you can send every
24 bit Rf-code via mqtt with the RF Bridge.

ascillato, 2018-06-04:
Hi

If your issue is solved, please close it. Thanks :+1: