arendst/Tasmota #2856: H801 Flashing RGB strip.

jelee1988, 2018-05-29:
Got some trouble selecting colors on the RGB template, the colors Red, Green, Blue and the effects are working fine on the smd5050 strip.

But when i select custom colors in HomeAssistant (The light colors for example, light blue, Yellow, or other colors on the inside of the RGB template, the rgb strip is going flashing, and then H801 a custom color, reconnect the strip.

What is causing the flashing and what is possibly a solution?

Status 0
```
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS = {"Status":{"Module":20,"FriendlyName":["Sonoff"],"Topic":"Woonkamer/RgbSnoer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:27","StartupUTC":"2018-05-29T11:45:28","Sleep":0,"BootCount":41,"SaveCount":149,"SaveAddress":"F7000"}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS2 = {"StatusFWR":{"Version":"5.14.0a","BuildDateTime":"2018-05-29T00:27:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["test123",""],"TelePeriod":120,"SetOption":["00008029","55818000"]}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","17B61798","00000000"]}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS5 = {"StatusNET":{"Hostname":"Woonkamer/RgbSnoer-5219","IPAddress":"192.168.1.3","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:6F:94:63","Webserver":2,"WifiConfig":3}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6F9463","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS7 = {"StatusTIM":{"UTC":"Tue May 29 11:45:55 2018","Local":"Tue May 29 12:45:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:53","Sunset":"20:41"}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS10 = {"StatusSNS":{"Time":"2018-05-29T12:45:55"}}
12:45:55 MQT: stat/Woonkamer/RgbSnoer/STATUS11 = {"StatusSTS":{"Time":"2018-05-29T12:45:55","Uptime":"0T00:00:27","Vcc":3.172,"POWER":"OFF","Dimmer":100,"Color":"00FFBF0000","HSBColor":"165,100,100","Channel":[0,100,75,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"test123","RSSI":56,"APMac":"10:7B:EF:B9:F5:11"}}}
```

ascillato, 2018-05-29:
Hi,

Seems to be a home assistant configuration issue. Please share the that part of you configuration.yaml file.

I have the led strip working fine using the same template that is on the wiki.

jelee1988, 2018-05-29:
configuration.yaml
https://pastebin.com/vdYKd7tV 

tried the H801 under 2 different modules; H801 and MagicHome. on multiple configurations.
But still no luck

ascillato, 2018-05-29:
Hi,

In your configuration for HomeAssistant you have 

`rgb_command_topic: "cmnd/Woonkamer/RgbSnoer/Color"`

and should be 

`rgb_command_topic: "cmnd/Woonkamer/RgbSnoer/Color2"`

so as to change the color using the actual dimmer value. The first configuration set the color and also the dimmer to 100

> But when i select custom colors in HomeAssistant (The light colors for example, light blue, Yellow, or other colors on the inside of the RGB template, the rgb strip is going flashing,

Flashing? turn on and off very quickly? in which color?

> and then H801 a custom color, reconnect the strip.

Sorry, I don't understand the phrase. What do you meant?

jelee1988, 2018-05-29:
Hi, 
THANKS  for your reply after changing color to color2 the problem was solved... 

It looked like some shade or purple, when I used the white channel it was blinking simultaneously.

It seemed as if the H801 got stuck, when I disconnected the rgb channels then the H801 resumed, then I selected "color 3" in the terminal and re-connected the rgb. and everything worked everything again.

```
- platform: mqtt
  name: "Woonkamer RGB Snoer"
  state_topic: "stat/Woonkamer/RgbSnoer/POWER"
  command_topic: "cmnd/Woonkamer/RgbSnoer/POWER"
  brightness_state_topic: "stat/Woonkamer/RgbSnoer/Dimmer"
  brightness_command_topic: "cmnd/Woonkamer/RgbSnoer/Dimmer"
  rgb_state_topic: "stat/Woonkamer/RgbSnoer/Color"
  rgb_command_topic: "cmnd/Woonkamer/RgbSnoer/Color2"
  brightness_value_template: "{{ value_json.Dimmer }}"
  rgb_command_template: "{{ '#%02x%02x%02x%02x%02x' | format(green, red, blue, 00, 00)}}"
  rgb_command_mode: hex
  brightness_scale: 100
  color_temp_command_topic: "cmnd/Woonkamer/RgbSnoer/CT"
  qos: 0
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
```

ascillato, 2018-05-29:
Great!

Happy to know that it is working for you.

Please, close this issue. Thanks :+1:

closed 2018-05-29 by jelee1988