arendst/Tasmota #2857: keep the status ON (or OFF) after a pulse

pakgio, 2018-05-29:
Hello everyone,

Writing here is my last option to get this thing solved.

Setting: Homeassistant + sonoff 4ch pro with tasmota + itching relay (first pulse turn ON/ second pulse, turn  OFF).

I can controll correctly the relay using the sonoff 4ch.
Homeassistant see and controls correctly all the 4 switches of the sonoff 4ch pro.

The point is: 
- when i give the pulse from the sonoff to the relay, the relay  turn ON (correctly) and home assistant see correctly the pulse ON/OFF on the sonoff.  Formally everything is working. In this way the status of the relay (and hece the bulb controlled by the relay) is not syncronized with homeassistant.

I want to give the pulse from the sonoff to the relay, but home assistant must keep the same status o the relay. 

Is there a way using MQTT to do this?

many thanks!

yoh-there, 2018-05-29:
Tasmota always send the changed state using stat/.../POWERx as topic. I know nothing about Homeassistant, but to track the state it needs to process this MQTT message.

I would advise to tell MQTT to retain the stat message. This way, any subscriber to the stat message always gets the current state when it connects. This is done using cmnd/.../POWERRETAIN ON.

Hope this helps

pakgio, 2018-05-29:
thanks yon-there! I'll give it a try for sure.

pakgio, 2018-05-30:
what about a virtual switch? 
how can I make communicate 2 switch using mqtt? 
I found out I can create as many virtual switch on the Home assistant. Those vrtual switch can be activated using mqtt (exaclty like a real switch).

I could use the virtual switch so when I turn it ON it send a command -may be toggle? - to the real sonoff switch that will pulse and activate the relay and vice versa (when I turn OFF the virtual switch it will send to the real sonoff switch a command to make it puls and turn off the relay).
Is it possible to do the same on the other side? so that when the real sonoff is toggled it will send a message to the virtual switch that will be turned on.

still meditating...but need some mqtt help.

yoh-there, 2018-05-30:
I think (but I am not sure) the problem you face is not so much MQTT, but it is using the reed switch as a controlling input for the relay. It seems to me you want the reed switch not do do anything with the relay, just use it as a feedback to your home automation hub and/or dashboard.

Probably the best way to address this is to configure the reed switch as switch2. To make that really work you might need to use "Wemos" as board, but I am not sure about that. I don't have a SV but can figure out what happens on a sonoff basic later if you are unable to do so yourself.

As for basic MQTT, please note that tools like mosquitto_sub and mosquitto_pub are indispensible. While MQTT is extremely simple, I took me a while too to wrap my head around things like the concept of retention, it's case sensitivity, etc etc. I found https://www.hivemq.com/blog/mqtt-essentials-part-1-introducing-mqtt a good starting point.

yoh-there, 2018-05-30:
Sorry, I was mixing up two different questions, D'uh. Forget everything I wrote except indeed the MQTT stuff.

ascillato, 2018-06-13:
Hi,

If your issue is solved, please close it. Thanks! :+1:

pakgio, 2018-06-13:
well...the issue is not closed...still trying to find a solution.

closed 2018-08-12 by stale[bot]