arendst/Tasmota #2858: Sonoff SV - If No MQTT Broker connection, reed switch causes the relay to fire.

rmarkle, 2018-05-29:
Hi!

I have a SONOFF SV connected to my garage door opener.  It has a reed switch on it to determine if the door is open or closed.  The reed switch is "closed" when the garage door is open.

If my sonoff SV looses connection to the MQTT broker ( which has been happening frequently, lately) and I manually open the garage door using the door opener's RF controller or the toggle switch on the sonoff, once the door hits the reed switch, the Sonoff automatically fires the the relay on.

Here is the logs from when I hit the toggle switch on the SONOFF.  There is a 1 sec PULSETIME set.
15:41:08 RSL: stat/sonoff2/RESULT = {"POWER":"ON"}
15:41:08 RSL: stat/sonoff2/POWER = ON
15:41:10 RSL: stat/sonoff2/RESULT = {"POWER":"OFF"}
15:41:10 RSL: stat/sonoff2/POWER = OFF
15:41:11 MQT: Attempting connection...
15:41:11 MQT: Connect failed to 192.168.1.247:1883, rc -2. Retry in 10 sec
15:41:22 RSL: stat/sonoff2/RESULT = {"POWER":"ON"} <-- Hits the reed switch here and it turns on the relay and closes the garage door.
15:41:22 RSL: stat/sonoff2/POWER = ON
15:41:22 MQT: Attempting connection...
15:41:22 MQT: Connect failed to 192.168.1.247:1883, rc -2. Retry in 10 sec
15:41:23 RSL: stat/sonoff2/RESULT = {"POWER":"OFF"}
15:41:23 RSL: stat/sonoff2/POWER = OFF
15:41:33 MQT: Attempting connection...

Any help?

yoh-there, 2018-05-30:
I think (but I am not sure) the problem you face is it is using the reed switch as a controlling input for the relay. It seems to me you want the reed switch not do do anything with the relay, just use it as a feedback to your home automation hub and/or dashboard.

Probably the best way to address this is to configure the reed switch as switch2. To make that really work you might need to use "Wemos" as board, but I am not sure about that. I don't have a SV but can figure out what happens on a sonoff basic later if you are unable to do so yourself.

ascillato, 2018-05-30:
Set to any unused pin as relay2, so the switch2 will toggle that one instead of relay1.

Frogmore42, 2018-05-30:
I would recommend you provide all the requested information from the template, especially status 0.  It looks like your device is having connection issues.  This might be because of the core that you are using.  It might or might not help the issue with the reed switch, but the rate of disconnection in the log indicates an issue.

rmarkle, 2018-05-30:
Setting the unused pins as relay2 seems to work.  

The MQTT connection issue is on the broker side, specifically HASS.IO is crashing frequently.  I'm working on that part.  I was more concerned about the Sonoff firing the relay if there wasn't any MQTT connection.

ascillato, 2018-05-30:
Happy to know that it is working for you. If your issue is solved, please close it.

For the next time, please use the templates as @Frogmore42 has explained. It makes a lot easier to solve issues. Thanks :+1:

rmarkle, 2018-05-30:
This issue is closed :-)

closed 2018-05-30 by rmarkle