arendst/Tasmota #2860: Wemos D1 mini: No NTP, constant clock resets

titomontes, 2018-05-29:
Hello again

I just flashed a Wemos D1 mini (still trying to figure out mqtt for reading motion sensor/luminance/temp/humidity). Problem is, it doesn´t get the time from the time servers provided. Every reboot resetd the clock, also.

> 00:19:55 MQT: stat/wemos2/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos2"],"Topic":"wemos2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
> 00:19:55 MQT: stat/wemos2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"wemoss","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T00:18:41","StartupUTC":"","Sleep":0,"BootCount":6,"SaveCount":18,"SaveAddress":"F9000"}}
> 00:19:55 MQT: stat/wemos2/STATUS2 = {"StatusFWR":{"Version":"5.14.0b","BuildDateTime":"2018-05-29T18:13:03","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
> 00:19:55 MQT: stat/wemos2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Montes01",""],"TelePeriod":300,"SetOption":["000280E9","55818000"]}}
> 00:19:55 MQT: stat/wemos2/STATUS4 = {"StatusMEM":{"ProgramSize":542,"Free":460,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17BE179A","00000000"]}}
> 00:19:55 MQT: stat/wemos2/STATUS5 = {"StatusNET":{"Hostname":"wemos2-5350","IPAddress":"192.168.1.49","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"68:C6:3A:CA:74:E6","Webserver":2,"WifiConfig":3}}
> 00:19:55 MQT: stat/wemos2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA74E6","MqttUser":"tito1","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
> 00:19:55 MQT: stat/wemos2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:19:55 1970","Local":"Thu Jan 01 00:19:55 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:38","Sunset":"16:58"}}
> 00:19:55 MQT: stat/wemos2/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:19:55","TSL2561":{"Illuminance":7.265}}}
> 00:19:55 MQT: stat/wemos2/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:19:55","Uptime":"0T00:18:41","Vcc":2.986,"Wifi":{"AP":1,"SSId":"Montes01","RSSI":68,"APMac":"34:97:F6:61:67:C0"}}}

Frogmore42, 2018-05-30:
It looks like you have MQTT set up.  You probably are having issues with core 2.4.0 like many others.  I would suggest recompiling with core 2.3.0 and see if that makes things better.

titomontes, 2018-05-31:
Yes , but the bug supposedly affects all boards, not just de wemos. I have another 10 sonoffs that work flawlessly with the correct time, so I wonder if this is a wemos related bug.

Frogmore42, 2018-05-31:
If you search the issues, I am sure you will find lots of examples where ONLY particular boards had an issue with 2.4.x.  If it is working for you great, use it.  If you are having a wired issue, one of the first things to try is change to 2.3.0.

Your "Heap":13 indicates this device is probably suffering from the issue.  That is a pretty low value for heap.

ascillato, 2018-06-13:
Hi,

Could you manage to solve it?

titomontes, 2018-06-15:
Nope, it's still an issue. I don't see how only the NTP time can be affected by the issue, Also, I have no clue on how to compile in another version...

Frogmore42, 2018-06-15:
See this issue for help with changing the core: https://github.com/arendst/Sonoff-Tasmota/issues/2689

If you are happy with the performance of what you have, please stay with it.  If not, the next thing to try is to use a core that is known to not have issues with WiFi connectivity (instead of one that is known to have strange issues sometimes).

The 2.4.x cores have caused a lot of people a lot of grief, but they have also worked flawlessly for a lot of people.  The usage of it might or might not be causing the issue you are seeing.  No one will know until you choose to try it.  Many others were surprised when it fixed their issue, maybe you will be too.

ascillato2, 2018-07-10:
Hi,

Please, try last release to see if solves your issue. Thanks

ascillato2, 2018-07-16:
Hi

Seems that you are using fixed IP. If that is the case, you need to check if your gateway and dns configuration are ok. If not, Tasmota can not connect to internet to sync Time with NTP Servers. 

If you managed to solve your issue, please close it. Thanks

ascillato, 2018-08-18:
Hi, any news on this issue?

ascillato2, 2018-08-22:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-22 by ascillato2