arendst/Tasmota #2862: How to install another firmware?

7h30n3, 2018-05-29:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : sonoff T1 R2 2CH EU 
- [ ] Provide the output of command ``status 0`` :

```
22:37:29 RSL: STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:37:29 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:11:48","StartupUTC":"2018-05-29T21:25:41","Sleep":0,"BootCount":30,"SaveCount":89,"SaveAddress":"FB000"}}
22:37:29 RSL: STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018.05.15 15:34:22","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:37:29 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wireless2",""],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
22:37:29 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
22:37:29 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-6835","IPAddress":"192.168.1.233","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:CA:7A:B3","Webserver":2,"WifiConfig":2}}
22:37:29 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue May 29 21:37:29 2018","Local":"Tue May 29 22:37:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:53","Sunset":"20:41"}}
22:37:29 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-05-29T22:37:29"}}
22:37:29 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-05-29T22:37:29","Uptime":"0T00:11:48","Vcc":3.279,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"wirelessnetwork","RSSI":82,"APMac":"B0:39:56:55:A1:99"}}}
```

I have some Issues with my sonoff and want to rule out a problem with tasmota, so I want to install: [](https://github.com/KmanOz/KmanSonoff ) .Therefore I need to get into flash mode.

How can I achive that?

Frogmore42, 2018-05-30:
Read the wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation

curzon01, 2018-05-30:
:confused: where do you get `Status 0` output if you want to know to install Tasmota on T1 R2?

>  Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)

your device is described within [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU):
[Sonoff T1 R2 EU Flashing](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU#flashing)

7h30n3, 2018-05-30:
@curzon01 I already have tasmota installed, but I thought that there is something wrong with this software, so I wanted to test another firmware, as mentioned above.

But nevermind. As it turns out I just had this problem: 

> If powered only with 3.3 V from debug connector long press will be detected by Tasmota on button 1

[https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU#known-issues-please-read](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU#known-issues-please-read)

ascillato, 2018-05-30:
Hi.

If your issue is solved, please close it. Thanks.