arendst/Tasmota #2865: Wall light switch on GPOI-0 (the button)

sparkgap, 2018-05-30:
I have RPI running openHAB and Mosquitto MQTT.  I have flashed several Sonoff Basics with Sonoff-Tasmota which are working fine using openHAB Rules.      I would like to add a standard USA wall light switch to override Rules for manual control. 

Prior to using Sonoff-Tasmota I wrote an Arduino C+ program for the sonoffs using HTTP Post commands issued from openHAB Rules.    To connect my light switch I soldered fine wires to the push button.   My C+ program looked at the button (GPOI-0) and controlled sonoff (locally) independant of openHAB.  In other words, if the RPI openHAB/wifi failed I could still control my lights.

How could this be implemented with Sonoff-Tasmota?   _________________ 

One solution would be to insert my C+ code somewhere in the Sonoff-Tasmota Void Loop.   If possible, where would the insertion point be?  __________________________________

As a last resort, I could connect my light switch to GPIO-0 or GPIO-14 and poll it with an openHAB Rule, but this would not provide redundancy in the event of wifi failure.   If I would implement this method, what would my MQTT topic look like look like?  ______________________________________

You might wonder why change if I had something working.   Well...... The ESP8266 locks-up about once a week, somehow in the WiFI library and don't seem to have a watch dog timer recovery feature.   All sonoffs do the same and even NodeMCU-ESP8266 modules running same code do the same.... lock-up.   I was hoping that Sonoff-Tasmota would be more reliable and not lock-up.

yoh-there, 2018-05-30:
I think you will be fine changing your module to a Wemos, then configure GPIO0 as button 1.

curzon01, 2018-05-30:
Be aware using [GPIO0 for button input](https://github.com/arendst/Sonoff-Tasmota/issues/2260#issuecomment-376070623) is not a good idea.

But anyway just wondering 
- why not using wires in parallel to the build in button
- why your are using additonal HTTP posts when you are already having a MQTT server running

The button on a Sonoff should be always usable as a manual switch. If this interferes yur logic, think about that.

curzon01, 2018-05-30:
Just wondering 
- why not using wires in parallel to the build in button
- why your are using additonal HTTP posts when you are already having a MQTT server running

The button on a Sonoff should be always usable as a manual switch. If this interferes your logic, think about that.

digiblur, 2018-05-30:
Not sure I get this post?  Just connect the light switch to GPIO14 on the Sonoff and make sure the options are set for GPIO14 as switch to change the state of the light anytime the state of GPIO14 changes.  I have several Tasmota devices like this in my home and they work great all through MQTT and never lock up.  If my HomeAssistant is down you can still control the lights manually.

Frogmore42, 2018-05-30:
Don't use gpio 0 with a switch (button is okay) If there is a power failure while the switch is closed the device will be in bootloader mode and not work.

sparkgap, 2018-05-30:
Digiblur - thanks
Now I  understand why to not use GPIO-0

On your previous post.......  where (and how) do I set options for GPIO-14? 
In the user_config.h ?????
or 
browse to the sonoff web based configuration interface   ????

In either case I don't see an option to make the relay follow GPIO-14

digiblur, 2018-05-30:
I believe you can force those, but just use the web interface.  Start on this video around 45 seconds: https://youtu.be/ab472a40-co?t=47s

I highly recommend a 10k pull up resistor though.  It fixed a couple ghost switchers for me.  
https://github.com/arendst/Sonoff-Tasmota/issues/665

sparkgap, 2018-05-31:
Thanks digiblur ........... Problem resolved:
In the web console under module paramaters, I set GPIO-14 to   "09 Switch1".   I was looking for this feature in the user_config.h , but I don't believe it can be set there.  I also had user_config.h switch mode messed up thinking it configured the onboard button..... but I set it back to  FOLLOW (inverted), so when my wall switch when in ON position , GPIO-14 is grounded.  Yes I'll add the 10K pullup.   When the programming header plug is removed, I'll connect 10K and my wall switch using different header plug..... this way there will be no need to solder in the field if I need to remove or replsace the sonoff.

In the video you recommended I learned that the web console can be used to change Switch Mode.   I could not find that in the wikis..... any suggestions where I can find those commands exactly as I would type them  ?????   So many tines this wiki stuff is obscured with syntax formats that I do not understand or parts of the command missing assuming I understand the full syntax.

closed 2018-05-31 by sparkgap

digiblur, 2018-05-31:
The Switchmode commands are described pretty well in the wiki here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Looking at the commands if it works like the pulsetime command I was playing with the other day, the <x> refers to the switch number.  So "switchmode1 1" would set switch 1 to Follow = ON.