arendst/Tasmota #2866: Split Interlock mode in groups

MadDoct, 2018-05-30:
There have been similar feature requests but they were specific to some hardware (sonoff 4ch).
I'd like for an option, preferably in the form of SetOptionXX, to allow interlocking in groups in any hardware with more than 3 relays. If the hardware (even moddified hardware) had 4 relays there would be 2 groups, if it had 6 relays, there would be 3 groups...

meingraham, 2018-07-21:
This would be much more elegant, flexible, and expandable than implementing it with rules (for which there are but a limited number of three).

https://github.com/arendst/Sonoff-Tasmota/issues/3204#issuecomment-406158303

ascillato2, 2018-08-23:
@meingraham You can concatenate the rules. The 3 slots are there but each rule that start with ON is independent from each other. So you can add more triggers if you want.