arendst/Tasmota #2867: APP_LEDSTATE LED_OFF not working for Sonoff Basic?

reibuehl, 2018-05-30:
**Describe the bug**

I did set #define APP_LEDSTATE LED_OFF (among other settings) in user_config_override.h. In the resulting binary, all custom settings are present as defined but the LED on the Sonoff Basic still switches on and off with the power relay. Is APP_LEDSTATE expect to not have an effect on the LED of the Sonoff Basic (eg. due to hardwired LED in that device) or should it work? 

- [X] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff Basic
- [X] Provide the output of command ``status 0`` :
`10:22:18 CMD: status 0
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff 01 OG_SZ_Nachtlicht"],"Topic":"OG_SZ_Nachtlicht","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T01:08:12","StartupUTC":"2018-05-30T08:14:06","Sleep":0,"BootCount":68,"SaveCount":151,"SaveAddress":"FB000"}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS2 = {"StatusFWR":{"Version":"5.14.0b","BuildDateTime":"2018-05-30T10:05:32","Boot":4,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["mySSID","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","00018000"]}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS4 = {"StatusMEM":{"ProgramSize":542,"Free":460,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS5 = {"StatusNET":{"Hostname":"sonoff01","IPAddress":"192.168.1.87","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:XX:XX:XX:XX","Webserver":2,"WifiConfig":2}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_96EFC3","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS7 = {"StatusTIM":{"UTC":"Wed May 30 09:22:18 2018","Local":"Wed May 30 10:22:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:52","Sunset":"20:43"}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS10 = {"StatusSNS":{"Time":"2018-05-30T10:22:18"}}
10:22:18 MQT: /SmartHome/Obergeschoss/OG_SZ_Nachtlicht/STATUS11 = {"StatusSTS":{"Time":"2018-05-30T10:22:18","Uptime":"0T01:08:12","Vcc":3.506,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mySSID","RSSI":100,"APMac":"14:CC:XX:XX:XX:XX"}}}`

**Steps to reproduce the behavior:**
Add the following to user_config_override.h, compile and flash to a Sonoff Basic:
`#undef  APP_LEDSTATE`
`#define APP_LEDSTATE           LED_OFF`

**Expected behavior**
LED of the Sonoff Basic should not switch on if the relay is switched on.

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2018-05-30:
Looks like your LED state is still set to 1 according to your data posted.  Go into the console on the web GUI and send over LedState 0

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

reibuehl, 2018-05-30:
After setting LedState 0 on the console, the LED behaves like it should. Any idea why the APP_LEDSTATE setting in user_config_override.h was ignored?

digiblur, 2018-05-30:
I usually set mine in the user_config.h when I compile things.  I'm not sure of the order of things on what is loaded when, but I seem to recall if I had a previous setting such as Ledstate 1 setup on the device and flashed a firmware with the ledstate set to 0 it wouldn't override the setting unless of course I did factory reset of the saved settings.

reibuehl, 2018-05-30:
Ah, okay, that could be the reason! The Sonoff Basic I used was not a first time install but was upgraded from Tasmota 4.0.8. Most probably the value was set by that version and the upgrade did not override it since it was already set. Thanks for the clarification!