arendst/Tasmota #2871: Real Time Clock

Lawrencezarb, 2018-05-31:
From my understanding the Timer function relies on the Sonoff unit have access to a NTP server.

If communication is lost to the NTP or the Wifi I assume that the Timers will cease to function as the Sonoff will not know what time it is.

Again I assume the Sonoff has no inbuilt clock to hold the time once it gets its Time from the NTP.

If this is correct would it be possible to add a Real Time Clock board to the Sonoff as used in a Raspberry pi?

ascillato, 2018-05-31:
> From my understanding the Timer function relies on the Sonoff unit have access to a NTP server.

Yes

> If communication is lost to the NTP or the Wifi I assume that the Timers will cease to function as the Sonoff will not know what time it is.
> Again I assume the Sonoff has no inbuilt clock to hold the time once it gets its Time from the NTP.

If the wifi or internet is lost **after** Tasmota sync the time from internet, the timers will continue working. The ESP8266 will continue counting the time. It has the internal hardware to do that. The problem is that the Clock in the ESP8266 is not backed-up with a battery, so with a reset or a power loss, it will forget the time.
If Tasmota boots and can not reach a Time server, the timers will never trigger. Because, Tasmota don't know the time at startup.

> If this is correct would it be possible to add a Real Time Clock board to the Sonoff as used in a Raspberry pi?

Raspberry Pi don't have a builtin Real Time Clock either. It has the same problem that the ESP8266. Can keep track of the time but it is not battery backed-up when losing power. You can add a RTC to a Raspberry as explained in this [link](https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/overview)

About the RTC on ESP8266 there is an article in this [link](http://www.instructables.com/id/TESTED-Timekeeping-on-ESP8266-Arduino-Uno-WITHOUT-/) that explains the ESP8266 precision for the Time.

And at this moment Tasmota does not have support for an external RTC hardware for reading and saving the time backed by a battery and to be used when there is no Wifi or Internet Connection.

Lawrencezarb, 2018-05-31:
Thanks for the information and clarificaton. With regards to my Raspberry Pi RTC, I was referring the an external RTC module that can be added to the Pi. 

It would be interesting to see this implemented.

trogper, 2018-06-03:
You could set up a NTP server on your Raspi, so the ESP would get it's time from there.

fritserasmus, 2018-06-17:
> 

You could set up a NTP server on your Raspi, so the ESP would get it's time from there.
--

trogper .. That would be awesome if the SOnOff can read the NTP from my MQTT Server.

Do you perhaps have a link that I could use for set-up procedure?

trogper, 2018-06-19:
I have raspbian jessie on my raspberry, it has ntp server installed by default, so you probably only have to set the sonoff to use raspberry as the ntp source.
[https://github.com/arendst/Sonoff-Tasmota/wiki/Commands](Commands reference)

Kaimane, 2018-06-25:
[Feature] Support for RTC DS3231

In one of my future projects I would like to use an Tasmota flashed ESP8266 for switching some relays by the tasmota timer feature. Nothing special ... But in the productive environment the ESP8266 can't reach my wi-fi for syncing with a NTP after a power failure. The AP would be too far away. At this point my configured timers would be useless.
To keep the time "in sync" without an established network connection a DS3231 for example could be a cheap, stable and easy to integrate solution. In my mind a simple and rudimentary integration in Tasmota would be desirable:
If Tasmota has received the current time via NTP (in case of setting up Tasmota) and a DS3231 is configured, the current time is automatically stored in the DS3231.
As soon as Tasmota fails to synchronize time with an NTP after a power failure for example (at this point no access point is within range of the ESP8266), Tasmota gets the time from the DS3231 easily.

So, to get timers working after a power failure without tasmota is connected to a ntp (local or worldwide) a DS3231 (or any other rtc) extension for Tasmota would be very useful I think.
What do you think?

ascillato, 2018-06-25:
Hi,

I tested as @trogper has explained and works really good.

I type in the console:

```
ntpserver1 192.168.1.117
ntpserver2 192.168.1.117
ntpserver3 192.168.1.117
```

where 192.168.1.117 is the IP of my raspberry pi running **Raspbian Stretch**. After restarting Tasmota it takes the time from my raspberry. So, if my raspberry has a RTC DS3231 attached, on an internet disconnection, the timers from Tasmota will work.

I tried also a raspberry pi running **Hassio** but does not work. Hassio seems not to support NTP as RASPBIAN.

@Kaimane your point is good, may be sometime in the future we will have an extension to add that, but remember that now Tasmota is mainly a firmware to add MQTT to sonoff devices for home automation, not a stand alone device.

trogper, 2018-06-26:
@ascillato I haven't usede hassio yet, but I suppose you do have SSH access to it, so you could just install the NTP package. In debian it is called just ntp

ascillato, 2018-06-26:
I have just found (5 minutes ago) a NTP add on for hassio. I will test it and if it works I will post it here.

ascillato, 2018-06-26:
I have tested the [NTP server addon for hassio](https://github.com/ascillato/hass-io-addons) and works really good. Tasmota can retrieve the time correctly from there. I made also the request to be added in the community hassio addons.

HobbytronicsPK, 2018-07-08:
I have problem with time update with Sonoff Tasmota 6.1.0. I am using **0.pool.ntp.org** as NTP Server. Time has been updated to 12:46:31 on Tasmota console which is 1 hour greater. It should be 11:46:31 as current time on my PC is 4:46 PM. I am in Pakistan GMT+5 zone.

HobbytronicsPK, 2018-07-08:
I have found the solution so though should mention it. Timezone was not correct. Changing time zone by `timezone +5` corrected the time output.

guyelg, 2018-10-07:
Hi
I want to use Tasmota in  a place without WIFI, but still enjoy the benefit of timers.
Is it possible to use external RTC like DS3231 ?
When WIFI is available then the switch will update the time/date in the external RTC  , and when there is no WIFI, after reboot the switch use meanwhile the external RTC as time referance.
Also , the external RTC is more accurate then the internal RTC (can be 4 min drift in a day in the internal )

andrethomas, 2018-10-07:
Currently, there is no driver added for any external RTC, but this is planned for future.

DS3231 is a good candidate.

guyelg, 2018-10-07:
Thanks for your answer.
Is there any document that explain the way to write driver ?
I will be happy to help with that.

andrethomas, 2018-10-07:
You can take a peek at https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API which provides the basic information. For the rest there's always the existing xsns_xx_xx.ino files that can be used as examples... there is also a  Tasmota discord chat channel which is useful for development ideas also.

ascillato2, 2018-10-10:
Added driver from @guyelg

Thanks.

closed 2018-10-10 by ascillato2

matteo2019, 2019-06-29:
Hello. I have read your conversations, I would like to know if today the ds3231 can be connected directly to the sonoff.

matteo2019, 2019-07-01:
if there is still no update, tell me the best method to take time from rtc 3231.
I don't have an internet connection where sonoff esp8266 works, when the electric network is lost the timers don't work because the sonoff loses the time calculation.

matteo2019, 2019-07-08:
can I please receive an answer? I need you

andrethomas, 2019-07-08:
@matteo2019 https://github.com/arendst/Sonoff-Tasmota/wiki/DS3231---External-RTC ?

matteo2019, 2019-07-08:
after this step what should I do on the sonoff?

Ottieni Outlook per Android<https://aka.ms/ghei36>

matteo2019, 2019-07-08:
I have to use the sonoff where there is no internet connection. It simply needs to work with time timers. When the mains power off the sonoff loses the time calculation, it does not remember the time. You need a Wemos d1 mini like you recommended. driving is clear but not complete. After installing tasmota on the wemos and connecting the RTC 3231 how should we proceed on the sonoff?

andrethomas, 2019-07-08:
In theory...

1. You need to compile a binary which includes the driver directive as noted in the wiki.
2. The RTC needs at least to be connected once to get the time after which it should keep time using its onboard battery (depending on the RTC board you have it should have a coin cell on it which will make it keep time)
3. If you plan to use the RTC on a device which has no internet connectivity at all you will need to first configure it on a wemos d1 mini or equivalent so that it can at least get the correct time once.
4. On the sonoff basic the only option is to use the TX/RX pins for SCL/SDA I2C needed to connect to the RTC module but you will need to make sure serial logging is disabled so that the pins are not inadvertently flooded with serial data.

This is rather off-label use though - having an RTC exclusively provide time where there is no internet connection so there are other risks involved such as the RTC losing some seconds over a long period and of course the fact that the coin cell will not last forever.

matteo2019, 2019-07-08:
is there anyone so good and willing to do a guide or tutorial? many of us ask for it

matteo2019, 2019-07-08:
however, we thank all those who are responding

matteo2019, 2019-07-10:
someone very kind could make a scheme of how to connect the wemos to the sonoff and how to configure them?

lalo-uy, 2019-07-10:
You use Tasmota in a Wemos or in a Sonoff
You want to connect a RTC to Sonoff ? Do it on TX/RX , previously
configured as SDI, SCK.

matteo2019, 2019-07-10:
please don't confuse my ideas.  As explained above, I want the sonoff to calculate the time in the absence of the internet and I want it not to lose the time before the power failure.  You advised me to flash the tasmota firmware on wemos D1 mini and connect the wemos to a 3231 rtc. I would like to know the next steps.

matteo2019, 2019-07-10:
- first step flash of the dual r2 sonoff with Tasmota firmware.  
 flash wemos d1 mini firmware tasmota.  
- connect rtc to wemos d1 mini

the other passages what are they?
I need to know specifically the steps to follow

ascillato, 2019-07-10:
@matteo2019 

1- Compile your Tasmota firmware with DS3231 enabled
2- Flash the device you want with the Firmware you have compiled
3- configure the GPIOS as explained in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/DS3231---External-RTC)
  D2 GPIO04 : 06 I2C SDA
  D1 GPIO05 : 05 I2C SCL
4- connect your DS3231 to those pins and also connect GND
5- power your device with internet connection
6- check the logs on the console (there will say if the time is saved inside your RTC)
7- power down and test it without internet connection. It should say in the console that it takes the time from the RTC.

matteo2019, 2019-07-11:
@ascillato

Thank you Ascillato for your reply.  Up to this point everything is clear.  This you described is the configuration on the wemos d1 mini and the connection with the wemos to the rtc ds3231.  the next step to connect the sonoff to the wemos what is it?

matteo2019, 2019-07-11:
@ascillato 
the sonoff (in my case sonoff dual r2 with tasmota) how should I connect it to the wemos?  in wifi or with cables?  can you describe the procedure in detail?

ascillato, 2019-07-11:
> the next step to connect the sonoff to the wemos what is it?

why you want to connect a sonoff to a wemos?

ascillato, 2019-07-11:
You ask how to connect a RTC to a wemos, so for that wemos have time and be used in a wifi network that does not have wifi. 

If you want a sonoff basic to have the time, just connect the RTC to that sonoff and forget your wemos

matteo2019, 2019-07-11:
@ascillato 
because the sonoff must take the time from the wemos to a sudden restart.  when there is no electricity supply

ascillato, 2019-07-11:
If you want a sonoff basic to have the time, just connect the RTC to that sonoff. You don't need a separated wemos for that.

ascillato, 2019-07-11:
it is a sonoff basic right?

matteo2019, 2019-07-11:
I repeat that my sonoff lives in a place with no internet.  Often there are losses of electrical network and its memory loses the time.  You advised me to configure a wemos D1 mini connected to a rtc ds3231.  After these steps how should I configure the sonoff?  the sonoff I think must be connected to the wemos.

matteo2019, 2019-07-11:
sonoff dual r2 flash tasmota stefanbode

ascillato, 2019-07-11:
You don't need a wemos. just connect your RTC to your sonoff dual. just that.

matteo2019, 2019-07-11:
@ascillato 
really???  you are the first to tell me this thing.  You're telling me that I rtc ds3231 can connect it directly to sonoff dual r2

ascillato, 2019-07-11:
yes, why not? it has free gpios to connect to.

As we are spamming this old issue I recommend you to address this to the Tasmota Support Chat.

matteo2019, 2019-07-11:
where do I connect ds3231 to sonoff?  the pin buttons 1 and the pin button 2 on the board I use them to connect 2 physical buttons to lower and raise the motor

ascillato2, 2019-07-11:
Do it on TX/RX , previously configured as SDI, SCK