arendst/Tasmota #2872: Calibrate temperature and humidity for DHT sensors

qingz2004, 2018-05-31:
Is there a way to calibrate temperature and humidity for DHT sensors?

**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Similar to power calibration

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-05-31:
Hello @qingz2004 
there is a big difference the pow has to be calibrated to show correct values and has
the hardware which support calibrating. None of the the DHT sensors supports calibrating features or support. If you not statisfied with the values you get, you have to do this with extern calibrating. This can be done (for example) with a Home Automation system. In my experience the the DHT sensors are not bad...
If you want it a little more accurate go and buy BMP280 or BME280 (Bosch) sensors. This sensors are supported by tasmota too.

qingz2004, 2018-05-31:
Thank you for the answer.
I understand the DHT sensors do not support calibrating features, but we can probably do it externally like ESPFormula from [ESPEasy](https://www.letscontrolit.com/wiki/index.php/EasyFormula) or at least to be able to set an offset.
I know some sensor like DHT11 has good resolution (16-bits ADC), but not very good accurate(+/2C).

Frogmore42, 2018-05-31:
Calibration is a complex subject.  Sensors can be off in many ways, eg zero point/offset, linearity, repeatability.  Humidity sensors are especially bad with the DHT11 ones likely not worth using, unless you are looking for a VERY gross indicator of humidity, i.e. is dry or wet.  They might be okay for that.

If you do a single point calibration and your sensor has very good repeatability, your results for measurements around that point will likely be better.  However, if the sensor has poor linearity and/or repeatability it might make it worse.

I have several BME280 and HTU21 and SHT15 sensors in my basement.  They all read slightly different, but I am not 100% sure if that is because they are not calibrated the same or the conditions really are that different.  I also have 4 pressure sensors.  I am pretty sure there is a slight offset on them, but it is well within the accuracy range specified for them, 1% of FS.  I could put some complex code in there to try and calibrate out any offset or non-linearity, but what would I use as my standard?

If you really want to calibrate your humidity sensors that is somewhat easier (but not really easy):
https://www.stevejenkins.com/blog/2014/06/how-to-calibrate-a-hygrometer-humidity-sensor-using-the-salt-test/ 
https://www.allaboutcircuits.com/projects/how-to-check-and-calibrate-a-humidity-sensor/
https://publiclab.org/notes/btbonval/07-14-2014/calibrating-dht11-sensors
http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

Since anything other than a single point offset correction is complex, it is probably best to handle the calibration in a layer above Tasmota.

andrethomas, 2018-06-05:
I've petty much abandoned the idea of using the DHT sensors completely... I wanted something to give accurate temperature report on regular interval for the purpose of climatic control automation via node-red using mqtt.

I had instances where replacing the DHT sensor in the same room with another sensor yielded completely differential results given the same operational conditions... especially with regard to humidity sensing but even temperature variations of up to 4 degree C were observed.

I switched to BMP180 and BMP280 (or is it BME280 I cannot recall now) which allowed me to implement climatic control with a hysteria level as low as 0.2 Degrees C which is pretty impressive - I don't particularly use the humidity reported by the sensors at this stage but I can also confirm what Frogmore42 indicates insofar that there are variations in humidity between different rooms in the house which I suppose is due to various other environmental factors such as airflow in a particular room, the temperature of that room and obviously how close it is to a source of water... I've also seen that changing the temperature of a room also increases the disparity of air pressure reported between different rooms and I think its important to remember that the air pressure and temperature readings coming from the bmp sensors may or may not be equivalent to outdoor readings with the same sensor as they are intended for altitude approximation and perhaps to some degree internally cross-compensate for that.

I've also tinkered with the LM75AD temperature sensors which are tiny, dirt cheap and locally available in my country and although I don't get humidity/air pressure from them they serve well for thermal measuring within 0.25 degree C accuracy from my experiments.

Frogmore42, 2018-06-06:
The BME280, HTU21, and SHT15 all have very high resolution for humidity even if they don't have that great absolute accuracy.  I never tried the DHT ones, because even their specs indicate they aren't that good.

However, while the ones I use are pretty accurate and have good resolution, they are not calibrated close enough to get useful absolute humidity measurements.  I was planning on using this to control the dehumidifier in my basement.  Unfortunately, the data does not support a good control signal.  So, instead I use a combination of relative humidity and time above set point.  That is probably similar to the I in PID, but simpler.

andrethomas, 2018-06-06:
Yeah, I do the same for temperature... if temp <= 21 it turns on heat... if temp >= 21.2 it turns off heating. Its impossible to set a < 21 and > 21 configuration with less hysteria as it will cause the relay to go on and off almost every few seconds due to small variances in the output of the temperature sensor.

With the DHT sensors I had to on at <21 and off at >23 to make it useful enough but that resulted in very high hysteria values from a PID controller perspective in my application.

Another important aspect which isn't present in a lot of breakout boards is that one needs to add good decoupling capacitors to give the ADC used internally by the sensors a very stable power source... as you say, given the high resolution even fluctuations of a few millivolts on the supply rail of the sensor will be having an impact on the output of the internal ADC's value given that the reference voltage is shifting along with the input power source.

I usually add a 100nF and 1uF on the VDD/VSS pins of the chip or breakout board to try and mitigate shifts in reference voltage.

One is starting to see more ADC type designs following the approach to creating internal reference voltage which is lower than the input voltage and more reliable to the internal ADC... a good example of this is Microchip's MCP342x ADC's - 2.048V ± 0.05% - This does, however, have a significant cost impact on the manufacturing of the chip and we won't see such enhancements in low-cost temperature humidity air pressure sensors anytime soon I don't think.

andrethomas, 2018-06-12:
For interest sake, here's my temperature controller data for one of the bedrooms using a bmp180
[image]

So yeah, post processing temperature data in something like node-red and making changes accordingly could be a potentially useful PID equivalent...

Avg temp and Hysteria is 30-minute cycles... as can be seen I have a ~0.1 deg C deviation in both directions, below and above the 30-minute average.

Also, note the instability of air pressure reading over the same period :)

Frogmore42, 2018-06-12:
Uhh, that's -/+ 0.06 Pascal over 30min.  I would say that is actually very stable.  Convert that to height difference in air and it comes out to about +/-5mm, so we'll within believable numbers.

Zoom in on your temperature scale and you will see some interesting things.

andrethomas, 2018-06-13:
Temperature pretty much just oscillating in response to the heater coming on and going off...

[image]

Air pressure over 12 hours:

[image]

Really impressed with the BMP180!

ascillato2, 2018-06-27:
Hi @qingz2004 

Now there are the new commands ADD, SUB, MULT for used in rules for calibration. See wiki.

Thanks

closed 2018-06-27 by ascillato2

markfickett, 2023-12-01:
There is also a [TempOffset command](https://tasmota.github.io/docs/Commands/#sensors): "-12.6..12.6 = Set calibration offset value for reported temperature telemetry. This setting affects all temperature sensors on the device."