arendst/Tasmota #2873: PIR triggered music play

chedxb, 2018-05-31:
I have a Wemos D1 Mini attached with a PIR motion sensor (https://www.adafruit.com/product/189) which works well and now I would like to attach a simple 8 ohm 0.25W speaker (https://www.adafruit.com/product/1891) to Wemos D1 Mini and play a .wav or .mp3 file whenever PIR is triggered.

Is it possible with Tasmota?

Jason2866, 2018-05-31:
Simply no, without a Home Automation system.
With a HA system this is easy to realize with Tasmota 
and a connected PIR. You get a Info via mqtt that the PIR has 
"fired" and you can start playing a music file with a device connected to the HA system too.
So long story short you need some more equipment.

Jason2866, 2018-06-01:
Yes works stand alone but need too many resources...
Sound quality isnt that good.

ascillato, 2018-06-01:
Hi,

Thanks for share your idea. It is a nice project but it uses ALL the available resources from the ESP8266. For Tasmota to support that It will need to hung all operations (MQTT, pushbuttons, etc) to reproduce a sound. The actual Tasmota is handling commands and process every 50ms. The ESP8266 speaker requires to be flashed to use the CPU at 160Mhz and that each part of the sound be executed every 0.03 ms. So, it will need to stop to be a firmware for a Sonoff to be a Speaker and then come back when finished playing.

So, as the main goal for Tasmota is to be a home automation device using sonoff hardware, I recommend the exact same thing that @Jason2866 has stated. Use a dedicated device for sound and a home automation software (like homeassistant, openhab, domoticz, nodered, etc) that can manage that logic.

If you want, you can use another esp8266 device like NodeMCU to be a dedicated Speaker.

Tasmota can send the PIR trigger to your home automation system and that system can make to reproduce your required sound to a sound capable device.

I don't have now the link but I have read on the homeassistant forum, that someone has achieved that a PIR triggers a light and also play a barking dog sound when there is movement in the backyard. All with separated devices.

Hope this helps you.

Jason2866, 2018-06-01:
I think you can build what you want with that
https://www.aliexpress.com/item/Hot-sale-TF-card-U-disk-MP3-Format-decoder-board-module-amplifier-decoding-audio-Player-Free/32804211507.html?spm=a2g0s.13010208.99999999.260.777d3c00otaINZ
Just switch the power from the module on, and it will play the stored sound.

ascillato, 2018-06-01:
Love it! :+1:

ascillato, 2018-06-04:
@chedxb

If your issue is solved, please close it. Thanks

closed 2018-06-04 by chedxb