arendst/Tasmota #2874: Firmware update to Minimal: Failed (Space) but 524kB Free Program Space

th-s-op, 2018-05-31:
Hardware used : (German) OBI WiFi Plug, see https://github.com/arendst/Sonoff-Tasmota/wiki/Obi-Socket

I want to upgrade firmware from 5.13.0a to 5.14.0. firmware.bin file has size 428 KB (438.768 Bytes) and there are 524kB Free Program Space. When uploading the file, it ends in a "Upload Failed Not enough space" error.

Any suggestions?

Is there any extremely small precompiled firmware I could use which connects to some fixed WiFi connection and is capable to install a new firmware?

Output of "status 0":
15:48:45 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:48:45 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:13","StartupUTC":"2018-05-31T14:37:32","Sleep":0,"BootCount":12,"SaveCount":259,"SaveAddress":"FA000"}}
15:48:45 RSL: STATUS2 = {"StatusFWR":{"Version":"5.13.0a","BuildDateTime":"2018-05-10T10:38:57","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
15:48:45 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX","YYY"],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
15:48:45 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":524,"Heap":22,"ProgramFlashSize":2048,"FlashSize":1024,"FlashMode":2}}
15:48:45 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota","IPAddress":"192.168.xxx.yyy,"Gateway":"192.168.xxx.zzz","Subnetmask":"255.255.255.0","DNSServer":"192.168.xxxaaa","Mac":"5C:CF:7F:XX:XX:XX","Webserver":2,"WifiConfig":4}}
15:48:45 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu May 31 14:48:45 2018","Local":"Thu May 31 15:48:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:12","Sunset":"20:12"}}
15:48:45 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-05-31T15:48:45"}}
15:48:45 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-05-31T15:48:45","Uptime":"0T00:11:13","Vcc":3.496,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","RSSI":100,"APMac":"00:E0:4B:XX:XX:XX"}}}

th-s-op, 2018-05-31:
I noticed the differento to another working example is:
"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F002780","0C000000","00000092","00000000"]
Not knowing how, but I probably flashed the wrong way. Can I repair this?

th-s-op, 2018-05-31:
Another note: if I try to flash by OTA, I receive the following message:
UPGRADE = Failed Update error: ERROR[8]: Flash config wrong real: 1048576 IDE: 2097152

arendst, 2018-05-31:
As you noticed the ide used flash size of 2048 does not fit the hardware present flasn size of 1024.

Solution is to use the only supported ide memory model of 1M(no spiffs) and reflash via serial.

th-s-op, 2018-05-31:
For some reason, I had to use the "ESP8266 DOWNLOAD TOOL" to flash the firmware on those WiFi Sockets. My problem of understanding "Flash Size" in this tool was that it states "MBit" as a unit, and while I think the 1024 you mention are KBytes, I thought the right selection would be "8Mbit" (=1MByte = 1024 KBytes), but that didn't work and I retried with smaller values until it flashed fine. There is no "spiffs" option in that tool.

Thank you VERY much, I will do the serial reflash if there's no way out of it by software.

My RESPECT and THANKS for providing Tasmota AND HELP!

closed 2018-05-31 by th-s-op