arendst/Tasmota #2875: Toggle naming

almostlunatic, 2018-05-31:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff basic
- [x ] Provide the output of command ``status 0`` :
17:49:54 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:49:54 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:34:49","StartupUTC":"2018-05-31T15:15:05","Sleep":0,"BootCount":84,"SaveCount":205,"SaveAddress":"F7000"}}
17:49:54 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0k","BuildDateTime":"2018-04-12T21:17:09","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
17:49:54 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["E7B8",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
17:49:54 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":515,"Free":488,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
17:49:54 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0119","IPAddress":"192.168.1.150","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:13:80:77","Webserver":2,"WifiConfig":5}}
17:49:54 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_138077","MqttUser":"mosse","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:49:54 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu May 31 15:49:54 2018","Local":"Thu May 31 17:49:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"06:13","Sunset":"21:18"}}
17:49:54 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-05-31T17:49:54","AM2301":{"Temperature":23.9,"Humidity":70.0},"TempUnit":"C"}}
17:49:54 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-05-31T17:49:54","Uptime":"0T00:34:49","Vcc":3.459,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"E7B8","RSSI":100,"APMac":"44:CE:7D:F9:E7:BE"}}}}

Hello,
I use a Sonoff Basic for manipulating the blinds. For the other members of the family, I would like to rename the buttons Toggle 2, 3, 4 as Up, Down, Stop. I can't find the right place for it, can you help me?

Frogmore42, 2018-06-01:
Your request is not clear.  If you have physical switches, you can label them however you wish.  If you are expecting your family to use the system via their phones, you probably should have one of the home automation solutions or make a dashboard in NodeRED.  I don't believe you can rename them for mqtt or the web interface, but neither of those are family friendly.

almostlunatic, 2018-06-01:
Thank you! I was not accurate, that's true. I want to rename the labels of the blue buttons on the Tasmotas web page "Toggle 1" up to "Toggle 4" that is confusing for the others. When I'll return from travelling, I for sure will set up Domoticz or Node-RED, but for now a little change in the HTML code would be a solution.

reloxx13, 2018-06-01:
wont be so easy as you think.
xdrv_02_webserver.ino around line 550
```
for (byte idx = 1; idx <= devices_present; idx++) {
        snprintf_P(stemp, sizeof(stemp), PSTR(" %d"), idx);
        snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("<td style='width:%d%'><button style=\"line-height:5.0rem\" onclick='la(\"?o=%d\");'>%s%s</button></td>"),
          100 / devices_present, idx, (devices_present < 5) ? D_BUTTON_TOGGLE : "", (devices_present > 1) ? stemp : "");
        page += mqtt_data;
      }
```

almostlunatic, 2018-06-01:
Thanks, reloxx13! I don't have that file, only webserver.ino, but the code you refer exist there. I found it myself too in an hour. :)

Markkyboy, 2019-02-27:
I also wanted to rename the 'Toggle' button. The line of code presented by reloxx13 is actually not the right place to make the edit, at least not in my case (sonoff basic module).

xdrv_01_webserver.ino at line: 314 - which originally looks like this;

const char HTTP_DEVICE_CONTROL[] PROGMEM = "<td style='width:%d%%'><button onclick='la(\"?o=%d\");'>%s%s</button></td>"

edited to look like this (look for 'On/Off');

const char HTTP_DEVICE_CONTROL[] PROGMEM = "<td style='width:%d%%'><button onclick='la(\"?o=%d\");'>On/Off</button></td>"

End result (with other changes);

[image]

BigGeorgeTx, 2020-12-27:
It may be that it was a feature added after this thread, or perhaps I misunderstand something, but it seems like the Tasmota WebButton<x> On/Off would solve this problem.  I came across this thread while search for an answer.  The WebButton command did for me what I wanted to do (change the labels on buttons 1 to 3 for a 3 outlet smart plug).

meingraham, 2020-12-27:
Yes, `WebButton` is a newer command that came along quite a long time after this issue.

BigGeorgeTx, 2020-12-27:
I thought that might be the case.  I wanted to leave a comment so if someone else stumbled across this thread, they would know where to look.  Thanks for letting me know.