arendst/Tasmota #2876: Do timers and rules work on a Sonoff RF Bridge?

haegar33, 2018-05-31:
I flashed Tasmota 5.14.0 successfully on my RF bridge. I can access the bridge via the Web terminal and for example can learn an RF code assigned to rfkey1. I can manually issue this command in the terminal and switch on a light source.
Now I would like to use the 16 clock timers to realise a simple on/off clock for a light source without the need for any external device like MQTT brokers etc..
I define a timer
`timer1  {"Arm":1,"Mode":0,"Time":"16:22","Window":0,"Days":"1111111","Repeat":1,"Action":3}`

and then a rule:
`rule on clock#timer=1 do rfkey1 endon`

Rules are enabled but at 16:22 nothing is triggered. I cannot see any error or execution status

Jason2866, 2018-05-31:
I see no error. Try -for testing- instead of sending the rfkey1 command to publish a mqtt or a other command. If this works, the timer fires, if not the problem is the timer.
I have a spare RfBridge. I will flash it tomorrow with latest firmware and will try your rule
Can you post the logs? And status 0

haegar33, 2018-06-01:
Ok, another test this morning:
- mqtt client started on my NAS, messages from Sonoff Bridge coming in
- Test timer:
`timer1 {"Arm":1,"Mode":0,"Time":"09:52","Window":0,"Days":"1111111","Repeat":1,"Action":3}`
mqtt Log:
`{"Timer1":{"Arm":1,"Mode":0,"Time":"09:52","Window":0,"Days":"1111111","Repeat":1,"Action":3}}`

- Test rule:
` rule on clock#timer=1 do status 7 endon`
mqtt Log:
`{"Rule":"OFF","Once":"ON","Rules":"on clock#timer=1 do status 1 endon"}`

no message is triggered at 9:52. Apparently there is an issue with the timers...

Status:
`{"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018.05.29 11:36:17","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
{"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Heimnetz",""],"TelePeriod":300,"SetOption":["01008009","55818000"]}}
{"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
{"StatusNET":{"Hostname":"sonoff-5417","IPAddress":"192.168.178.39","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:A3:XX:XX","Webserver":2,"WifiConfig":3}}
{"StatusMQT":{"MqttHost":"192.168.178.23","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A31529","MqttUser":"haegar","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Fri Jun 01 07:57:11 2018","Local":"Fri Jun 01 09:57:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"05:01","Sunset":"21:43"}}
{"StatusSNS":{"Time":"2018-06-01T09:57:11"}}
{"StatusSTS":{"Time":"2018-06-01T09:57:11","Uptime":"0T00:12:01","Vcc":3.138,"Wifi":{"AP":1,"SSId":"Heimnetz","RSSI":78,"APMac":"E0:28:6D:ED:XX:XX"}}}
`

Jason2866, 2018-06-01:
Try the already compiled sonoff.bin from release branch.
Your settings will survive...
If timers works here your compiled version is the reason why

Jason2866, 2018-06-01:
Hi again,

i tried it, doesnt work too.

`00:00:00 Project sonoff RF-Bridge_2 (Topic sonoff-82589D, Fallback sonoff-82589D, GroupTopic sonoffs) Version 5.14.0b-2_3_0
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-82589D-6301...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-82589D-6301 with IP address 192.168.2.127
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/sonoff-82589D/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff-82589D/POWER = 
00:00:06 MQT: tele/sonoff-82589D/INFO1 = {"Module":"Sonoff Bridge","Version":"5.14.0b","FallbackTopic":"sonoff-82589D","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff-82589D/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-82589D-6301","IPAddress":"192.168.2.127"}
00:00:06 MQT: tele/sonoff-82589D/INFO3 = {"RestartReason":"Software/System restart"}
09:39:43 MQT: tele/sonoff-82589D/STATE = {"Time":"2018-06-01T09:39:43","Uptime":"0T00:00:14","Vcc":3.181,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
09:43:13 CMD: timer1 {"Arm":1,"Mode":0,"Time":"10:45","Window":0,"Days":"1111111","Repeat":1,"Action":3}
09:43:13 MQT: stat/sonoff-82589D/RESULT = {"Timer1":{"Arm":1,"Mode":0,"Time":"10:45","Window":0,"Days":"1111111","Repeat":1,"Action":3}}
09:43:22 CMD: rule on clock#timer=1 do rfkey1 endon
09:43:22 MQT: stat/sonoff-82589D/RESULT = {"Rule1":"OFF","Once":"OFF","Free":479,"Rules":"on clock#timer=1 do rfkey1 endon"}
09`

haegar33, 2018-06-01:
Ok, I tried a reflashing, for over OTA (but was not sure whether is was actually working) and the a serial upload with esptool.py of the pre-compiled V5.14.0. 
Timers don't trigger anything...
Latest Status:
`{"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:34","StartupUTC":"2018-06-01T08:53:04","Sleep":20,"BootCount":11,"SaveCount":73,"SaveAddress":"F5000"}}
{"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018.05.15 15:34:22","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
{"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Heimnetz",""],"TelePeriod":300,"SetOption":["01008009","55818000"]}}
{"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
{"StatusNET":{"Hostname":"sonoff-5417","IPAddress":"192.168.178.39","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:A3:15:29","Webserver":2,"WifiConfig":3}}
 #{"StatusMQT":{"MqttHost":"192.168.178.23","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A31529","MqttUser":"haegar","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Fri Jun 01 08:55:38 2018","Local":"Fri Jun 01 10:55:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"05:01","Sunset":"21:43"}}
{"StatusSNS":{"Time":"2018-06-01T10:55:38"}}
{"StatusSTS":{"Time":"2018-06-01T10:55:38","Uptime":"0T00:02:34","Vcc":3.140,"Wifi":{"AP":1,"SSId":"Heimnetz","RSSI":78,"APMac":"E0:28:6D:ED:6F:38"}}}
`

haegar33, 2018-06-01:
Accidentally closed

Jason2866, 2018-06-01:
I have the same behaviour. Timer doesnt fire rule
May we both are overlooking something, but i have no idea what.
I think it is a bug.

ascillato, 2018-06-01:
Hi,

Both have **"Rule1":"OFF"**

you need to turn rules on. On the console please type: **rule 1**

Jason2866, 2018-06-01:
Hello @ascillato 
sometimes the solution is so easy (when you know it).
In german we say: Den Wald vor lauter Bäumen nicht gesehen. 
-> You didnt see the forrest because there are so many trees

THX!

haegar33, 2018-06-01:
My „forrest“ was the rule status message where I read rule:OFF,Once,ON instead of realising that „Once“ is a separate parameter with its own value ON/OFF :o)
It also works for me now ;) Thanks to all for helping out!

closed 2018-06-01 by haegar33