arendst/Tasmota #2878: Static IP vs Dynamic IP (pros and cons)

sparkgap, 2018-06-01:
I'm thinking of changing from dynamic to static IP.   When using the Web Console to make a configuration change or to check status of my Sonoff Basic modules I need to browse to it's IP address.  My wifi router, after a reset, will assign different IP addresses, therefore every time I need to access a specific Sonoff, I need to go into the router settings to determine it's IP address.   This is an extra step.  It would be much easier to use fixed IP's and have a scheme similar to this: 
sonoff-1 = 192.168.1.101 
sonoff-2 = 192.168.1.102
sonoff-3 = 192.168.1.103
etc,  etc.
Is this a good or bad idea and why?????

znanev, 2018-06-01:
@sparkgap Regarding the IP addresses, you could use MAC reservations in your router to assign pseudo-static IP addresses to your sonoffs, i.e. if sonoff-1 has MAC address aa:bb:cc:dd:ee:ff, make a DHCP reservation for this MAC to assign IP 192.168.1.101 etc.

You can usually forward any unused ports to internal devices when using NAT. Let's say you could map port 8001 to be forwarded to port 80 of IP 192.168.1.101, port 8002 to be forwarded to port 80 of IP 192.168.1.102 and so on. But I would advise you not to directly expose your sonoffs to the Internet, even though you can set a password for the Web UI (command is WebPassword _password_).

It is better if you can setup a VPN to your home network and once you establish a secure connection from outside, you could access your internal devices like if you were at home.

Even better is if you can setup some kind of home automation system (Home Assistant, OpenHAB, Domoticz etc.). Then connect to your home automation system from outside, either through SSL (HTTPS), or after establishing a VPN connection to your home network.

Jason2866, 2018-06-01:
Just my 2 cents. Static IP is a "trap". If you overlook sometime something you are hunting strange errors...
If you cant always the same IP-Address (with MAC Binding), i would change this limited router!

reloxx13, 2018-06-01:
use mdns, so you dont need to know the ip

e.g.
http://kitchen.local

sparkgap, 2018-06-02:
I was wrong.  My Linksys E1200 does not require port numbers with static IPs.  I was looking at the port forwarding configuration section, which I do not want to use.   So I see no risk difference between a static and dynamic IP.   So, looks like I'll go with static, for matching topic and IP pairs for convenience.
sonoff-1 = 192.168.1.101
sonoff-2 = 192.168.1.102
sonoff-3 = 192.168.1.103
etc.... etc.....

Frogmore42, 2018-06-02:
There is a reason DHCP was invented decades ago.  Static IP management is hard to do well.

Good news, your router does support reserved addresses
https://www.linksys.com/us/support-article?articleNum=135673

So use that, but don't set Tasmota to be static unless you keep VERY good records.

sparkgap, 2018-06-02:
OK DHCP wins.  Thanks to all.

closed 2018-06-02 by sparkgap