arendst/Tasmota #2879: Sonoff POW - MQTT messages contains inaccurate energy / power values

Stefan300381, 2018-06-01:
Hi,

this is a beginner question. I've searched the wiki/forum etc. but couldn't find anything.
First, here is status 0 output:

`09:45:08 CMD: status 0
09:45:08 MQT: stat/sonoffpow1/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff"],"Topic":"sonoffpow1","ButtonTopic":"0","Power":1,"PowerOnState":4,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:45:08 MQT: stat/sonoffpow1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T01:19:01","StartupUTC":"2018-06-01T07:26:07","Sleep":0,"BootCount":9,"SaveCount":29,"SaveAddress":"F7000"}}
09:45:08 MQT: stat/sonoffpow1/STATUS2 = {"StatusFWR":{"Version":"5.14.0b","BuildDateTime":"2018-06-01T07:24:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
09:45:08 MQT: stat/sonoffpow1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["schoofiwlan",""],"TelePeriod":0,"SetOption":["00008009","55818000"]}}
09:45:08 MQT: stat/sonoffpow1/STATUS4 = {"StatusMEM":{"ProgramSize":557,"Free":444,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
09:45:08 MQT: stat/sonoffpow1/STATUS5 = {"StatusNET":{"Hostname":"sonoffpow1-6336","IPAddress":"192.168.2.174","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"2C:3A:E8:4E:98:C0","Webserver":2,"WifiConfig":3}}
09:45:08 MQT: stat/sonoffpow1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4E98C0","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:45:08 MQT: stat/sonoffpow1/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 01 08:45:08 2018","Local":"Fri Jun 01 09:45:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:51","Sunset":"20:45"}}
09:45:08 MQT: stat/sonoffpow1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:45:08 MQT: stat/sonoffpow1/STATUS10 = {"StatusSNS":{"Time":"2018-06-01T09:45:08","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.000}}}
09:45:08 MQT: stat/sonoffpow1/STATUS11 = {"StatusSTS":{"Time":"2018-06-01T09:45:08","Uptime":"0T01:19:01","Vcc":3.475,"POWER":"ON","Wifi":{"AP":1,"SSId":"schoofiwlan","RSSI":68,"APMac":"78:8A:20:D7:68:F7"}}}`

Here is my issue:
Currently I'm testing my POW with a 11W LED bulb. The web interface shows accurate information in regards to power consumption. Sometimes it changes between 11W and 12W and back. But the MQTT messages show big differences. While the bulb is on or rather inserted into the socket, it lists up to 22W and while it's not in the socket MQTT messages show values between 0W and 2W. Any idea why?

`09:20:15 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:20:15","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":10,"Factor":0.32,"Voltage":234,"Current":0.134}}
09:20:17 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:20:17","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":20,"Factor":0.75,"Voltage":234,"Current":0.114}}
09:20:18 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:20:18","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":11,"Factor":0.40,"Voltage":234,"Current":0.114}}
09:21:14 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:21:14","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":21,"Factor":1.00,"Voltage":234,"Current":0.089}}
09:21:15 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:21:15","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":10,"Factor":0.48,"Voltage":234,"Current":0.089}}
09:22:44 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:22:44","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":21,"Factor":0.43,"Voltage":233,"Current":0.212}}
09:22:45 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:22:45","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":11,"Factor":0.21,"Voltage":233,"Current":0.212}}
09:23:29 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:23:29","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":22,"Factor":1.00,"Voltage":234,"Current":0.092}}
09:23:31 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:23:31","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":10,"Factor":0.45,"Voltage":234,"Current":0.096}}
09:23:49 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:23:49","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.079}}
09:29:02 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:29:02","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":20,"Factor":0.00,"Voltage":234,"Current":0.000}}
09:29:04 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:29:04","ENERGY":{"Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":0,"Power":10,"Factor":0.43,"Voltage":234,"Current":0.104}}
09:30:43 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:30:43","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":2,"Factor":0.05,"Voltage":234,"Current":0.128}}
09:30:53 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:30:53","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.063}}
09:33:37 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:33:37","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":1,"Factor":0.00,"Voltage":234,"Current":0.000}}
09:33:47 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:33:47","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.065}}
09:34:31 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:34:31","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":2,"Factor":0.11,"Voltage":234,"Current":0.086}}
09:34:41 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:34:41","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.000}}
09:35:34 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:35:34","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":2,"Factor":0.00,"Voltage":234,"Current":0.000}}
09:35:44 MQT: tele/sonoffpow1/SENSOR = {"Time":"2018-06-01T09:35:44","ENERGY":{"Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":0,"Power":0,"Factor":0.00,"Voltage":234,"Current":0.097}}`

Thanks in advance

Stefan

Frogmore42, 2018-06-01:
The sensor is not accurate, especially at low power levels.  Think about it this way.  Full scale for the sensor is about 20A and 300V, which is 6000W.  Even if it were accurate to 0.1% of full scale that puts the error at 6W.  A more typically error is 1-2%, which would make the error 60-120W.

Having said that, the reason the number is jumping around so much is because it is determining the power factor to be very different at each of those measurements.  Put an old school incandescent light build there and you will get much better results.

The number is 20W when the power factor is 1, but 10W when the factor is 0.4 or so.  This is consistent.  It also indicates the sensor has a hard time figuring out the power factor of an led bulb.  This too is expected as they usually have very unusual power utilization signatures that do not work well with simple measurements.

This is not a Tasmota issue.

Stefan300381, 2018-06-01:
alright, thanks a lot for the explanation. the strange thing is just that I never saw such exceptional values when watching the web interface with the "realtime" display.

closed 2018-07-31 by stale[bot]