arendst/Tasmota #2880: POW PowerDelta tresholds

Lu-Fi, 2018-06-01:
Please add or use existing tresholds (eg. PowerHigh / PowerLow) for PowerDelta.
It's currently not really usable if Consumption is normaly ~0-5W.
Every minimal sensor difference will fire an MQTT Message.

IVI053, 2018-07-07:
I have the same problem. Only percentage for `powerdelta` has the problem that it fires to much on low values and to less on high.
So for high i can set `teleperiod` according to my demands but as the measurements are not too accurate you have noise when nothing draws power and 0 to 1-2W is already 100-200%. While that change is insignificant setting it to 200% to filter this out means at arround 100W already getting a too huge needed change IMHO.

ascillato, 2018-08-26:
Hi,

What about using rules to set exactly the energy report that fit your requirements like:

```
rule 1
rule 4
rule on Energy#Power>5 do publish stat/sonoff/energy %value% endon
```

ascillato2, 2018-08-30:
Hi,

Please, share your results when you have time. Thanks.

closed 2018-08-30 by ascillato2

IVI053, 2018-09-09:
Wouldn't it flood mqtt with every change above 5 with this rule?

ascillato, 2018-09-09:
Hi, no because the rule is one-shot