arendst/Tasmota #2881: Info Sonoff Dual R2 command for manage relay

opendore, 2018-06-01:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff Dual R2 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

I have a Sonoff Dual R2 that I'm using to open and close a shutter and it's working very well.
On the sw1 and sw2 I connected the wall switches to open and close the shutter with physical buttons.
I would like to know if there is a command to send via MQTT (I have configured homebridge and Mosquitto with my Raspberry 3) in order to raise the shutter or lower it not completely.
There is a command that keeps the relay1 / 2 2 on for a number of seconds that I decide?

Thanks in advance for your support.

ascillato, 2018-06-01:
Hi,

Please, expand a little the explanation for this troubleshooting.

How is connected?

The relays are controlling the motor or and sending the up and down command?

opendore, 2018-06-01:
The relays controlling the motor.

ascillato, 2018-06-01:
And in your shutter, the motor stops by itself when reaching the full open and close?

ascillato, 2018-06-01:
What I understood from your explanation is that you want to have a full open and full close command and also a % open and % close command. That is right?

If that is your case, you can make this new commands using rules:

_Using last Tasmota v5.14.0b_

* Turn Rules On.

On the console: `rule 1`

* Command **open by x seconds** and command to **close by x seconds**:

On the console: `rule on event#open do backlog power1 on; ruletimer1 %value% endon on rules#timer=1 do power1 off endon on event#close do backlog power2 on; ruletimer2 %value% endon on rules#timer=2 do power2 off endon`

You can call that new commands
* by console as: `event open=1`
* by console as: `event close=1`
* by mqtt as: `cmnd/sonoff/event open=1`
* by mqtt as: `cmnd/sonoff/event close=1`

where 1 is 1 second.

opendore, 2018-06-02:
ascillato thank you!

It was just what I was looking for !!!
At the time I was able to configure also homebridge and I would like to be able to choose the percentage of opening / closing of the shutter through Siri and HomeKit .... you say that it is possible?
At the moment I can control it vocally to open or close the shutter completely.

ascillato, 2018-06-03:
For the percentage of opening and close, you need to experiment how much time it takes and make the conversion from % to Seconds in homebridge. So you can say open 50% and your homebridge should send the command `cmnd/sonoff/event open=4` for example

opendore, 2018-06-08:
I managed to implement the rule and it works correctly.
What I still can not do is integrate this function into Homebridge, here I put a part of the configuration file.
If I go to see Consolle tells me command unknown, if I launch it directly from Consolle works.

Do you have any ideas?

Thanks to everyone in advance.

{
                "accessory": "mqtt-switch-tasmota",

                "name": "Fessure SU",

                "url": "mqtt://192.168.0.103",
                "username": "mqttuser",
                "password": "kawasaki",

                "topics": {
                             "statusGet": "stat/sonoff/POWER1",
                             "statusSet": "cmnd/sonoff/event open=1"
                          }
}

ascillato, 2018-06-08:
what mqtt message do you get on the console?

opendore, 2018-06-08:
20:36:34 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}

ascillato2, 2018-06-08:
that is the result. I meant the previous log message. what tasmota is receiving.

Please, type `weblog 4` in the console and repeat the test and post the results

opendore, 2018-06-08:
20:43:37 CMD: weblog 4
20:43:37 MQT: stat/sonoff/RESULT = {"WebLog":4}
20:43:44 CFG: Salvato nella flash in F6, Conteggio 3156, Bytes 2560
20:43:52 RSL: Topic Ricevuto cmnd/sonoff/event open=1, Dimensione Dati 2, Dati ON
20:43:52 RSL: Gruppo 0, Indice 1, Comando EVENT OPEN=, Dati ON
20:43:52 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
20:44:01 WIF: Controllo connessione...
20:44:01 WIF: Connesso
20:44:18 RSL: Topic Ricevuto cmnd/sonoff/event open=1, Dimensione Dati 3, Dati OFF
20:44:18 RSL: Gruppo 0, Indice 1, Comando EVENT OPEN=, Dati OFF
20:44:18 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
20:44:23 RSL: Topic Ricevuto cmnd/sonoff/event open=1, Dimensione Dati 2, Dati ON
20:44:23 RSL: Gruppo 0, Indice 1, Comando EVENT OPEN=, Dati ON
20:44:23 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
20:44:24 RSL: Topic Ricevuto cmnd/sonoff/event open=1, Dimensione Dati 3, Dati OFF
20:44:24 RSL: Gruppo 0, Indice 1, Comando EVENT OPEN=, Dati OFF
20:44:24 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}

ascillato, 2018-06-08:
Hi

I set the following simple rule on my sonoff: `on event#test do power %value% endon`

And from my MQTT I sent `cmnd/cocina/event test=1`

and the results in the console are:
```
18:10:14 SRC: MQTT
18:10:14 RSL: Topic Recibido cmnd/cocina/event, Tamaño de Datos 6, Datos test=1
18:10:14 RSL: Grupo 0, Índice 1, Comando EVENT, Datos test=1
18:10:14 MQT: stat/cocina/RESULT = {"Event":"Done"}
18:10:14 RUL: EVENT#TEST performs "power 1"
18:10:14 SRC: Rule
18:10:14 RSL: Topic Recibido /power, Tamaño de Datos 1, Datos 1
18:10:14 RSL: Grupo 0, Índice 1, Comando POWER, Datos 1
18:10:14 MQT: stat/cocina/RESULT = {"POWER":"ON"}
18:10:14 MQT: stat/cocina/POWER = ON
```
So as it is working for me in last Tasmota v6.0.0a, and think that in your case you are using an old version of tasmota. Can you provide the STATUS 0 output?

opendore, 2018-06-09:
Hi, this is my log:

 

21:11:06 CMD: status 0

21:11:06 RSL: Topic Ricevuto /status, Dimensione Dati 1, Dati 0

21:11:06 RSL: Gruppo 0, Indice 1, Comando STATUS, Dati 0

21:11:06 RSL: stat/sonoff/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

21:11:06 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:42:59","StartupUTC":"2018-06-09T19:28:07","Sleep":200,"BootCount":1315,"SaveCount":3169,"SaveAddress":"F9000"}}

21:11:06 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:36:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}

21:11:06 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["NETGEAR24",""],"TelePeriod":60,"SetOption":["00804009","55818000"]}}

21:11:06 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}

21:11:06 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4224","IPAddress":"192.168.0.3","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:9F:30:80","Webserver":2,"WifiConfig":3}}

21:11:06 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.103","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F3080","MqttUser":"mqttuser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}

21:11:06 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 09 20:11:06 2018","Local":"Sat Jun 09 21:11:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:47","Sunset":"20:51"}}

21:11:06 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-09T21:11:06","Switch1":"ON","Switch2":"ON"}}

21:11:06 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-09T21:11:06","Uptime":"0T00:42:59","Vcc":3.183,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR24","RSSI":46,"APMac":"B0:7F:B9:D2:3F:88"}}}

21:11:16 MQT: Tentativo di connessione...

21:11:16 MQT: Connesso

21:11:17 MQT: tele/sonoff/LWT = Online (salvato)

21:11:17 MQT: cmnd/sonoff/POWER = 

21:11:17 MQT: Sottoscrivi a cmnd/sonoff/#

21:11:17 MQT: Sottoscrivi a cmnd/sonoffs/#

21:11:17 MQT: Sottoscrivi a cmnd/DVES_9F3080/#

opendore, 2018-06-21:
I try to explain myself better, if I run the console command it works fine.
My problem is that I can not integrate this command into the homebridge configuration file.
Currently I can raise or lower completely my 4 blinds.
What I would like to be able to manage by sending a command that closes the contact for a few seconds and integrate it into the Homebridge config file.

{
"accessory": "mqtt-switch-tasmota",

            "name": "Fessure SU",

            "url": "mqtt://192.168.0.xxx",
            "username": "xxx",
            "password": "xxx",

            "topics": {
                         "statusGet": "stat/sonoff/POWER1",
                         **"statusSet": "cmnd/sonoff/event open=1"**
                      }
}

if I integrate this, the command in the homebridge config file does not work and MQT comes out on the console: stat / sonoff / RESULT = {"Command": "Unknown"}

if I write directly from the console the command works correctly.
cmnd/sonoff/event open=1

Any idea?

ascillato, 2018-06-21:
Your home automation is sending by MQTT the command

EVENT OPEN=, Dati OFF

Instead of

EVENT OPEN=1

So, is an issue in your home automation. I don't know about homebridge. I
can't help you with that.

ascillato2, 2018-07-10:
Hi,

Could you manage to solve your issue?

closed 2018-08-22 by ascillato2