arendst/Tasmota #2883: How to set up the timer?

seanchs, 2018-06-01:
Apologize for posting this question here since I can't find an appropriate place for questions. 

I understand the 5.14 can set up timer for Tasmota (I am using Sonoff Basic). However, I tried to set it up but it won't work. I looked through the wiki and release note but I am not very coding savy to figure it out. What I am trying to do: 
1) Turn on the switch at 7:25am every morning 
2) Turn off the switch after 30 minutes or at 7:55am 

Highly appreciate any help in advance!

Lawrencezarb, 2018-06-01:
When you set the timer you need to make sure you have the ARM box ticked.
Also it is important that the Sonoff obtains the correct time from the NTP server.

Frogmore42, 2018-06-01:
> 

Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)

seanchs, 2018-06-01:
Thanks Lawrencezarb and Frogmore42. I searched the discussion group but no luck. I'll post this question in the discuss group.

ascillato, 2018-06-01:
Hi, seems to be a problem in your configuration.

Please provide the output of the `status 0` command in the console, as the template asks. Also a screenshot of the timer configuration page.

Lawrencezarb, 2018-06-01:
Are you sure that your has the correct time. "STATUS 7" in the console window will tell you what time/date your unit thinks it is.

seanchs, 2018-06-01:
Hi Lawrencezarb,

My local time is: 4:17pm. With "STATUS 7", the console shows:

00:16:46 MQT: stat/SW-HotWater/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 01
23:16:46 2018","Local":"Sat Jun 02 00:16:46 2018","StartDST":"Sun Mar 25
02:00:00 2018","EndDST":"Sun Oct 28 03:00:00
2018","Timezone":1,"Sunrise":"04:50","Sunset":"20:46"}}

I guess my time is not right in Sonoff Basic. Would you mind sharing how to
set the time of Sonoff Basic?

Thanks a lot,

Sean

ascillato, 2018-06-01:
Hi,

You have to use `timezone `command, Please read the wiki.

seanchs, 2018-06-01:
Hi Adrian,

Got it. I just set up the right timezone via Console. Looks like it works
now. Thanks a lot.

Sean

closed 2018-06-01 by ascillato

BlackH137, 2019-01-21:
Thanx :heart:

une18, 2019-11-09:
Hi! need help with this command. I don´t understand what to entes in the command line.

`00:05:02` MQT: tasmota_maja_ees/tele/STATE = {"Time":"1970-01-01T00:05:02","Uptime":"31500T19:47:42","UptimeSec":2721671262,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Minu nett1","BSSId":"F8:1A:67:C0:EA:C6","Channel":12,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:06:03 CMD: TimeDST 0,2,11,7,13,120
00:06:03 MQT: tasmota_maja_ees/stat/RESULT = {"TimeDst":{"Hemisphere":0,"Week":2,"Month":11,"Day":7,"Hour":13,"Offset":120}}
00:06:22 CMD: time
00:06:22 MQT: tasmota_maja_ees/stat/RESULT = {"Time":"1970-01-01T00:06:22"}``

une18, 2019-11-09:
I feel, that the Instructions can not be written more complicated.
[image]

Got it finally fixed!
Enter - Time 14:16:37   - and it sets the right time 
Enter -  Timezone +2   - and your time zone will be +2

Why is it not written like that in: https://github.com/arendst/Tasmota/wiki/Commands#how-to-use-commands?????????

meingraham, 2019-11-09:
The vast majority of users use NTP or an RTC to ensure that the time on their devices is set properly. Your device will not retain it's time if it restarts since you set the time manually.

The documentation as written was able to successfully get you to the setup you needed. Because your setup is one out of 4 million Tasmota installations, the documentation has to cover configurations for far more scenarios than just yours. Documenting all of these is impossible.

Do recall that this is an open source project which you are getting to use at no cost to you. All the code and documentation is performed by volunteers (i.e., unpaid). Please forgive us for not catering to your specific situation with more care.

Now that you have resolved your issue, please close this issue.

Regards.