arendst/Tasmota #2887: Raw RF Code over mqtt on Sonoff Basic or other ESPs with separate RF Receiver

raddi, 2018-06-01:
Heyho,

i'm not good at coding, but in the 5.12 version i was able to implement the RC-Switch library into the Tasmota code (sorry for playing with the code :D ) and added some code snippets to receive RF codes with an RF receiver on GPIO14 (Sonoff Basic) and send the code in an individual MQTT topic. 
Now in 5.14 the code has completely changed (MQTT is now a seperate code tab) and i just can't get my solution back to work.

Why do I want this feature? 
I have a lot of 433mhz RF transmitters (like RF wall switches or remotes, which send unique rf codes) and i use Node Red to receive the code via mqtt, analyse the code in functions and switch different things with it over mqtt (for example: if mqtt-rfcode message "783847384" --> switch sonoff 1, 2 and 3 on/off over MQTT). 

First i used a Wemos D1 Mini to receive the codes from the RF Receiver and send them over mqtt but i was able to implement my code in the Tasmota firmware for my Sonoff basics. I connected the RF receiver to GPIO14 and was able to use Tasmota on the basic AND receive the rf code and send it as a string over mqtt on the same device (no need for the Wemos anymore, yay)

I know there are some RF features in the Tasmota firmware for the Sonoff RF and RF Bridge but it would be really cool to just add a RF receiver to a Sonoff basic or Arilux/Magichome on GPIO14 (or any other unused GPIO) and send the raw code over MQTT.

I could use the 5.12 Tasmota i customized but i really want AriLux with HSB Color so i need the 5.14 Tasmota Firmware. :)

So my dream Tasmota-device would be: AriLux LC01 with HSB Color control and an RF Receive on GPIO14 for raw RF Code.

Please be kind,
thank you!

Jason2866, 2018-06-02:
Hello @raddi 
kenmcc implemented RCSwitch library. I use it with a WemosD1 mini. It should work with a Sonoff too...
You can find the Tasmota Fork here:
https://github.com/kenmcc/Sonoff-Tasmota/tree/rcswitch

mientki, 2018-12-31:
I bought a WiFi module: https://www.aliexpress.com/item/Wifi-Remote-Switch-1CH-AC-DC-7-36V-10A-Smart-Home-Relay-Module-433Mhz-Receiver-Support/32902426871.html 
[image]
How should I configure it correctly to publish in the mqtt broker all RF codes that I receive? I would like him to behave exactly like Sonoff RF Bridge. I would like this module, together with two RF contact sensors, to be used for the garage door. My device seems to be similar SV sonoff.

The following settings are not correct.
[image]

Jason2866, 2018-12-31:
Device is not supported. We dont have this device....
You have to figure it out by yourself if possible and how

mientki, 2018-12-31:
Can you give me a hint, guide me, how to do it to work?

Jason2866, 2018-12-31:
Where is the Rf receiver? You have to follow where the output pin from the receiver goes to.
If it goes to a ESP Gpio pin you can use it as Rf receiver mqtt gateway. If not you are stuck

Jason2866, 2018-12-31:
Easiest solution buy a sonoff sv and add a rf-receiver to a free GPIO pin.
This one will work https://www.aliexpress.com/item/Transmitter-Module-Relay-Receiver-315MHz-Code-Learning-433MHz-Remote-Control-Transmitter-Receiver-Module/32825621368.html?ws_ab_test=searchweb0_0,searchweb201602_4_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=fb02f930-e1c9-4b05-a77b-1485a00cc03e-0&algo_pvid=fb02f930-e1c9-4b05-a77b-1485a00cc03e&transAbTest=ae803_3

GianCann, 2019-01-01:
@Jason2866 can you explain me how to configure Tasmota on D1 Mini to work with RF Receiver?
I connected the receiver on pin D2 and SetOption28 to 1 but RfRaw command from console is 'Unknow'.

Thank you (and Happy New Year!)

Jason2866, 2019-01-01:
You have to trace where the signal from the 2 inner pins (the are connected) from the rf receiver
goes. Does it go to the ESP? If yes just check which GPIO and configure it as rf receive.
Thats all. If no cut trace(s) and connect it to a free GPIO pin from ESP
The rf button on board will have no function

Jason2866, 2019-01-01:
@GianCann 
Use sonoff.bin from here http://thehackbox.org/tasmota/
You have configured as module generic and gpio as rf-receive?

mientki, 2019-01-01:
Middle pins are not connected to ESP. Solder wire to GPIO14 and set to Tasmota RFrecv. It works.

lordzid, 2020-12-17:
can i use esp 8266?