arendst/Tasmota #2888: Timer commands return MQT: stat/pow_12/RESULT = {"Command":"Unknown"}

MimbaMonkeyHouse, 2018-06-01:
I'm trying to set timers in Tasmota. From the Wiki, I understand that the command I need to use is:
```
Timer1 {"Arm":1,"Mode":0,"Time":"10:00","Window":0,"Days":"0","Repeat":1,"Output":1,"Action":1}

```
I tried the actual examples in the Wiki, but I always get:
```
MQT: stat/pow_12/RESULT = {"Command":"Unknown"}
```
I downloaded the last classic binary firmware, but no luck either.
Here is my Status 0:
```
07:32:32 CMD: status 0
07:32:32 MQT: stat/pow_12/STATUS = {"Status":{"Module":6,"FriendlyName":["POW12 Heater"],"Topic":"pow_12","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:32:32 MQT: stat/pow_12/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:07","StartupUTC":"2018-06-01T23:16:25","Sleep":0,"BootCount":6997,"SaveCount":14141,"SaveAddress":"F7000"}}
07:32:32 MQT: stat/pow_12/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:18:02","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:32:32 MQT: stat/pow_12/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"SysLog":3,"LogHost":"192.168.86.101","LogPort":514,"SSId":["indebuurt1","monkey_house"],"TelePeriod":30,"SetOption":["00000009","55818000"]}}
07:32:32 MQT: stat/pow_12/STATUS4 = {"StatusMEM":{"ProgramSize":487,"Free":516,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
07:32:32 MQT: stat/pow_12/STATUS5 = {"StatusNET":{"Hostname":"pow_12-2099","IPAddress":"192.168.86.72","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"2C:3A:E8:1A:68:33","Webserver":2,"WifiConfig":3}}
07:32:32 MQT: stat/pow_12/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.86.101","MqttPort":1883,"MqttClientMask":"POW_12","MqttClient":"POW_12","MqttUser":"test","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:32:32 MQT: stat/pow_12/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 01 23:32:32 2018","Local":"Sat Jun 02 07:32:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":8}}
07:32:32 MQT: stat/pow_12/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:32:32 MQT: stat/pow_12/STATUS10 = {"StatusSNS":{"Time":"2018-06-02T07:32:32","ENERGY":{"Total":141.188,"Yesterday":3.339,"Today":2.579,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
07:32:32 MQT: stat/pow_12/STATUS11 = {"StatusSTS":{"Time":"2018-06-02T07:32:32","Uptime":"0T00:16:07","Vcc":3.182,"POWER":"OFF","Wifi":{"AP":2,"SSId":"monkey_house","RSSI":48,"APMac":"70:3A:CB:34:37:DD"}}}
```

ascillato, 2018-06-01:
You can set timers with the menu. Please go to configuration/timers

MimbaMonkeyHouse, 2018-06-02:
I have no configuration timer menu on my Tasmota 5.14 (classic bin). This is a Sonoff POW unit. Do you need to activate this menu before it appears?

ascillato, 2018-06-02:
ahhhh, you are right. Sorry.

Clasic does not have timers. Please, use another one.

ascillato, 2018-06-02:
Please, see the [readme](https://github.com/arendst/Sonoff-Tasmota/blob/development/README.md), for a list of features that are included or not in each firmware version.

MimbaMonkeyHouse, 2018-06-02:
Thanks for that @ascillato. I assumed Classic was the regular firmware. Updating from 5.12 Classic, it looks like I need to first upload the minimal firmware. My workload just increased, but we'll get there...

closed 2018-06-02 by MimbaMonkeyHouse